theyosh/TerrariumPI #649: [BUG]: Relays not switching through TerrariumPI but do through python script in console

mink8mink, 2022-01-22:
**Setup:**
 - TerrariumPI version: Version: 4.1.0
 - Python version: Python 2.7.16 
 - Raspberry PI: Pi4 Model B

**Describe the bug**
 Relays not switching through TerrariumPI but do through python script in console

I have tried everything under the sun. TerrariumPI will install the relay but will not switch them???? I can switch then through a test script in python though. I am desperate. I've googled to the ends of the earth and cannot get this to work. I did get it to work with TP v3.......

PLEASE HELP!

theyosh, 2022-01-23:
This is just a simple GPIO relay. But in your script you have BCM port numbering. That needs to be translated to physical numbers. https://pinout.xyz/

So, based on:

`pinList = [2, 3, 4, 17, 27, 22, 10, 9]`

You should use the translated number as address at a GPIO relay.

2 = address 3
3 = address 5
4 = address 7
17 = address 11
27 = address 13
22 = address 15
10 = address 19
9 = address 21

The above script will never work, as it runs in a loop and does not give the correct output.

So I would suggest to add a GPIO relay with address 3, and see if that works. Then just add all the other relays as GPIO relay type with the translated address in the list above.

mink8mink, 2022-01-23:
Yosh, 

You are the man! Thank you so much...Something so simple. I did what you said and am now getting it to recognize the relay..One more thing...After I added them I tried to set them up in an area with a timer and get an error when i save...

**SAVE ERROR 
Area could not be added. Invalid isoformat string:"**

It then shows this under status of the area:

**Day: Invalid date - 9:46:46 PM (Invalid date)**

Thank you for creating this awesome piece of software and for supporting it's users as well as you do. We appreciate you man :)

mink8mink, 2022-01-26:
Ok, I think I got it. I believe it was the safari on Mac that was causing the save issue. I got no issue when using the phone browser or Firefox. 

I also noticed that setting up areas are touchy. If you mess up delete everything, reboot and try again. I also had to setup my relays as GPIO inverse as they were doing the opposite from what they were supposed to do. Hope this helps someone!

theyosh, 2022-01-27:
Hi, the deleting of areas did have a bug #652 . That will be fixed in the next release. And the area part is pretty new, and complex, so there could be still some bugs... 

Ah, crap, yes, I use the browser locale to translate the date and time to a timestamp. I am unable to test Safari, but it was working for Chrome, which is almost... almost the same engine.. But lately, Apple and Safari are doing strange stuff...

About the GPIO relays. Somehow there are two version, or you can connect them differently, so I do not know why sometimes the inverse version is needed. But as I have had this in the past, I just added an inverse option to it.

mink8mink, 2022-01-27:
Thanks Yosh! Its been running perfect ever since. 

The great thingabout it is that regardless of the internet connection the relays will stay on schedule....This is the opposite of all the "smart" switches out there which won't turn on or off if the connection drops connected. My setup is in the garage and wireless is spotty but TPI works regardless.

mink8mink, 2022-01-28:
Yes. Sir!

 

So can we close this issue?

closed 2022-01-28 by theyosh