theyosh/TerrariumPI #650: [BUG]: TerrariumPi does not install on Raspberry Pi 3B+

vlreinier, 2022-01-25:
**Setup:**
 - TerrariumPI version: Latest
 - Python version: -
 - Raspberry PI: 3B+

**Describe the bug**
During installation the following error code appears and TerrariumPi can not be started:
100 cannot access 'venv/lib/python*      packages/bluepy/bluepy-helper' No such file or directory.
Line 204: bc: command not found ./install.sh

**To Reproduce / Installation steps**
Clean install of Raspberry Lite on Raspberry Pi 3B+ no desktop version, SSH enabled and changed default port.

**Expected behavior**
A working TerrariumPi webserver :)

**Screenshots**
[image]

vlreinier, 2022-01-25:
It works on the Raspberry Lite Legacy image, awesome! :)

theyosh, 2022-01-25:
Hmm, yeah, the latest image is not working. I thought I made some fixes for that, but it can be that I did not release that code..

So I am happy you figured it out. For now, stick with the old legacy image.

vlreinier, 2022-01-25:
Ah so it was not me, good to know!

Is there support for creating custom lighting schemes with leds?

theyosh, 2022-01-27:
For now, I think I close this issue. There are some code fixes in the queue for the next release: https://github.com/theyosh/TerrariumPI/commits/4.x.y.z

closed 2022-01-27 by theyosh

timmakken, 2022-02-16:
Hi Yosh, im having the same issue with latest TerrariumPi. any thoughts on how i should fix this i installed bc manually so no issues there but the open-csv-headless is just stuck without any errors. ( same specs as @vlreinier )

vlreinier, 2022-02-16:
Easiest way to fix this is using the legacy 32bit no desktop version. That one works for sure. Dont forget a correct networking interfaces file, or you will encounter sleep mode issues.

theyosh, 2022-02-16:
If it does not work please follow the installation instructions: https://theyosh.github.io/TerrariumPI/install/#raspberry-pi