theyosh/TerrariumPI #651: [FEATURE]: Custom Intensity Lighting schemes for RGBCCT leds

vlreinier, 2022-01-25:
First off, very nice software!

Is there a possibility to create custom lighting schemes? It would be cool to control the intensity for LEDS over the day for different colors. Below some examples:

[image]
[image]

vlreinier, 2022-01-26:
Apparantly im great at answering my own questions :P I hope my impatience and questions are useful for other enthousiasts who are looking for the same functionality. 

So what did i do? I used the dimming duration in a new area to accomplish slowly increasing and decreasing the light intensity.

However, am i correctly stating that there is no possibility to do a increase or decrease again later in the day for the same relay (without creating new areas or relays)? If that is the case, then i will keep this feature request open for using a graph to schedule lighting within one area like so:
[image]

And i still dont understand the dimming frequency not visibly changing?

theyosh, 2022-01-28:
Hi,

the only way I can think of is making your own script dimmer. In this script you can then make the logic for the dimming values of the different channels based on the 'dim' value of TP. Then you have one single dimmer which is easier to control.
And how the frequencies work with all kind of dimmers and leds is for me a riddle. Everybody has different experiences, and there a lot of different kind of leds that all behave differently. I only added the option as the code has support for it. If it does not work, to bad, I can't fix it :(

For lights, there is no variation option. But it is a nice idea to make. As all the other areas have the variation option, it should not be hard to make. But let me think a bit about this.

And you posted a nice graph, but making that in TP is very hard. So that will not be doable for me to make. Because I can I know how many lights there should be. And I have been experimenting with small graphs in the environment part, but that gets cluttered, and does not look nice.

So, I would go for a custom dimmer script, to make it a bit easier to control in TP.

vlreinier, 2022-01-29:
I will look into it!

A graph (with multiple overlays) might be hard but below i described how i would envision the proces for a somewhat doable solution. Adding a graph to visualize this would be optional.

- Create modus 'clock' for a certain type
- Choose a start and end time for day or night
- Create a (or multiple) intensity point(s) for  certain timepoints within the start and end time

The rate of intensity increases and decreases could be calculated by looking at the slope between each two points (time relative to intensity) I dont know if this would be easy to implement in the current software, but i will also take a look at it.

Visualizing this would result in something like:
[image]

theyosh, 2022-02-03:
Hmm, I have thought about this, but I am afraid that it is to much now to add to TP4. And there is a way to to this your own using script dimmers. As a script dimmer is asked every 30 seconds an update, you can make your own logic to change the dimmer value. The only downside is, that you cannot show it in the TP4 graphs. Because when you change the value of the dimmer state, TP4 will try to get it back to the max value.

So, the graph will show only 100% power, but actually, you change the dimmer value based on your own logic.

Making all kind of timers is tricky and difficult. So for now, I am tending to leave this up to you with a script dimmer...

vlreinier, 2022-02-03:
I understand, ive been looking into the code but it would need a big refactor.

When i have some spare time, i will write the code for this and let you know :) If you are willing to implement this feature/change?

For now i will use the script dimmer or the normal dimmer solution. Thanks!

theyosh, 2022-02-07:
I am always open for PRs. I have done that in the past. It could be that the solution is not quite what I want, and then it could be that there is some code updates needed. But I am willing to help if the big part is done.

closed 2022-02-25 by theyosh