theyosh/TerrariumPI #653: Cool Free Tip for TPI users.

mink8mink, 2022-01-27:
Sign up for the free PiTunnel account and be able to view the TPI web interface and remote ssh into your setup without messing with port forwarding or VPN's...Really cool tool and its FREE for one device.

https://www.pitunnel.com/

mink8mink, 2022-01-27:
Both pretty cool options. Perfect for mobile use.

yotitote, 2023-01-05:
Hey, could you please send me a screenshot of your settings in the pitunnel? I tried to connect mine raspberry and it doesn't work.  Thank you in advance for your answer.

theyosh, 2023-01-05:
@yotitote I guess you run this: `pitunnel --port=80 --name=mytunnel --http` you should change the portnumber to 8090 like this: `pitunnel --port=8090 --name=mytunnel --http`

yotitote, 2023-01-05:
@theyosh actually I run the good one and I get the error message "Tunnel Connection Failed".

theyosh, 2023-01-05:
Than I do not know. I do not use pitunnel

yotitote, 2023-01-05:
Sure, thanks.