theyosh/TerrariumPI #654: Database IntegrityError: UNIQUE constraint failed: Sensor.id -> Can't add DS18B20

Dave0408, 2022-01-28:
Good evening,

for some reason I was missing a temperature sensor DS18B20. Unfortunately I can't add him back either manually or via the automatic detection.

Error message:
```
Error Speichern
Sensor could not be added. Object 1wire temperature named 'Temperatur_oben' at address '28-011928001632' cannot be stored in the database. IntegrityError: UNIQUE constraint failed: Sensor.id
```

```
pi@raspberrypi:/sys/bus/w1/devices $ ls
28-011927c83e7d  28-011927f38213  28-011928001632  w1_bus_master1
```

In http://ipadress:port/api/sensors/ it is not listed.

28-011927c83e7d is up and running
28-011927f38213 is up and running

Any idea how to fix my database issue?! Can I manually connect to TerrariumPI database and execute some SQL (Drop * WHERE Sensor.ID = '28-011928001632') ? 

Thanks. 

Dave

theyosh, 2022-01-28:
Yeah, I have seen those also. This happens to me when during the startup the sensor is not correctly loaded, but during the scanning it is working correctly and found. Strange, but happens.

In the 4.x.y.z branch this is already fixed. There I will then still use the known sensor in the database and load that one.

So, if you are able to switch to the 4.x.y.z branch, do a reinstall `sudo ./install.sh` then this issue should be fixed. And you have a lot of small bug fixes. But switching to the 4.x.y.z branch needs a rerun of my installer!

Dave0408, 2022-01-28:
I am already using Terrarium PI V. 4.1.0
But maybe not at the latest one....

theyosh, 2022-01-28:
4.1.0 is the latest released version. But we have updates :) https://github.com/theyosh/TerrariumPI/compare/4.1.0...4.x.y.z

So this bug will be fixed in the 4.2.0 release, but if you are in a hurry, you can switch to the 4.x.y.z branch, so you are a bit ahead of the latest release.

Dave0408, 2022-01-28:
OK. Thanks for this information. I will close this issue as it is already known and fixed for future updates.

closed 2022-01-28 by Dave0408