theyosh/TerrariumPI #655: SHT20 Temperature Measurement incorrect/inaccurate

wolli1234, 2022-02-01:
Hi Yosh,

I switched my sensor setup to SH20 for humidity and temperature.

The Humidity shows and updates correctly, but the temperature values behave a bit weird. Mostly the temperature stays at 19.394 C, through a along period of updates it does not make any change.

I compared this to a different sensor python script: [https://github.com/feyzikesim/sht20](https://github.com/feyzikesim/sht20)

Executing this script, both values (humidity and temperature) are different to the values in TP. But while the humidity value updates regularly and is just a bit off to the script value, the temp value does not update for a very long time and the offset is getting more obvious.

I played around a bit [here](https://github.com/theyosh/TerrariumPI/blob/main/hardware/sensor/sht2x_sensor.py) changing the values for wait time and timeout, but it did not change anything.

Do you have an idea why this might happen and why it only affects temperature but not humidity?

**Setup:**
 - TerrariumPI version: 4.1
 - Python version: 3
 - Raspberry PI: 3B

theyosh, 2022-02-01:
Hi,

to be short: the results of https://github.com/feyzikesim/sht20 are better :) I have no explanation for that. The current code for this sensor is also found somewhere/donated by someone. So I have not written the code myself.

So, I have included that library in my software, but it looks like that there is not much change for me. But I have put the sensor in soil and measure the temperature of the soil of a plant.

So, if you say that the other library works better, I can just use that.

wolli1234, 2022-02-01:
That would be great! I would try it myself to include but I'm not familiar enough to know how and where to put the code exactly.

theyosh, 2022-02-01:
Ok, I have updated the code. It is only available in the 4.x.y.z branch for now. And that branch is a bit ahead with new features.

Therefore, you need to rerun the installer after getting the latest 4.x.y.z code. And the restart TP4 and it should just work. Hopes you will get better readouts

wolli1234, 2022-02-02:
> Ok, I have updated the code. It is only available in the 4.x.y.z branch for now. And that branch is a bit ahead with new features.
> 
> Therefore, you need to rerun the installer after getting the latest 4.x.y.z code. And the restart TP4 and it should just work. Hopes you will get better readouts

Thank you!

For documentation:

    git remote set-branches 4.x.y.z
    git fetch -v
    git checkout main
    git pull

Is that correct to switch back from main to 4.x.y.z?

wolli1234, 2022-02-02:
I switched to 4.x.y.z and the new code for sht20 was updated. Unfortunately after a while I could not access the frontend anymore.
This happened several time before on main 4.1.0 but I did a backup of the data folder und by removing terrariumpi.db the frontend was back again. But noe this is not working anymore.

In general the missing frontend seemed to be related to enclosures and area setups with using dependencies feature. Might be a different bug, but i want to mention it here, as i switched to 4.x.y.z and can't get TP running properly again.

wolli1234, 2022-02-02:
Thank you. Regarding GUI bug, I forgot to rerun the installer and that solved the problem. I still think there is some weird bug with the GUI, because it happened several times, but I will open another ticket if it reappears ;).

wolli1234, 2022-02-02:
Anyway SHT20 gives way smoother values now!

theyosh, 2022-02-03:
Yes, same here. So then I close this issue now and will this be added to the next release. Thanks for find this library and testing!

closed 2022-02-03 by theyosh

wolli1234, 2022-02-04:
Thank you for the great support!