theyosh/TerrariumPI #656: [BUG]:

93420, 2022-02-03:
**Setup:**
 - TerrariumPI version: 4.1.0 (docker image) (but same issue with v4.0.0 installed manually
 - Python version: 3
 - Raspberry PI: 3B

**Describe the bug**
No area detail for second enclosure in Dashboard page

**To Reproduce / Installation steps**
2 enclosures, at least 1 area in each enclosure, then go to dashboard page. Only area for enclosure 1 are displayed.

**Expected behavior**
Area details should be there for all enclosures

**Screenshots**
No specific error
[https://ibb.co/ZJ11857](https://ibb.co/ZJ11857)
[https://ibb.co/ZJ11857](https://ibb.co/c8mrTqw)

**Additional context**
[image]
[image]

theyosh, 2022-02-03:
Hi, this is not correct! So I have to look into this. I think I have an idea why this is happening, but I need to do some testing.

I hope to have an answer in 48 hours...

theyosh, 2022-02-03:
Euh... did you click on the second enclosure title? Only one enclosure is shown at a time.

93420, 2022-02-03:
Oh.. yes if I click on the second, it changes...
Is there a way to show all?

So.. my bug becomes a suggestion: Add a clickable icon on the title.;-)

Thank you for your quick response an investigation !

-------- Message d'origine --------

theyosh, 2022-02-05:
Hmm, I am not sure. I like the clean interface, and adding some icons or other indicators to that part, makes it a but messy.

So I made the click area in the title a bit bigger, so when you move the mouse over it, you will see a hand.
Also, now you know it. And you are the first that mention this. So, maybe it is something that is maybe not direct clear, but I guess the most people will figure it out.

And when you had a third area, it was also clear that you have to click on the title, as you would have 2 closed parts.

So for now, I think I leave it as is. I just made the clickable area bigger and hoping people will notice it when hovering over the areas

theyosh, 2022-02-13:
So, I made up my mind. I leave the GUI as is. Adding extra icons for open en closing is not looking nice. And once you know it, it is clear.

closed 2022-02-13 by theyosh