theyosh/TerrariumPI #657: [HELP]: Updated to version 4 (finally), how to disable automatic MiFlora detection?

Swiftnesses, 2022-02-05:
Hello,

Finally pulled apart the coop for a rebuilt and updated to V4.

My Coop is surrounded by 15 MiFlora sensors and they're overwhelming the system, can I stop them being automatically added?

theyosh, 2022-02-05:
Hmm, not sure if it will work, but the only way, is to let TP4 running, and try to login. Then go to every sensor and use the 'Ignore' option in settings menu (wrench icon)

[image]

This will take some time, but this is for now the only way to ignore sensors. Once ignored, they will not be used in TP4 anymore.

Swiftnesses, 2022-02-05:
I'll try that. It seems V4 might not work for me anyway :-(

Before I used webhooks to send raw information to Home Assistant (it was JSON), now it appears the messages are not JSON, and instead are subject / body style messages? I setup the webhook as a service and created a notification type for relay change, but nothing works. I also tried MQTT, but had no luck getting information to show. 

Any ideas?

theyosh, 2022-02-05:
I would suggest first to fix your sensor issues. For notifications please open new issue.

There is also a more technical way to exclude the sensors, but then you need at least have TP4 started up. So that all sensors are scanned once.

If this all takes more then 2 minutes, it could be that TP4 is automatically restarting. If that is the case, we have to fix that first. You can see in the `/var/log/syslog` if there are restarts. As long there are restarts, TP4 will never fully start, and therefore, we cannot ignore sensors.

Swiftnesses, 2022-02-05:
Perhaps it's my Pi - clicking ignore says success and reloads, but they're still there. Delete says success too, but when it reloads they're still there...

theyosh, 2022-02-05:
They can still be showing on the page, but they should not be updated anymore. And after 1 or 2 minutes, they should also not been shown anymore. This is due to the caching of the data. So give it a bit of time.

Also, all the ignored sensors can be found at the settings page. There should be a list of ignored devices. If that list is filled, then I am sure TP4 will not update those anymore. And they will remove from the interface.

Swiftnesses, 2022-02-05:
OK, I'll keep going and check back later. With regards getting my webhook working again, is there a way to get JSON output like in version 3 (rather than subject / body). Worst case it looks like I can poll the API endpoint, but I'd rather not. I do have MQTT, if I can get updates via that service?

Swiftnesses, 2022-02-05:
Sorted out the MiFloras (took over an hour of clicking!). Figured out the API and updated all my NodeRed flows, just need to get the webhopok sending JSON data again now. Can you help @theyosh?

theyosh, 2022-02-05:
Nice, I am looking into the webhook code. Not sure if I can post the same information. But you should get all the data that you would get trough an API call plus the subject and message you can customize with variables.

But, I am fixing in the 4.x.y.z branch for now. So give me some time. I guess that in 1 or 2 hours I can have an update

Swiftnesses, 2022-02-05:
As long as it's JSON I can translate it and adjust my NodeRed flows.

Before I used information for the door sensor (so I could tell / confirm, if it had opened or closed) and relays (if they changed state - they drive a linear actuator for the door) via the webhook data. 

I assume this was removed then when you moved from V3 --> V4?

theyosh, 2022-02-05:
Hmm, I think you should switch to 4.x.y.z to get the updates. I changed the code a bit. But in order to know which data is available by which names, you should open the notification message, and look at the placeholders. But that is not yet in the main branch I think. Then it will be hard(er) to figure out the data. A help could be: http://[RPI]:8090/api/swagger/#/Button/getButton and look at the Buttons get action. Or https://theyosh.github.io/TerrariumPI/api/ should give the same information.

So if you are able to switch to 4.x.y.z branch, do a git pull, rerun the installer and restart, then it should all work.

Swiftnesses, 2022-02-05:
I did a pull and am restarting... to confirm, the data sent to the webhook endpoint will now be valid JSON?

theyosh, 2022-02-05:
It should be. I used an old test script on my server that I used for TP3 webhook testing. And that code is just working. So then I conclude that the JSON data is there again, and valid. Else my test script would not accept the post data

Swiftnesses, 2022-02-05:
[image]

I tried this as a test, but nothing is received at the webhook (if I make a manual post it works, so the webhook is okay). The webhook only accepts POST and JSON.

theyosh, 2022-02-05:
If I am correct, you are still on the main branch. I am not sure how well the notification works in the version. If you are able to switch to 4.x.y.z branch, rerun the installer, and clear browser cache. You should see placeholder data here.

If you are on the 4.x.y.z branch, it should work. Then I suspect browser cache.

Swiftnesses, 2022-02-05:
What's the best way to change branch?

theyosh, 2022-02-05:
- sudo service terrariumpi stop
- git remote set-branches origin '*'
- git fetch -v
- git checkout 4.x.y.z
- git pull
- sudo ./install.sh
- sudo service terrariumpi start

That should do it.

Swiftnesses, 2022-02-05:
Installing now, I'll update soon - THANK YOU :)

Swiftnesses, 2022-02-05:
I now see the placeholders and extra notification options, but no luck getting data to the webhook.

Swiftnesses, 2022-02-05:
Hang on, I setup the 30 second relay type and got something, let me hunt further!

theyosh, 2022-02-05:
The contents could be different, and when TP4 starts, there are some broadcast messages. You will get those as well as 'a gift' ;)

But that data is different then what you expect. So I am hoping your NodeRed can ignore post messages not containing the data it needs. But all message should be JSON valid.

Swiftnesses, 2022-02-05:
[image]

Okay, so the 30 second update one works (I get nice JSON for each relay). The one on change doesn't seem to fire.

theyosh, 2022-02-05:
Hmm, maybe not all are correct implemented. But I think you should use the `relay_toggle` option for that.

theyosh, 2022-02-05:
I have probably duplicated the notification types from sensors. So the update should be left out for relays, and toggle is the right one

Swiftnesses, 2022-02-05:
Yes, toggle seems to fire for on / off for relays. Perfect.

When I get the door sensor attached tomorrow (magnet), do I use toggle too?

theyosh, 2022-02-05:
Could it be that you have accidentally ignored that one? If you look at the settings page at the list of ignored sensors. Could it be that it is in that list? Remove it from the list and save the settings.

This is error is here, because you have already added it.. Or at least TP4 has a record of it in the database.

Swiftnesses, 2022-02-05:
That did it. Weird...

BTW what's the best way to do a clean install of x.y.z, add `--branch 4.x.y.z` to the clone statement?

Swiftnesses, 2022-02-05:
Oh, and so many issues are fixed on this branch for me - for example on main, the toggles in the UI were not updating for relays, I thought it was my rubbish Pi3! Nice to know it wasn't :)

theyosh, 2022-02-05:
git clone --branch 4.x.y.z https://github.com/theyosh/TerrariumPI.git

That should clone directly into it. But this will also download all the git changes. But it will allow you to switch back to the main branch if you want. The 4.x.y.z is used for new features. So the main branch you can always update without problems. The 4.x.y.z could have more bugs... 

But I am happy if you run the 4.x.y.z branch, because that means another one is testing new code :D

theyosh, 2022-02-05:
An try this: https://matrix.to/#/#terrariumpi:theyosh.nl

Swiftnesses, 2022-02-05:
Sadly, the ducks were taken one night by a fox (really upsetting), there's some chickens now and better security. The fox has been back, but not succeeded, thankfully!

theyosh, 2022-02-05:
Damm... hmmm, motion detection with animal recognition and sounding an alarm... that would be a nice feature to have.. But to bad of the ducks.. But you can't really blame the fox. It is nature...

Swiftnesses, 2022-02-06:
It was there before the upgrade and vanished, I'll check.

Swiftnesses, 2022-02-06:
Getting there, the only thing I can't get working is the door sensor (button), I have it across GND and GPIO24 and have entered address 18 into the UI, it doesn't seem to update?

Pinout: https://i.stack.imgur.com/Oj8z8m.jpg

Swiftnesses, 2022-02-06:
OK, actually it does update, but takes 30 seconds or so and isn't instant like V3, any way to improve this @theyosh ?

Swiftnesses, 2022-02-06:
Oh man, can't get the camera to rotate either, the pi live rotational settings don't seem to have an affect :(

Swiftnesses, 2022-02-06:
It appears that none of the Pi live cam setting work - resolution, rotation etc etc.

theyosh, 2022-02-06:
Hmm, it could be it need a restart to make the changes work for the live cam. Because I have 180 degrees rotation what does work.

So, I look into the part of reloading the camera after saving. If you setup the camera correctly and restart, is it then ok?

Swiftnesses, 2022-02-06:
I'm trying, for some reason it started to display a black picture (although raspistill / raspivid still work). Trying to a reinstall.

V3 was absolutely solid, sometimes you shouldn't fix what isn'y broken (in my case, it seems!).

theyosh, 2022-02-06:
In this kind of cases, I would use a new SD card, install a clean OS, and install the new version. Then you can still swap back to the old system.

If the image is black, could it be that a white balance setting is not set correctly?

Swiftnesses, 2022-02-06:
HI @theyosh I've done that multiple times, camera is back now, but relay statuses are back to acting all funky in the UI (don't update, and show off when they're on, it's a bit of a mess). I might pull the card and do a clean install of V3 sometime, but it's already a weekend lost! Perhaps one to revisit in the future, for now I've just enabled remote GPIO and will issue commands from Home Assistant.

Swiftnesses, 2022-02-06:
Update, no idea how, but relays are now working and the camera came back! I've re-setup all my Node-RED flows and Home Assistant logic and it's all working 100%. While V4 seems a little more unstable, it's worth it in the end - especially as I know you'll continue to work hard to make it perfect. Thanks @theyosh, you can now consider me one of the x.y.z pull crew :)

closed 2022-02-07 by Swiftnesses

theyosh, 2022-02-07:
Hi, I fixed the RPI live webcam reloading. It was totally broken... :( But it is now fixed.

About the door status... and slow status update. As TP4 is using a DB way more then then TP3, it is also a bit slower as more data is red and written. I have noticed this also. But I am almost done with the full rewrite and all features, and working on a new GUI which should be way faster. And with that, I will look into the slow status updates due to the database actions. I know it can be improved, but it will take a lot of code changing. So that will happen later.

My experience is that when using a Raspberry PI 4 with a USB3 stick as OS drive, it goes really fast. Also using a USB disk on a TP3 would improve.