theyosh/TerrariumPI #658: [BUG]:  (V4, x.y.z) - ResourceWarning: Enable tracemalloc to get the object allocation traceback

Swiftnesses, 2022-02-07:
Version: Latest x.y.z

Noticed this in the syslog, is it related to the webcam stream, perhaps we need to close the chunks gracefully?

```
Feb  7 07:36:31 raspberrypi python[512]: ResourceWarning: Enable tracemalloc to get the object allocation traceback
Feb  7 07:36:31 raspberrypi python[512]: /home/pi/TerrariumPI/venv/lib/python3.7/site-packages/gevent/pywsgi.py:930: ResourceWarning: unclosed file <_io.BufferedReader name='/home/pi/TerrariumPI/webcam/4e96d2cfe49351a5d47209f6889f3bdd/chunk_159.ts'>
Feb  7 07:36:31 raspberrypi python[512]:   for data in self.result:
```

Swiftnesses, 2022-02-07:
[image]

Also seen in the daemon.log

Swiftnesses, 2022-02-07:
I updated, but the web UI isn't loading, seems to be looping...

Swiftnesses, 2022-02-07:
Made another pull and the UI is back...

theyosh, 2022-02-13:
Hmm, I think I close this issue, as I cannot fix this. I have tried to change the code a bit which should reduce those errors. I do not see them over here. 

Also, it is a warning, and therefore TP4 will keep running.

closed 2022-02-13 by theyosh