theyosh/TerrariumPI #660: Startup Loop / No Interface after 4.1 Update

naichalt, 2022-02-08:
Hello, 

after the update from 4.0.0 to 4.1.0 i have no interface/UI.
Tried to restart service, reboot system. 
The errors from the am2302 are known. (I think its broken :) ) but that was not a Problem until now.

It seems that the system is trapped in some kind of loop? (Hence the starting TerrariumPi 4.1.0 part in the log)

```
2022-02-08 17:03:34,867 - INFO    - terrariumEngine       - Added new new sensor 1wire temperature named '1-Wire sensor measuring temperature' at address '28-012042cbf179' to database with value 27.31C in 0.88 seconds.
2022-02-08 17:03:44,025 - INFO    - terrariumEngine       - Loaded 2 sensors in 12.65 seconds.
2022-02-08 17:03:44,026 - INFO    - terrariumEngine       - Loading existing relays from database.
2022-02-08 17:03:44,033 - INFO    - terrariumEngine       - Scanning for new relays ...
2022-02-08 17:05:39,497 - INFO    - terrariumEngine       - Starting TerrariumPI 4.1.0 ...
2022-02-08 17:05:39,568 - INFO    - terrariumEngine       - Loaded 28 settings in 0.07 seconds.
2022-02-08 17:05:40,027 - INFO    - terrariumEngine       - Loading existing sensors from database.
2022-02-08 17:05:40,961 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named '1-Wire sensor measuring temperature' at address '28-0120425e575c' with value 26.62C in 0.93 seconds.
2022-02-08 17:05:41,825 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named '1-Wire sensor measuring temperature' at address '28-012042cbf179' with value 27.12C in 0.86 seconds.
2022-02-08 17:05:41,826 - INFO    - terrariumEngine       - Scanning for new sensors ...
2022-02-08 17:05:52,745 - INFO    - terrariumEngine       - Loaded 2 sensors in 12.72 seconds.
2022-02-08 17:05:52,747 - INFO    - terrariumEngine       - Loading existing relays from database.
2022-02-08 17:05:52,753 - INFO    - terrariumEngine       - Scanning for new relays ...
2022-02-08 17:07:48,230 - INFO    - terrariumEngine       - Starting TerrariumPI 4.1.0 ...
2022-02-08 17:07:48,301 - INFO    - terrariumEngine       - Loaded 28 settings in 0.07 seconds.
2022-02-08 17:07:48,764 - INFO    - terrariumEngine       - Loading existing sensors from database.
2022-02-08 17:07:49,680 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named '1-Wire sensor measuring temperature' at address '28-0120425e575c' with value 26.56C in 0.91 seconds.
2022-02-08 17:07:50,547 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named '1-Wire sensor measuring temperature' at address '28-012042cbf179' with value 27.00C in 0.87 seconds.
2022-02-08 17:07:50,549 - INFO    - terrariumEngine       - Scanning for new sensors ...
2022-02-08 17:08:01,454 - INFO    - terrariumEngine       - Loaded 2 sensors in 12.69 seconds.
2022-02-08 17:08:01,454 - INFO    - terrariumEngine       - Loading existing relays from database.
2022-02-08 17:08:01,460 - INFO    - terrariumEngine       - Scanning for new relays ...
2022-02-08 17:09:56,992 - INFO    - terrariumEngine       - Starting TerrariumPI 4.1.0 ...
2022-02-08 17:09:57,064 - INFO    - terrariumEngine       - Loaded 28 settings in 0.07 seconds.
2022-02-08 17:09:57,527 - INFO    - terrariumEngine       - Loading existing sensors from database.
2022-02-08 17:09:58,406 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named '1-Wire sensor measuring temperature' at address '28-0120425e575c' with value 26.44C in 0.87 seconds.
2022-02-08 17:09:59,264 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named '1-Wire sensor measuring temperature' at address '28-012042cbf179' with value 26.88C in 0.86 seconds.
2022-02-08 17:09:59,266 - INFO    - terrariumEngine       - Scanning for new sensors ...
2022-02-08 17:10:10,202 - INFO    - terrariumEngine       - Loaded 2 sensors in 12.67 seconds.
2022-02-08 17:10:10,203 - INFO    - terrariumEngine       - Loading existing relays from database.
2022-02-08 17:10:10,213 - INFO    - terrariumEngine       - Scanning for new relays ...
2022-02-08 17:12:05,668 - INFO    - terrariumEngine       - Starting TerrariumPI 4.1.0 ...
2022-02-08 17:12:05,739 - INFO    - terrariumEngine       - Loaded 28 settings in 0.07 seconds.
2022-02-08 17:12:06,200 - INFO    - terrariumEngine       - Loading existing sensors from database.
2022-02-08 17:12:07,126 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named '1-Wire sensor measuring temperature' at address '28-0120425e575c' with value 26.38C in 0.92 seconds.
2022-02-08 17:12:07,987 - INFO    - terrariumEngine       - Loaded sensor 1wire temperature named '1-Wire sensor measuring temperature' at address '28-012042cbf179' with value 26.75C in 0.86 seconds.
2022-02-08 17:12:07,990 - INFO    - terrariumEngine       - Scanning for new sensors ...
2022-02-08 17:12:18,979 - INFO    - terrariumEngine       - Loaded 2 sensors in 12.78 seconds.
2022-02-08 17:12:18,982 - INFO    - terrariumEngine       - Loading existing relays from database.
2022-02-08 17:12:18,992 - INFO    - terrariumEngine       - Scanning for new relays ...
```

theyosh, 2022-02-08:
This is very strange. The problem is that scanning for new relays never looks like to finish.

> 2022-02-08 17:05:52,753 - INFO    - terrariumEngine       - Scanning for new relays ...
> 2022-02-08 17:07:48,230 - INFO    - terrariumEngine       - Starting TerrariumPI 4.1.0 ...

As I restart TP4 after 2 minutes of no activity. The 2 minutes should be enough to start up and start updating.

But somehow, your system gets stuck on relay scanning. Which is strange, and I have never seen that before.

Could it be that a USB relay is not fully connected? It also looks that it does not load any relays. Did you configure any?

Else I am out of options. Other then reinstalling it all over again to see if it then works. Or try to switch to the 4.x.y.z branch which has maybe newer versions of certain libraries that could cause this.

theyosh, 2022-03-09:
Closing due to no response

closed 2022-03-09 by theyosh