theyosh/TerrariumPI #674: [FEATURE]: Sensirion SHT4X sensor

Jexkin, 2022-03-06:
**Is your feature request related to a problem? Please describe.**
I tried to set my SHT40 sensor up using the SHT3x settings but it didn't work, so I think something must be added for the SHT40/SHT4X's to work?

**Describe the solution you'd like**
SHT4x sensor readouts for temperature and humidity.

**Additional context**
I already have the hardware so I could test everything, just like last time (in case you remember my name).
If it matters somehow: I got the SHT40 sensor Adafruit sells (adafruit.com/product/4885)
Thanks!

theyosh, 2022-03-09:
Hmm, I have to look into this. I found a library that could read out this sensor. But that takes some time to implement. Give me a week. I think I know more then.

Jexkin, 2022-03-09:
Sure, take all the time you need. I'm out of town so I couldn't test it anyway at the moment. I should be able to once you figure something out! Thanks!

theyosh, 2022-03-09:
Hmm, sorry, it is already added :)

But it is only on the 4.x.y.z branch. You can switch to it using the following commands:

- git remote set-branches origin '*'
- git fetch -v
- git checkout 4.x.y.z
- git pull

And then rerun the installer `sudo ./install.sh` and restart TP4. This should add a SHT4X sensor.

Jexkin, 2022-03-10:
I see, I'll do that as soon as I am home again!
Will keep you updated if everything works :)

Jexkin, 2022-03-10:
Hey, so I just switched to the 4.x.y.z branch (did all the steps) but now I can't access my admin panel anymore. Already rebooted the Pi4 a few times but still nothing.
I probably am just overlooking something. IP is correct, so I'm unsure what I'm doing wrong.
The switch to 4.x.y.z was flawless.

theyosh, 2022-03-17:
Hmm, without any error it is hard to judge. It is the development branch, but I am able to run it. So if you are willing to try once more, and try this: https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/

theyosh, 2022-03-17:
Hmm found the issue. Do a `git pull` and rerun the installer. Then it should all work.

Jexkin, 2022-03-17:
Hey, so I tried it once again after the update and I'm still not able to access the admin panel.
It still should be <pi-ip>:8090, right?
Anyway, I ran the debug and this the output:

[image]

I'm a bit confused because it still says `Starting TerrariumPI 4.2.0`, shouldn't it say `Starting TerrariumPI 4.x.y.z`?

theyosh, 2022-03-17:
Sorry, but I made a small mistake in the installer. So try a final `git pull` and rerun the installer. That should take care.

Jexkin, 2022-03-17:
I did `git pull` but it seems like it didn't update anything?
I just checked out the branch on GitHub and it says that it was last updated 3 hours ago.
Still can't connect to the admin panel and I still get the exact same output when using the debugging tool.

theyosh, 2022-04-01:
Hi, I think our lasts messages just crossed. Because this should all be fixed now, and install the correct python libraries.

Could you test once more? `git pull` and then `sudo ./install.sh` and restart TP4.

theyosh, 2022-04-01:
And can I then ask to look at #210 ?

Jexkin, 2022-04-01:
Hi, I'll do it ASAP. Coincidentally I am on a train again.

In regards of the pics, I actually ordered new enclosures and once I am done with them, I'll make sure to post them :) Should be pretty interesting. Though it may take another 2 months because it definitely needs some time to finish. Hope that's alright!

Also I probably will open up another Issue in regards on how to dim flourescence bulbs with a dimmable electrical ballast, which hopefully should be possible in some way. I may just try to dim it with my current IRF3708 module but that may not work.
But that's future stuff!
(Reason behind that is that LEDs usually don't output UVA and I just want to give my reptiles as naturalistic light as possible and there are some floursescence bulbs that offer such a thing)

theyosh, 2022-04-01:
Hmm, regards to dimming, that could be hard. There are multiple questions regarding to dimming, so I would suggest also to look in the closed issues. Not all dimming issues are solved, but at least they all have quite some information and tests that could help.

The IRF3708 should work with the existing RF- dimmer type... I think. Can't test that.

Jexkin, 2022-04-02:
I'll do that once I have all the stuff but it'll probably still take a few weeks because _money_ lol.

Also, I honestly don't know if it's just me but I still have problems with 4.x.y.z ...
I do it as you say ->
[image]

then i run the installer with `sudo ./install.sh`, this is what I see once it's done
[image]

but after the reboot I am not able to access the admin panel via `TPi-ip:8090` anymore and I honestly don't know why. Any clue?

As soon as I switch back to the main branch and run the installer everything works as usual.

theyosh, 2022-04-02:
This is my fault. Made a typo in the requirements.txt file. So I would like to ask once more to do :+1: 

- `git pull`
- `sudo ./install.sh`

And then it should be working again.

Jexkin, 2022-04-02:
Works now, thanks!
 Just as some sort of proof:
 
[image]

[image]

Thank you very much! Will the SHT4x be added to TPi 4.3 / main branch? Or should I just stay on the 4.x.y.z branch?
In regards of the fluorescent lights and dimming, I'll probably going to open a new issue IF I run into issues. If not, then I'll hopefully remember to send some neat pictures in the pinned issue `#210`.

Jexkin, 2022-04-02:
Also I just looked at issue #643 
I see that you added a `Dimmer max dim value` and it's at max 100.
Would it theoretically possible to increase it to 1024, as the IRF520 module has a value range between 0 and 1024?
Or am I missing some knowledge? And sorry if I should have posted it in the issue.

theyosh, 2022-04-06:
[image]

You should be able to enter 1024 in the 'Dimmer max dim value'. When you enter a value higher then 0 then the old dimming code will activate. Else I use a new way of dimming. But it seems that the new way does not always work well.

So you should just be able to enter the 1024 to get the dimmer working the 'old' way. This is a new feature and is not yet documented

Jexkin, 2022-04-06:
So, I just tested it and if I put it above 100 the lights are just on at whatever the value in `Max power in %` is set

If I put the `Dimmer max dim value` to something in between 0 and 100, the dimming works just as usual.
Also I was more so referring to this 
[image]
` _DIMMER_MAXDIM     = 100`, that's why I assumed that the max is 100.

Jexkin, 2022-04-07:
I tested it with pin 33, I hope the settings should be alright. Should be equivalent as if I'd put the `May power in %` to `60`. And so far it works the same.

[image]

but as soon as I put a value above 100 into the `Dimmer max dim value`, it doesn't dim at all.

theyosh, 2022-04-08:
Hi, I made a small change, so do a `git pull` and then it should work with the value of 1024. But I do not get the feeling that it will not fix your issue.

In the name you have LED. Which suggests that you have a LED device. And dimming LED devices is hard. I test with normal light bulbs. Which dim much more easier.

So, if you are trying to dim a led, I would like to ask to google a bit more about the LED device and dimming. For example: #645 

And I would like to close this issue, as the new sensor is working... You can also use #643 for updates.

Jexkin, 2022-04-08:
Sure, I'll post updated in the mentioned issue!
Thanks a lot!

closed 2022-04-08 by theyosh