theyosh/TerrariumPI #675: [FEATURE]: 433 Mhz RF Transmitter/Receiver as a relay

vlreinier, 2022-03-07:
Is there any support for using a 433Mhz RF transmitter – receiver kit (more specifically MX-05V+MX-FS-03V) to control 433 Mhz devices such as Klikaanklikuit (KAKU) products?

I could make this with a script relay and above specified hardware but maybe there is a better way to control KAKU products? Or a suggestion for a new feature?

theyosh, 2022-03-09:
Hi,

I am sorry to say, but this is not supported. As this kit is a general kit. So you need some kind of script to send out the right pulses to toggle KAKU relays. As there could be more then 1.

So, the only solution is to make scripts that toggle a single relay, and use that in TP4.

Different kind of hardware that could be used. But it is way more expensive: http://www.rfxcom.com/epages/78165469.sf/en_GB/?ViewObjectPath=%2FShops%2F78165469

But that needs still scripted relays.

vlreinier, 2022-03-09:
I was thinking of a new relay type with some kind of 'listen' button to retrieve the relay information. But i will create it using a script relay. Thanks.

closed 2022-03-09 by vlreinier