theyosh/TerrariumPI #676: [BUG]: Different decimal output on LCD for multiple temp sensors

Sitnalta, 2022-03-11:
**More info**
Please look first at the [FAQ](https://github.com/theyosh/TerrariumPI/wiki/FAQ) or [Known Issues](https://github.com/theyosh/TerrariumPI/wiki/Known-issues) for more information.
Or search in the [closed tickets](https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed) to see if it should been fixed in the past

**Setup:**
 - TerrariumPI version: 4.2
 - Python version: 3
 - Raspberry PI: 3B+

**Describe the bug**
I have 8 temp sensors (1-wire) connected.
Notificatons on my 20x4 i2c LCD screen sensors show different amount of decimals, I have some with 1 decimal (what I prefure), some have 2 or even 3.

**To Reproduce / Installation steps**
Not relevant I guess.

**Expected behavior**
Same output on screen for all temp sensors, I prefure 1 decimal (Example 26.1)

theyosh, 2022-03-17:
Hi,

Yeah, you are right, so in the 4.x.y.z branch, if is not possible to use number formatting like: `${variable:.2f}` where `:.2f` does a 2 decimal rounding. Of in your case `:.1f`. If you do not use that formatting, you will get about 6 decimals for humidity values.

This code in only on the 4.x.y.z branch now, and will be merged with a new release. You can always try it by switching to the 4.x.y.z branch. But if you are not that familiar with that, you could wait till the next release.

theyosh, 2022-04-01:
Any change for you to test this new number formatting?

theyosh, 2022-04-11:
Closing due to implemented in the latest release (4.3.0) and no response.

closed 2022-04-11 by theyosh