theyosh/TerrariumPI #679: [BUG]: Watchdog issue - following upgrade from 4.1 to 4.2

morrell7, 2022-03-16:
Hi Joshua,

Just upgraded to 4.2 after waiting a while  - but the service isn't able to start:

Debugged and found missing modules: 

sensirion_i2c_sht
sensirion_i2c_driver

went into venv and pip3 installed them and resolved :) I'm getting better!

Just thought you should know. Checked requirements.txt:

pi@pi-terrearium:~/TerrariumPI $ cat requirements.txt | grep sensirion
sensirion-i2c-sht==1.0.0

sensirion_i2c_driver isn't there! but was the first thing my service failed on.

Best regards,

Mark

theyosh, 2022-03-17:
Hmm, are you sure? Because the new Sensiron driver is only available in the 4.x.y.z branch, which is kind of development.

So in the main branch, this driver is not used yet, and should not need the Sensiron driver ( https://github.com/theyosh/TerrariumPI/blob/main/hardware/sensor/sht2x_sensor.py ) . The 4.x.y.z branch does have the Sensiron driver in the requirements: https://github.com/theyosh/TerrariumPI/blob/4.x.y.z/requirements.txt#L50

So I get the felling you have not the latest code if you want to run on the 4.x.y.z branch. Or you should switch back to the main branch, and update that with a git pull.

My suggestion is to go back to the main branch and do a git pull and rerun the installer. That should do the trick

theyosh, 2022-03-17:
Hmm, the sensirion_i2c_driver is indeed missing. Strange, I thought it would be installed with the other Sensiron driver. But this is fixed now.

But still this is in the 4.x.y.z branch, not the main branch

theyosh, 2022-03-28:
Mark,

is this still an issue? The main branch should just work, and the 4.x.y.z also if you rerun the installer after pulling the latest code.

Swiftnesses, 2022-04-01:
@theyosh I just did a git pull and ended up with this issue.

How can I install the missing modules?

P.S. I'm on X.Y.Z

Swiftnesses, 2022-04-01:
Here's error for your info:

`(venv) pi@raspberrypi:~/TerrariumPI$ python terrariumPI.py
2022-04-01 21:50:31,626 - INFO    - terrariumEngine       - Starting TerrariumPI 4.2.0 ...
2022-04-01 21:50:31,710 - INFO    - terrariumEngine       - Loaded 28 settings in 0.08 seconds.
Traceback (most recent call last):
  File "terrariumPI.py", line 16, in <module>
    terrariumEngine = terrariumEngine(__version__)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 113, in __init__
    self.webserver = terrariumWebserver(self)
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 54, in __init__
    self.__routes()
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 236, in __routes
    self.api.routes(self.bottle)
  File "/home/pi/TerrariumPI/terrariumAPI.py", line 146, in routes
    all_sensor_types = '|'.join(terrariumSensor.sensor_types)
  File "/home/pi/TerrariumPI/terrariumUtils.py", line 37, in __get__
    return classmethod(self.fget).__get__(None, owner)()
  File "/home/pi/TerrariumPI/hardware/sensor/__init__.py", line 105, in sensor_types
    for sensor in __cls__.available_sensors:
  File "/home/pi/TerrariumPI/terrariumUtils.py", line 37, in __get__
    return classmethod(self.fget).__get__(None, owner)()
  File "/home/pi/TerrariumPI/hardware/sensor/__init__.py", line 88, in available_sensors
    for (hardware_type, sensor) in __cls__.available_hardware.items():
  File "/home/pi/TerrariumPI/terrariumUtils.py", line 37, in __get__
    return classmethod(self.fget).__get__(None, owner)()
  File "/home/pi/TerrariumPI/hardware/sensor/__init__.py", line 62, in available_hardware
    imported_module = import_module( '.' + file.stem, package='{}'.format(__name__))
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/TerrariumPI/hardware/sensor/shtxx_sensor.py", line 6, in <module>
    from sensirion_i2c_sht.sht2x import Sht2xI2cDevice
ModuleNotFoundError: No module named 'sensirion_i2c_sht'
`

theyosh, 2022-04-01:
first activate python virtual env: 
`cd /home/pi/TerrariumPI/` 
`source venv/bin/activate`

then install the following:

`pip install sensirion_i2c_driver sensirion_i2c_sht`

Then it should work

Swiftnesses, 2022-04-01:
Indeed, I just sorted it. Thanks!

theyosh, 2022-04-01:
But it is still strange... in the 4.x.y.z they should be installed with the installer: https://github.com/theyosh/TerrariumPI/blob/4.x.y.z/requirements.txt#L50-L51=

morrell7, 2022-04-01:
> Mark,
> 
> is this still an issue? The main branch should just work, and the 4.x.y.z also if you rerun the installer after pulling the latest code.

Hi Joshua,

No I fixed it myself, I just raised this to make you aware of it and it seems someone else has had the issue.

It is weird, when I was first installing your software when is was just 4.0.0 I had similar weird things you helped with.
That's why I was able to find the problem again. 

I'm happy for you to close as it doesn't seem too common.

Cheers

theyosh, 2022-04-01:
Then I am for closing. It is possible to search in closed issues as well. So people should then do a bit more research ;)

The problem self should be fixed, that is the main thing to fix. And I think I did, so I close it then now.

Thanks for the response

closed 2022-04-01 by theyosh