theyosh/TerrariumPI #682: [FEATURE]:Is it possible to use a BME680 instead of the BME280?

ewiandr73, 2022-03-31:
Is it possible to use a BME680 instead of the BME280?

I tried to connect one but was getting a temperature reading, not sure if its the way I connected it or if the sensor is configure different and the BME280 setting in Terrarium SW is not picking it up.

I'm using the 4.x.y.z branch on a RPi4 (running really well now with 5 different enclosures).

I'll have another play later and report back if anything changes,

Regards

Andy

theyosh, 2022-03-31:
Hi,

at the moment BME680 is not supported. It is a different chip then the BME280. So I have to add some extra code. I hope the add it this evening, as it looks pretty straight forward.

If you can test it, that would then be nice. Code will be in the 4.x.y.z branch

theyosh, 2022-04-01:
Hi, added support for BME680 sensors. At the moment I do not use the gas value. Not sure how to use that....

Do a `git pull` and rerun the installer `sudo ./install.sh` and then restart TP4. You should then be able to add a BME680 sensor.

All code is untested...

theyosh, 2022-04-01:
Also, I guess this is your first issue, so there is a small fee ;) #210

ewiandr73, 2022-04-02:
I install it and seem to have lost the Gui.

I tried renaming the Terrarium directory and re-installing the TerrariumPI SW, still no gui.
I also tried a new SD card and fresh install and still no gui.

Any ideas?

Regards

Andy

theyosh, 2022-04-02:
I am afraid something strange is going on with the installer. Follow the steps here, and it should work: https://github.com/theyosh/TerrariumPI/issues/679#issuecomment-1086315900

If this solves it, please report back. Then I have to look into the installer...

theyosh, 2022-04-02:
Ok, found the issue. Made some stupid typos in the requirements file.

So could you try to do a `git pull` and rerun the installer: `sudo ./install.sh` and see if it works now? My installer does work correctly now.

ewiandr73, 2022-04-02:
The gui launches after trying the fix from #679 however when I click login I get the following message:

_Error: 401 Unauthorized
Sorry, the requested URL 'http://192.168.86.248:8090/login/' caused an error:

Authenticate to make any changes_

I then tried the git pull and sudo .install.sh and I seem to have lost the gui again.

theyosh, 2022-04-02:
This looks more like a browser issue. Clear browser cache and try again? 

based on your info, I would expect a login window be shown by the browser...

ewiandr73, 2022-04-03:
I tried to clear browser cache, unable to lauch the gui.

I tried a fresh install and again I can't reach the gui.
I tried stopping and start the Terrarium process and get the following:

pi@raspberrypi:~$ sudo service terrariumpi stop

pi@raspberrypi:~$ sudo service terrariumpi start
Job for terrariumpi.service failed because a timeout was exceeded.
See "systemctl status terrariumpi.service" and "journalctl -xe" for details.
pi@raspberrypi:~$ 

pi@raspberrypi:~$ systemctl status terrariumpi.service
â— terrariumpi.service - TerrariumPI
   Loaded: loaded (/etc/systemd/system/terrariumpi.service; enabled; vendor preset: enabled)
   Active: activating (start) since Sun 2022-04-03 11:28:07 AEST; 4s ago
 Main PID: 921 (python)
    Tasks: 1 (limit: 3596)
   CGroup: /system.slice/terrariumpi.service
           â””â”€921 /home/pi/TerrariumPI/venv/bin/python /home/pi/TerrariumPI/terrariumPI.py

Apr 03 11:28:07 raspberrypi systemd[1]: Starting TerrariumPI...
pi@raspberrypi:~$

I even tried a LAN cable and a different RPi4 to make sure it wasn't a network of HW issue (SSH works fine)

theyosh, 2022-04-03:
Strange, could you try: https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/

ewiandr73, 2022-04-03:
I tried on both.

1. The original Pi, seemed to have an issue with the DB, copied the back I had of the DB which seemed to fix it.
**"sqlite3.DatabaseError: database disk image is malformed"**
 I think I have lost my orignal data, which is fine but I still can't login?
[image]
[image]

2. The New install some sort of import error?
**ImportError: numpy.core.multiarray failed to import**

[image]

theyosh, 2022-04-03:
Which Raspian OS is this? I see python 3.9, where I work on 3.7. I guess you have used Bullseye OS, and not the old legacy.

I have not used the new version, as the webcam is not support on the new os... So I am not sure if the installer is working correctly on the new Raspian OS. I had no time to test yet.

ewiandr73, 2022-04-03:
It was the latest OS, which one should i use?

Also I was messing around with my android phone and managed to get the login working ok the original one, which is very odd. I'll set it all up again and test the BME680 and let you know how i get on.

theyosh, 2022-04-03:
Legacy OS: https://theyosh.github.io/TerrariumPI/install/#raspberry-pi

ewiandr73, 2022-04-06:
The BME680 works fine.

I performed a fresh install and reconfigured everything again a few times now, it's a little unstable at the moment and one of my Sonoff relays seems to have decided to die.
I will give it a a few days to settle down.

On another note I flashed a 4 socket smart powerboard (similar to Meross power switches) with Tasmota and connected it in the same way as the Sonoff devices and it seems to work ok,  it's really useful if you have any transformers.  Let me know if you want any logs or anything to verify, I thought you might want to add it  to you supported list.

https://grid-connect.com.au/smart-home-products/powerboards/
 
You can close this case and I'll continue to keep an eye on the install, I will upload some photos once I have tidied up the cabling a bit.

theyosh, 2022-04-06:
Thanks for the update and feedback.

About the Sonoff and Tasmota maybe I should change that in the GUI. But that means also code changes and database changes. Because all Tasmota based devices should be able to use. As long as the urls are the same... 

For now, I think I just update the documentation with this extra information.

theyosh, 2022-04-06:
For now, I close this, as the BME680 is working. Thanks for testing!

closed 2022-04-06 by theyosh