theyosh/TerrariumPI #683: [BUG]:

grahnema, 2022-04-01:
I set up an area to control the humidity in my enclosure but I think I have a two fold issue, the first one may be contributing to the second. One is that when the humidity sensor receives a reading outside the hi or low alarm values the Humidity gauge doesnt show the  current reading and nothing is displayed but I can see the error warning pop up that tells me the humidity is outside the range which also changes the the current conditions in the area overview to show a reading of 0%(0%-0%) which I believe is causing the Second issue which is that when the humidity reaches the high alarm the fan relay I have set up does not turn on to reduce humidity.  

I can manually turn the fan relay on and when the humidity level falls between the high and low the readings and gauge show properly again as well as the area view. but then the problem repeats itself.  
[image]

[image]
Here you can see where Humidity Control Area and Misting System the readings all move to 0%
[image]
Fresh Air is the name of the fan relay which should turn on when I reach the high alarm 

[image]
In this photo you can see when the humidity fall within range everything shows properly.

theyosh, 2022-04-01:
You have the **limit min** and **max** values to close to the alarm values. Put you alarm max to 100 and alarm min to 30 or lower.

It is correct that when the sensor value is higher then the limit max value, I see that as an error. And after a while the sensor goes to 0% due to no data. So in this case, increase the limit max value to 100%.

theyosh, 2022-04-01:
More info: https://theyosh.github.io/TerrariumPI/setup/#sensors

grahnema, 2022-04-02:
Outstanding! issue resolved. After your comment here and reading the instructions again it makes sense.  Thank you!

closed 2022-04-02 by grahnema