theyosh/TerrariumPI #685: [BUG]: Updated to 4.2.0 and cannot start UI

Swiftnesses, 2022-04-01:
Hi @theyosh,

Following my git pull, I'm getting the error:

`Traceback (most recent call last):
  File "terrariumPI.py", line 16, in <module>
    terrariumEngine = terrariumEngine(__version__)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 145, in __init__
    self.__load_existing_webcams()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 968, in __load_existing_webcams
    value = self.webcams[webcam.id].update(relays)
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/decorator.py", line 232, in fun
    return caller(func, *(extras + args), **kw)
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/retry/api.py", line 74, in retry_decorator
    logger)
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/retry/api.py", line 33, in __retry_internal
    return f()
  File "/home/pi/TerrariumPI/hardware/webcam/__init__.py", line 466, in update
    self.__raw_image = Image.open(image)
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/PIL/Image.py", line 3009, in open
    "cannot identify image file %r" % (filename if filename else fp)
PIL.UnidentifiedImageError: cannot identify image file <_io.BytesIO object at 0x6a368e70>
`

I can no longer access the UI. Any ideas?

Swiftnesses, 2022-04-01:
I deleted the db and restored my settings as I need to open it remotely tomorrow. All sorted.

closed 2022-04-01 by Swiftnesses

theyosh, 2022-04-01:
Hmm, for the next time, just move the DB to a different location. So we can try to make the changes in the DB so that you do not have to reconfigure everything.

In this case it looks like issues with a webcam. We could just only delete the webcams in the DB so that the rest is still there..

But that is for a next time :)