theyosh/TerrariumPI #686: [Guidance]: Area (Heating) Ignore Low Alarm

ewiandr73, 2022-04-06:
Setup
 - TerrariumPI version: 4.2.0
 - Python version: 3
 - Raspberry PI:  4 Model B Rev 1.1 

Advise needed
After re-installing the Terrarium SW, I went through setting up the enclosures, when looking at the heating area there is an option to 'Ignore low alarm' in the low alarm tab.

The only option it will let me select is to select the Ignore low alarm option, I can find and info on what this option does ? 

Screenshot
[image]

ewiandr73, 2022-04-06:
Also seem looks the the system is trying to stay at the higher temperature setup and not dropping to the lower setting before turning the heater on.

[image]
 
I'm sure I've probably set something wrong but I can't seem to find what

theyosh, 2022-04-06:
The first screen is an error which is just fixed. So for that do a `git pull` and rerun the installer `sudo ./install.sh`. As there are new libraries in use.

The option 'ignore low alarm' when toggled on, will do what is shown in the second image. It will take the high alarm value as reference. When you have updated the TP4 code, you can toggle this out, and then the temperature will stay withing the low and high alarm value.

The idea is that you can have multiple areas with the same sensors, but with different behavior on when the toggle on. This is just a new feature that I made last 2 weeks.

ewiandr73, 2022-04-07:
That fixed it, thanks :)

closed 2022-04-07 by ewiandr73