theyosh/TerrariumPI #687: [BUG]: couldn't find remote ref refs/heads/master

Hemibon, 2022-04-10:
Hi there!

We've been trying to reconnect to our PI website but it hasn't been working. We can connect to the PI via ssh, see attachment. From there we wanted to update with git pull, but we get an error message 'fatal couldn't find remote ref refs/heads/master'. What does that mean? How can we solve the issue? Version of the TerrariumPI is in the screenshot.

Thanks a lot for your help!
Best wishes,
Hemibon

[image]

theyosh, 2022-04-10:
Hi, it is impossible to upgrade from a 3.X to a 4.X. See: https://github.com/theyosh/TerrariumPI/releases/tag/4.0.0 and https://theyosh.github.io/TerrariumPI/install/#migration-from-v3-to-v4.

So basically, running a 4.X version needs a reinstall of my software. You can use the same Raspberry PI. Easiest way is to move the current folder `TerrariumPI` to `TerrariumPI.old` and do a fresh install with the 4.X version.

Also you are on the master branch, which is deleted on Github. That is why you have the git error.

Hemibon, 2022-04-10:
Hi, thanks for the quick answer. I will go to v4. The problem is that I can't even stop it terariumpi, see attachement...

[image]

theyosh, 2022-04-10:
I think you need the command `screen -r` and then press Ctrl+C twice: https://github.com/theyosh/TerrariumPI/wiki/FAQ#screen-1=

Then it should stop. Then move the folder TerrariumPI and start a new installation https://theyosh.github.io/TerrariumPI/install/#manual

Hemibon, 2022-04-10:
Yeah, it's not liking it... I still can't stop after the screen is terminated...?

[image]

Hemibon, 2022-04-10:
Never mind.. I went ahead with the moving to TerrariumPi.old and it worked and started the reinstallation but it gave a lot of warnings... I am going to start from scratch, from the beginning (Balena etcher and so forth..). We'll see and I'll close this subject. Thank you anyway

closed 2022-04-10 by Hemibon

theyosh, 2022-04-10:
Hmm, to bad. That will take some more time. But, make sure you use the Legacy OS of Raspberry PI. https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-legacy

Look for:

Raspberry Pi OS Lite (Legacy)

    Release date: April 4th 2022
    System: 32-bit
    Kernel version: 5.10
    Debian version: 10 (buster)
    Size: 284MB