theyosh/TerrariumPI #688: failed to start component /etc/update-motd.d/05terrariumpi

Smashedcrab83, 2022-04-17:
I have tryed to install with differant OS but nothing seems to work, 
Keeps coming up with "failed to start component /etc/update-motd.d/05terrariumpi"

Any help would be good .
Thanks

theyosh, 2022-04-18:
Hmm, this is strange. When do you see this error? When you login with SSH?

Then it looks like the software is not installed correctly. A few things you could do:

1. Rerun the installer once more.
2. Run in debug mode: https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/
3. Running on Debian Buster 32bit Lite legacy should work https://theyosh.github.io/TerrariumPI/install/#raspberry-pi

theyosh, 2022-05-01:
Any update on this? I cannot replicate this issue.

theyosh, 2022-05-16:
Closing due to no repsonse

closed 2022-05-16 by theyosh