theyosh/TerrariumPI #692: [BUG]:  docker PCA9685-dimmer error

cmonicob, 2022-04-24:
**Setup:**
 - TerrariumPI version:  4
 - Python version: 3
 - Raspberry PI: 3B+

**Describe the bug**
Could not load hardware for relay PCA9685-dimmer dimmer named 'light' at address '1,0x40': 121

**To Reproduce / Installation steps**
Docker compose

theyosh, 2022-04-25:
The error message is a bit cryptic. I guess there should also be some like `OS error`. Because the number 121 looks like the I2C hardware is not found/connected...

But I do not own this hardware, so I cannot test this. So I need more info, to see what could be wrong in the code... Could you try to see if there is more info about the error?

cmonicob, 2022-04-30:
all the sensors and actuators with which I work are through i2c, the only one that gives the error is the PCA9685, when doing an i2c scan it appears as connected and when sending actions directly over the bus it also works correctly.
It only happens on docker since if I put the native code it works correctly.
I have been looking for the cause of the error and with docker it is difficult for me to be able to debug correctly.
If I can launch the tests in some specific way, please tell me the command to be able to find the error

theyosh, 2022-04-30:
Hmm, I have no clue. So to be clear. You made some small test code, and that works directly outside docker. The same code does not work in Docker? And other I2C does work both in and outside docker?

Then I do not know... I do not have the hardware to test. And I do not have special tests for it. What I basically do is make a small python script to test the device with some basic code/example code. And then I integrate it with my code. And hope it still works ;) So that is how far I can help you with this.

This is very strange... and very annoying... as other I2C does work...

theyosh, 2022-05-05:
So, this is now fixed right? With your PR? Can we close this issue then?

cmonicob, 2022-05-05:
bug fixed

closed 2022-05-05 by cmonicob