theyosh/TerrariumPI #693: SD card corrupted - can I get the settings from 3.99 to work on 4.*

martynwheeler, 2022-04-30:
My SD card won't boot - it was working on 3.99 version of the software.  I can access the filesystem to get the settings files.  Is there are simple way to restore these to a new system that is working on version 4?

I would appreciate any guidance to get back to my old system.  Thank you very much.
Martyn

theyosh, 2022-04-30:
Hmm, there is no upgrade route to version 4. So in you case I would install an old version: https://github.com/theyosh/TerrariumPI/tree/3.x.y.z#installation=

1. Install Raspian Buster (Legacy) Lite on a new SD card.
2. Install Git: `sudo apt install git`
3. Install TerrariumPI 3 (old) version: ` git clone --recursive --branch 3.x.y.z --depth 1 https://github.com/theyosh/TerrariumPI.git`
4. Run the installer for Python3: `sudo ./install.sh 3`. Do **not** reboot yet!!

Then you could restore the following files: https://github.com/theyosh/TerrariumPI/wiki/FAQ#which-files-to-backup=

And reboot the Pi. That should give you the old version back.

An upgrade to version 4 is a different job. So I would again use a new SD card for that, so you can easily swap between old and new version. More info about version 4: https://github.com/theyosh/TerrariumPI/releases/tag/4.0.0

But I would suggest to get it working with TP3 first....

martynwheeler, 2022-05-01:
I followed the instructions exactly - but when I reboot it claims to have started terrariumpi during the boot messages but it is not accessible.  It is not giving me the motd screen when I log in (in fact there is an error finding motd..!)
After step 4 I did not reboot and copied the old files across (from my old 3.9.9 system).

I'm a bit desperate as I am having to manually control everything from a single power switch.

theyosh, 2022-05-01:
If you are logged in, and enter the folder TerrariumPI you should be able to start it manually so you get the debug/error on screen:

`python3 terrariumPI.py`

This should start it, and if there is an error, it will show. I need error messages, else I cannot provide solutions.

martynwheeler, 2022-05-01:
Thanks, this is what I get after running that:
Traceback (most recent call last):
  File "terrariumPI.py", line 24, in <module>
    from terrariumEngine import terrariumEngine
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 31, in <module>
    from terrariumSensor import terrariumSensor
  File "/home/pi/TerrariumPI/terrariumSensor.py", line 516, in <module>
    from terrariumGPIOSensor import terrariumYTXXSensorDigital, terrariumDHT11Sensor, terrariumDHT22Sensor, terrariumAM2302Sensor, terrariumHCSR04Sensor
  File "/home/pi/TerrariumPI/terrariumGPIOSensor.py", line 6, in <module>
    import Adafruit_DHT
ModuleNotFoundError: No module named 'Adafruit_DHT'

martynwheeler, 2022-05-01:
It is also saying:

failed to stat component /etc/update-motd.d/05-terrariumpi: no such file or directory

Thanks again for your help.

theyosh, 2022-05-01:
Could you try the following command: `sudo pip3 install Adafruit_DHT` and then try to start again `python3 terrariumPI.py`

martynwheeler, 2022-05-01:
Thanks, now I get a new error:

Traceback (most recent call last):
  File "terrariumPI.py", line 24, in <module>
    from terrariumEngine import terrariumEngine
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 32, in <module>
    from terrariumSwitch import terrariumPowerSwitch
  File "/home/pi/TerrariumPI/terrariumSwitch.py", line 14, in <module>
    import brightpi
ModuleNotFoundError: No module named 'brightpi'

theyosh, 2022-05-01:
Hmm, it looks like some submodules are not installed correctly. Could you go the the folder 'Bright-Pi' and run the command: `sudo python3 setup.py install`

And restart again. Somehow the installer should install all those libraries, but it did not somehow..

martynwheeler, 2022-05-01:
Okay, i run that and get this:
pi@terrariumpi:~/TerrariumPI/Bright-Pi $ python3 setup.py install
running install
/usr/local/lib/python3.7/dist-packages/setuptools/command/install.py:37: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  setuptools.SetuptoolsDeprecationWarning,
/usr/local/lib/python3.7/dist-packages/setuptools/command/easy_install.py:147: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  EasyInstallDeprecationWarning,
error: can't create or remove files in install directory

The following error occurred while trying to add or remove files in the
installation directory:

    [Errno 13] Permission denied: '/usr/lib/python3.7/site-packages/test-easy-install-971.write-test'

The installation directory you specified (via --install-dir, --prefix, or
the distutils default setting) was:

    /usr/lib/python3.7/site-packages/

Perhaps your account does not have write access to this directory?  If the
installation directory is a system-owned directory, you may need to sign in
as the administrator or "root" account.  If you do not have administrative
access to this machine, you may wish to choose a different installation
directory, preferably one that is listed in your PYTHONPATH environment
variable.

For information on other options, you may wish to consult the
documentation at:

  https://setuptools.pypa.io/en/latest/deprecated/easy_install.html

Please make the appropriate changes for your system and try again.

martynwheeler, 2022-05-01:
so I now tried using sudo

pi@terrariumpi:~/TerrariumPI/Bright-Pi $ sudo python3 setup.py install
running install
/usr/local/lib/python3.7/dist-packages/setuptools/command/install.py:37: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  setuptools.SetuptoolsDeprecationWarning,
/usr/local/lib/python3.7/dist-packages/setuptools/command/easy_install.py:147: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  EasyInstallDeprecationWarning,
running bdist_egg
running egg_info
writing brightpi.egg-info/PKG-INFO
writing dependency_links to brightpi.egg-info/dependency_links.txt
writing top-level names to brightpi.egg-info/top_level.txt
reading manifest file 'brightpi.egg-info/SOURCES.txt'
writing manifest file 'brightpi.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
creating build/bdist.linux-armv7l/egg
creating build/bdist.linux-armv7l/egg/brightpi
copying build/lib/brightpi/__init__.py -> build/bdist.linux-armv7l/egg/brightpi
copying build/lib/brightpi/brightpilib.py -> build/bdist.linux-armv7l/egg/brightpi
byte-compiling build/bdist.linux-armv7l/egg/brightpi/__init__.py to __init__.cpython-37.pyc
byte-compiling build/bdist.linux-armv7l/egg/brightpi/brightpilib.py to brightpilib.cpython-37.pyc
creating build/bdist.linux-armv7l/egg/EGG-INFO
installing scripts to build/bdist.linux-armv7l/egg/EGG-INFO/scripts
running install_scripts
running build_scripts
creating build/bdist.linux-armv7l/egg/EGG-INFO/scripts
copying build/scripts-3.7/brightpi-test.py -> build/bdist.linux-armv7l/egg/EGG-INFO/scripts
changing mode of build/bdist.linux-armv7l/egg/EGG-INFO/scripts/brightpi-test.py to 755
copying brightpi.egg-info/PKG-INFO -> build/bdist.linux-armv7l/egg/EGG-INFO
copying brightpi.egg-info/SOURCES.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying brightpi.egg-info/dependency_links.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying brightpi.egg-info/top_level.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
zip_safe flag not set; analyzing archive contents...
creating 'dist/brightpi-2.0-py3.7.egg' and adding 'build/bdist.linux-armv7l/egg' to it
removing 'build/bdist.linux-armv7l/egg' (and everything under it)
Processing brightpi-2.0-py3.7.egg
Removing /usr/lib/python3.7/site-packages/brightpi-2.0-py3.7.egg
Copying brightpi-2.0-py3.7.egg to /usr/lib/python3.7/site-packages
brightpi 2.0 is already the active version in easy-install.pth
Installing brightpi-test.py script to /usr/bin

Installed /usr/lib/python3.7/site-packages/brightpi-2.0-py3.7.egg
Processing dependencies for brightpi==2.0
Finished processing dependencies for brightpi==2.0

martynwheeler, 2022-05-01:
but it still gives the same error:

pi@terrariumpi:~/TerrariumPI $ python3 terrariumPI.py
Traceback (most recent call last):
  File "terrariumPI.py", line 24, in <module>
    from terrariumEngine import terrariumEngine
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 32, in <module>
    from terrariumSwitch import terrariumPowerSwitch
  File "/home/pi/TerrariumPI/terrariumSwitch.py", line 14, in <module>
    import brightpi
ModuleNotFoundError: No module named 'brightpi'

theyosh, 2022-05-01:
Last one is ok. The command needs sudo.

So, what happens now when you start again: `python3 terrariumPI.py`

theyosh, 2022-05-01:
Hmm, then it is hard to figure this out. Another option can be by editing the file `terrariumSwitch.py`

I do not know how much experience you have on the console. But you could use nano for editing:
`nano terrariumSwitch.py` and then find the line `import brightpi` and add a `#` in front of that line. Save the file. This will disable Bright Pi support. If you do not use it, it should be save to disable it.

Then try again: `python3 terrariumPI.py`

martynwheeler, 2022-05-01:
still the same error: 
pi@terrariumpi:~/TerrariumPI $ python3 terrariumPI.py
Traceback (most recent call last):
  File "terrariumPI.py", line 24, in <module>
    from terrariumEngine import terrariumEngine
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 32, in <module>
    from terrariumSwitch import terrariumPowerSwitch
  File "/home/pi/TerrariumPI/terrariumSwitch.py", line 14, in <module>
    import brightpi
ModuleNotFoundError: No module named 'brightpi'

I have also tried rebooting

martynwheeler, 2022-05-01:
Thanks,  next error is:

pi@terrariumpi:~/TerrariumPI $ python3 terrariumPI.py
Traceback (most recent call last):
  File "terrariumPI.py", line 24, in <module>
    from terrariumEngine import terrariumEngine
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 32, in <module>
    from terrariumSwitch import terrariumPowerSwitch
  File "/home/pi/TerrariumPI/terrariumSwitch.py", line 31, in <module>
    from lib8relay import set as relay8SetV3
ModuleNotFoundError: No module named 'lib8relay'

martynwheeler, 2022-05-01:
okay have fixed that by installing the module:  it is now up and running.  I have no idea why it did not inastall all of the those modules

theyosh, 2022-05-01:
Ok, now it works again... Strange...

So does it run with your old settings and database? So it should now be running like you had before?

Then I would suggest to take a brake ;) and then next week start an upgrade? As I do not support TP3 anymore, I cannot guarantee that you can keep reinstalling TP3.

martynwheeler, 2022-05-01:
It seems to be up and running so fingers crossed.

I will look into the upgrade soon.  It should be straighforward as I only have:
A single light
A single heater on a dimmer (MPDMv3)
two ds18b20 sensors