theyosh/TerrariumPI #695: ds18b20 temperature sensors disappeared?

martynwheeler, 2022-05-01:
I have just installed the latest docker image and I see the ds18b20 sensors - but after about five minutes they have disappeared and now i cannot access them (rebooted the pi).

I am new to docker, is it better to do a manual installation for these sensors?

Thanks

Martyn

theyosh, 2022-05-01:
To be honest, I would only use Docker is you are familiar with it. I am also not a big Docker user, so I also relay on other people.

So, I think you should not use it for now. And about the missing sensors, that is a bit strange. There should be folders in `/sys/bus/w1/devices/` staring with the number 28

`ls -l /sys/bus/w1/devices/`

`lrwxrwxrwx 1 root root 0 May  1 19:46 28-0115b231f3ff -> ../../../devices/w1_bus_master1/28-0115b231f3ff
lrwxrwxrwx 1 root root 0 May  1 19:46 28-0315b2966fff -> ../../../devices/w1_bus_master1/28-0315b2966fff`

If that is not the case, I can't fix it, Then the RPI has problems reading the sensor. Sometimes it could help to power it down, and unplug the PI. So there is a totally power drain.

martynwheeler, 2022-05-01:
Okay, I'll revert to manual installation - I know what I'm doing and can
find my way around.

It all works ok if I switch sd card back to 3.10 so it must be something in
the docker image rather than a hardware issue.

Martyn

martynwheeler, 2022-05-02:
After a manual installation all is good.  Thanks for all your hard work on this!

closed 2022-05-02 by martynwheeler