theyosh/TerrariumPI #697: What does the time mean in the heater area?

martynwheeler, 2022-05-02:
I have just upgraded to 4.1.3 and it all looks good.  I think I have configured everything.  I was just wondering if you could explain what the time for the heater area means in the image below?
[image]

Here are the area settings
[image]

Thanks again for all your hard work

theyosh, 2022-05-04:
Hi,

Hmm not sure about the total duration of the heating. That looks wrongly calculated.

You are screenshot shows that you are using sensors as the trigger. And you have a dimmer, so that means that once the dimmer is powered on, it will try to keep the temperature in the middle of the min and max alarm value.

The Power on time is now '0' which is ignored. If you fill in a value there, the relay should toggle on when alarm low is active, and then shutdown after this amount of seconds. It is mostly used for sprayers. Here you can specify the time the relays are on when needed.

The Settle time is the time it takes to toggle the relays back on when they are shutdown. Again, this is more for spraying. This will give the sensors and software to get the new values before the alarm goes off a second time.
But you have a dimmer relay selected, so in this case, the dimmer will only update once 1 minute. Normally that will be 30 seconds. The lower is better. So I would say, set the Settle timeout to '0' to disable/ignore that also. Then you dimmer will update every 30 seconds based on the measured temperature.

I still have to make documentation about setting up area's. But it is a lot of work. Also, click on the question mark icon next to the modal title to get some more information.

martynwheeler, 2022-05-04:
thank you for your response.  That all makes sense.  Not sure why the calculated time is wrong - I was fiddling with the settings at the beginning so it is probably left over from there.

One last thing on this topic.  What does the "depends on" do - it says "Select the areas where this area depends on." but I am not clear what this means exactly?

Thanks again for all your hard work.

theyosh, 2022-05-05:
The depends on field lets you select other areas where this area depends on. This is mostly for spraying system with a water tank area. If the water level is to low, the water tank area will get an error. Then the humidity area where the sprayer is connected to, will not turn on, because the water tank is in alarm state.  This will prevent from dry pumping.

The same goes for heating the water tank. If there is not enough water, you do not want to turn on the heater in the water tank. So you create then an area for the heating, that depends on the water tank. Again. low level water, will trigger an alarm, and therefore stop or not start the heating.

And maybe there are more scenarios possible. I hope it is clear.

martynwheeler, 2022-05-05:
Brilliant, thanks for the info.

theyosh, 2022-05-05:
Ok, then I close it now.

closed 2022-05-05 by theyosh