theyosh/TerrariumPI #698: MPDMv4.1 - flicker on 0%

martynwheeler, 2022-05-02:
Hi Again,

Now that I am on version 4.3 my heater bulb is flickering at night.  The settings have the bulb on 0%.  It is using pin 12 as its control.  What are the recommended calibration settings please?

Thanks

theyosh, 2022-05-05:
I guess you did not change the hardware? So old version also used pin 12.

So I guess there are two solutions. You could use the Dimmer offset calibration and use the value '1'. So when you put the dimmer at 0% it will actually be at 1%. Not sure if this will work, but you could try this. If the light is off then, this is an easy fix.

Else you can use the field `Dimmer max dim` and enter the value of 870 and use a `Dimmer frequency` value of 5000. That should then trigger the old dimmer logic, which could also help. But this is not fully tested, and it depends on the light bulb it seems. As I do not have any issues with the dimmer.

More info at #643

martynwheeler, 2022-05-06:
Okay, I have used  the field Dimmer max dim and enter the value of 870 and use a Dimmer frequency value of 5000 - that seems to have made it work with no flicker.

Would you mind explaining the changes between the old and new?  Is hardware PWM not as good as software?

Thanks again

theyosh, 2022-05-07:
In the past it was always try and error with the dimmer values like frequency and max dim. So in the new version I was using a more generic dimmer library, which also supports software PWM. And therefore you could use all GPIO pins for dimming.
So that looked nice. And for me it worked.

But as you can see in the [closed tickets](https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed+dimmer) you can see there is a lot of differences what works.

And you also confirm that the old way works better for you. For me, the new way also works. Why... I do not know. It could be the type of light bulb. Could be the AC power... And I am not a electronic technician..

theyosh, 2022-05-07:
Also, hardware PWM works way better, but limits in the GPIO pins you can use... So that is why I tried a new library for PWM dimming

martynwheeler, 2022-05-07:
Okay, that is great. Thanks for explaining. I will close the issue.

closed 2022-05-08 by theyosh

martynwheeler, 2022-10-11:
Thanks for the response.

Is software PWM better?  I thought that the hardware option was more
stable.  Do you recommend using a different pin from 12?  I have tried a
few options and have had it working if I lower the frequency down to
1000Hz  - I will look at your suggestions later and report back

Thanks again