theyosh/TerrariumPI #700: Widgets not showing on old Ipad

martynwheeler, 2022-05-06:
I was hoping to use an old ipad (ios 9.3.5) as permanent screen for viewing terrarium pi - however, the pages are not rendering properly (dials and graphs not loading) - I can't install chrome or firefox so im stuck with safari.  Anyone got any ideas?

Thanks

Martyn

theyosh, 2022-05-06:
I am afraid that I cannot fix that. I looked in the code, but some external libraries uses code that needs iOS 10.X at least.

So that means a new device. I can get an Ipad mini 2 for about 60 euro. Which has iOS 12.x so that should work.

But I cannot fix this.

martynwheeler, 2022-05-06:
No worries. I was just trying to find a use for an old ipad. Issue is closed

closed 2022-05-06 by martynwheeler