theyosh/TerrariumPI #701: Gradually dim heater when light goes off

martynwheeler, 2022-05-06:
Is it possible to gradually dim the heater at night time after the main lights go off? Thank you

theyosh, 2022-05-07:
You mean the difference between day and night? Nope, not at the moment. That is an instant change.

I could make it more dynamic, but that means that you have to enter a duration in which the temperature drops, but also rises. It goes two directions. So that would mean 2 extra fields to answer....

This is how I see it then. Or did you have a different idea?

martynwheeler, 2022-05-07:
Yes, that is how I saw it. Some kind of fade in/fade out time so.
What I have done for now is to delay the light and let the heater bulb come
on for 15 minutes before. This is okay as it feels less abrupt.

theyosh, 2022-10-23:
Marty,

sorry for the long delay, but I am afraid that this is an option I will not implement. This is due to the fact a lot of more logic is needed in order to make this. And that has to do with restarts during a dimmer action.

And you found a work around and if that is still working, I would like to close this issue.

theyosh, 2022-12-02:
Closing due to not being implemented

closed 2022-12-02 by theyosh