theyosh/TerrariumPI #705: [BUG]:

Dragonfly-terra, 2022-05-31:
**Setup:**
 - TerrariumPI version: 4
 - Python version: 3
 - Raspberry PI: 4

**Describe the bug**
When adding a BME280 sensor on the i2c bus adres 3 [3x76] it gives the error:
`Unable to load sensor BME280 sensor temperature named 'Tempsensor 2' at address '3x76': invalid literal for int() with base 16: '0x3x76'`

**To Reproduce / Installation steps**
I have changed the config file in the boot folder, the i2c settings are as followed:
`
# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

dtoverlay=i2c-gpio,bus=5,i2c_gpio_delay_us=1,i2c_gpio_sda=5,i2c_gpio_scl=6
dtoverlay=i2c-gpio,bus=4,i2c_gpio_delay_us=1,i2c_gpio_sda=23,i2c_gpio_scl=24
dtoverlay=i2c-gpio,bus=3,i2c_gpio_delay_us=1,i2c_gpio_sda=27,i2c_gpio_scl=22
`

**Expected behavior**
It shows the sensor when use the command  `sudo** i2cdetect -y 3`
The sensor that is connected to bus adres 0x76 is working fine

**Screenshots**
[image]

theyosh, 2022-05-31:
Hi,

using bus 3 needs this address: `0x76,3` use `,3` for i2c bus https://theyosh.github.io/TerrariumPI/hardware/#i2c-bus

Dragonfly-terra, 2022-06-01:
Hi,

thx for your reply. I tried that but for some reason it doesn't work.

[image]

The i2c files are in the /dev/ folder:
[image]

I will test a different sensor.

Quick question I am using different GPIO number then where mentioned in the tutorial, don't know if that is a problem.
In the tutorial it says GPIO 17 will be the SDA and GPIO 27 will be the SCL for i2c bus 3. 
I used SDA = GPIO27 and SCL = GPIO22

Dragonfly-terra, 2022-06-01:
And now it is working after a few minutes of waiting.
[image]

Don't know why it wasn't working. I did not touch it after my reply.

Dragonfly-terra, 2022-06-01:
I am currently trying to test with a multiplexer TCA9548A to check if that works.

I have edited the config file in the folder /boot/
`# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on
dtoverlay=i2c-mux,pca9548=0x70`

But i got a lot of errors, they are showing some values but it doesn't seem to work as it should.
Am I to ambities and it isn't supported or am I missing something?

theyosh, 2022-06-02:
I have not tested such a device. But using `dtoverlay=i2c-mux,pca9548=0x70` should give 8 new I2C buses according to: https://forums.raspberrypi.com/viewtopic.php?t=141517

But you have a '#' in front of that line, which means that it is a comment and not used. So I expect that you tried to add: `dtoverlay=i2c-mux,pca9548=0x70`. Make sure the address of 0x70 is correct for your device.

I am not sure what will happen with the existing I2C bus where this device is connected to.

So it could be that adding this device, will overrule the existing I2C bus and all connected devices. That could give errors when you run my software as the devices are not found anymore.

So, first action is to see if you can enable the TCA9548A, that you will get 8 I2C buses. And then you probably should reconnect al your hardware to the TCA9548A device. As I expect it will override the existing I2C bus. use `i2cdetect` to check if all hardware is available on the I2C buses.

theyosh, 2022-06-17:
Do you have an update? As it looks like loading the mutliplexer through RPI boot config seems the most easy solution.

Dragonfly-terra, 2022-06-17:
Sorry for my late reply.
I still had some issues but I suspected the hardware wiring. So I am currently busy with complete new E-cabinet.
Hope to finish it this weekend and then I will report back to you.
[image]

Dragonfly-terra, 2022-06-19:
[image]

theyosh, 2022-06-19:
Nice!

Could you do me a small favor, and explain how you did it? Did you use the https://forums.raspberrypi.com/viewtopic.php?t=141517 and do you have now also 8 I2C buses? So basically, add the 

Then I can add some extra documentation how to use this.

Dragonfly-terra, 2022-06-19:
Here is the description what I did.

I2C Multiplexer - TCA9548A

Wiring diagram:

| Raspberry 4b  | I2C Multiplexer |
| ------------- | --------------- |
| 5Vdc [pin 02] | VIN [pin 01]    |
| GND [pin 06]  | GND [pin 03]    |
| SDA [pin 03]  | SDA [pin 05]    |
| SCL [pin 05]  | SCL [pin 07]    |

If you use more than one multiplexer you need to change the adres

| A0 [pin 11] | A1 [pin 13] | A2 [pin 15] | Mux Adress |
| ----------- | ----------- | ----------- | ---------- |
| 0           | 0           | 0           | 0x70       |
| 1           | 0           | 0           | 0x71       |
| 0           | 1           | 0           | 0x72       |
| 1           | 1           | 0           | 0x73       |
| 0           | 0           | 1           | 0x74       |
| 1           | 0           | 1           | 0x75       |
| 0           | 1           | 1           | 0x76       |
| 1           | 1           | 1           | 0x77       |

Next step is to go change the config.txt file.
This file is located in the boot folder and can only be changed when in root access.

Open the connection with putty. 
Use the following command:

<code> cd /boot/ </code>  
<code> sudo nano config.txt </code>  

It should open the file, go down to find the following line:  
<code> # Uncomment some or all of these to enable the optional hardware interfaces </code>

Make sure the following option is on: (if it is not there add it)  
<code> dtparam=i2c_arm=on </code>

Add the following line after it:  
<code> dtoverlay=i2c-mux,pca9548=0x70 </code>

Explanation:  
<code>
Name:   i2c-mux  
Info:   Adds support for a number of I2C bus multiplexers on i2c_arm  
Load:   dtoverlay=i2c-mux,<param>=<val>  
Param:  pca9548     Select the NXP PCA9548 device  
Val:    addr        Change I2C address of the device (default 0x70)  
</code>

press ctrl - x and save the file.

Reboot the raspberry

To check if the Multiplexer is working correctly open the connection with putty and type the following instruction:

<code>sudo i2cdetect -l</code>

You should get something like this: (address could differ from example)  
```
i2c-25  i2c     i2c-1-mux (chan_id 3)       I2C adapter
i2c-23  i2c     i2c-1-mux (chan_id 1)       I2C adapter
i2c-1   i2c     bcm2835 (i2c@7e804000)      I2C adapter
i2c-28  i2c     i2c-1-mux (chan_id 6)       I2C adapter
i2c-26  i2c     i2c-1-mux (chan_id 4)       I2C adapter
i2c-24  i2c     i2c-1-mux (chan_id 2)       I2C adapter
i2c-22  i2c     i2c-1-mux (chan_id 0)       I2C adapter
i2c-29  i2c     i2c-1-mux (chan_id 7)       I2C adapter
i2c-27  i2c     i2c-1-mux (chan_id 5)       I2C adapter
```

i2c-1 is the I2C adapter from the raspberry

i2c-22 till i2c-29 are the i2c adapters from the Multiplexer

If it is not appearing wright away you could do the following:  
* Reset the I2C Multiplexer by putting 5Vdc on the RST [pin 09]
* Check if the SDA en SCL are correctly wired
* Check if the connection is good, breadboard could give connection issues
* Check folder /dev/ if files with i2c-[number] are present
* Shutdown raspberry, completely disconnect power-supply, restart and check again.

Now you can use multiple sensors with the same i2c-address.
For example I used 4 GYBMEP (BME/BMP280 i2c) sensor without changing the address.
The default address of the sensor is 0x76.
I used the following pinning:  
| Sensor   | Pinning  | address |
| -------- | -------- | ------- |
| sensor 1 | SD0, SC0 | 0x76,22 |
| sensor 2 | SD0, SC0 | 0x76,23 |
| sensor 3 | SD0, SC0 | 0x76,24 |
| sensor 4 | SD0, SC0 | 0x76,25 |

If you use the following command in putty you should the address:  
<code>sudo i2cdetect -y 22</code>  
**The number 22 is a example, change it to the number of the address you got from the i2c scan you did above**

It should look something like this:  
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f  
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --  
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --  
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: UU -- -- -- -- -- 76 --
```

Where UU is the Multiplexer and 76 is the sensor (of course if you have more i2c devices on the same port you should have more address showing)

The next step is using the sensors in TerrariumPI.  
Go to **Sensors** and press **New sensor**  
For the example we used the BMP/BME280 i2c sensor so select in hardware BME280 Sensor.  
For now we use the temperature for the type.  
Now the address of the first sensor is <code>0x76,22</code>  
**The number 22 is a example, change it to the number of the address you got from the i2c scan you did above**  

Fill in a name, the alarm, limit and Max diff and save.
It should look something like this:  
[image]

Now you could add the next sensors by using the different addresses like <code>0x76,23</code> and <code>0x76,24</code>  and <code>0x76,25</code>

Sources  
[I2C Multiplexer - TCA9548A Info](https://www.instructables.com/TCA9548A-I2C-Multiplexer-Module-With-Arduino-and-N/)  
[Raspberry config edit](https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README)  
[Hardware info on TerraiumPi](https://theyosh.github.io/TerrariumPI/hardware/)

theyosh, 2022-06-19:
Thanks a lot!!!

theyosh, 2022-06-26:
Closing now. I added extra documentation with a link to your explanation: https://theyosh.github.io/TerrariumPI/hardware/i2c-multiplexer/ Thanks a lot for testing and describing it.

closed 2022-06-26 by theyosh