theyosh/TerrariumPI #707: Problem with switch humidifier

vikrilow, 2022-06-04:
Hello
I have Terrarium PI 3.10.0 installed since 1 year ago. It has control over temperature and humidity, mainly, in addition to light control.

These sensors cause you to turn the humidifier, ceramic bulb, and fans on or off.
I am observing that the connector where I have the humidifier does NOT disconnect the humidifier, it keeps leaving it on. Two days ago I realized this and changed the switch, thinking that it could be from the connector itself (it had a spare plug, I did tests and it worked correctly).
Today I realize that this plug acts in the same way and I don't know if it could be due to some configuration issue.

Please could you help me with:
1) Know the origin of the possible problem with the humidifier (this is causing me problems, because since it does not turn it off, I find the terrarium at 100% humidity and with a lot of fog and my animal suffers)

2) I have searched for information to update, but it is complex for my knowledge, could you guide me a little how to update to the latest version?

vikrilow, 2022-06-04:
I attach picture from installation and configuration
You can see that switch 3 and 4 are disabled but however Humidifier are running... 

In Addition, I send a screenshoot about switches from  "RESERVA" and "Himifier" shut, but running

[image]

[image]
[image]
[image]

theyosh, 2022-06-04:
Hi,

if you manually toggle the relay, does it react to action in the interface. So when you click on the relay button in the GUI, does the relay change state?

Because I get the feeling that this is faulty hardware. If 1 relay out of 4 is acting strange/wrong, then I am afraid it is hardware related. So upgrading the software will not fix that.

Sometimes it can help to power it totally down, like draining all the hardware. That would sometimes help. It is like a hard reset.

theyosh, 2022-06-17:
Hi,

First of, there is not a migration path to TP4. So basically, you do a reinstall. So, that means you have to recreate your current setup from scratch. The most safe way, is to start with a new SD card. So if it goes wrong with TP4, you can just put the SD card with TP3 back in the PI and it should work again.

Also, make sure you backup the following files: https://github.com/theyosh/TerrariumPI/wiki/FAQ#which-files-to-backup=

For a new SD card, you can follow this setup: https://theyosh.github.io/TerrariumPI/install/

When the installation is done, reboot and then you can setup your system again. https://theyosh.github.io/TerrariumPI/setup/

When that is done, you can transfer old relay data to TP4, so that the total costs are still correct. If you do not care about the total power usage and costs, you can skip this step: https://theyosh.github.io/TerrariumPI/install/#migration-from-v3-to-v4

I hope this is clear, and that you are able to upgrade to TP4 with al the new goodies :)

theyosh, 2022-06-26:
Is the above explanation enough to upgrade? It should not be that hard to do. Only a bit of work.

vikrilow, 2022-07-05:
Hi friend

I'm sorry for delay, I've been with a lot of work and I could not be able to try it.
Today I've configured new SD and I'm configuring new parameters. I think all rigth!! But I think I will have to open new issue in query mode hehe

You can close this issue. If I had any problem, I would open a new issue.

Thanks for your help

closed 2022-07-07 by theyosh