theyosh/TerrariumPI #709: [DEVICE]: Support for I2C 4CH AC LED Dimmer Module

Dragonfly-terra, 2022-06-19:
Request for adding support for a 4 channel Dimmer module over I2C.

With this module, you can control the intensity of incandescent lamps, LED dimmable bulbs, power heating elements or fan speed controlled through a microcontroller or Arduino/Raspberry boards.
[image]

Technical info

Compatible with any Arduino, Raspberry or microcontroller.

* Dimming method - leading edge
* Working voltage - 110 or 240 VAC
* Current per channel - up to 3A
* Compatible with 50Hz and 60Hz AC Line
* AC Line autodetect function
* 100kHz I2C bus interface
* I2C bus pull up switch on board
* I2C device address select switch on board
* Can be connected up to 8 devices on I2C bus (8 devices x 4 = 32 channels!!!)
* 50HZ LED - show detected AC 50HZ Line
* 60HZ LED - show detected AC 60HZ Line

**Pinning description**  
[image]

**I2C Addressing**
[image]

Communication
[image]

| Channel | Address      |
| ------- | ------------ |
| 1       | 0x80 |
| 2       | 0x81 |
| 3       | 0x82 |
| 4       | 0x83 |

Required settings
I2C address
Channel number (1 until 4)

Example code light full (no dimming) channel 1
```
Wire.beginTransmission(i2c-address); // device address
Wire.write(channel-address);             // select 1st channel 0x80
Wire.write(value);               // write dimm value "0" to 1st channel
Wire.endTransmission();
```

Example code light off channel 3
```
Wire.beginTransmission(i2c-address); // device address
Wire.write(channel-address);             // select 3rd channel 0x82
Wire.write(value);               // write dimm value "100" to 3rd channel
Wire.endTransmission();
```

Example code light 75% dimming channel 2
```
Wire.beginTransmission(i2c-address);    // device address
Wire.write(channel-address);            // select 2nd channel 0x81
Wire.write(value);                      // write dimm value "75" to 2nd channel
Wire.endTransmission();
```

Sources  
[I2C 4CH AC LED Dimmer Module](https://www.tindie.com/products/bugrovs2012/i2c-4ch-ac-led-dimmer-module/)

Dragonfly-terra, 2022-06-25:
@theyosh 

So I have no idea what I am doing and I am not a programmer, but I am trying to learn a little bit of what is going on in the software.
So I added the _init_.py file in hardware/relay folder and added the 4dimmer_i2c.py file.

It would be great if you could review it and perhaps give me some tips and advice to make it work. That would be great.

Thank you very much for all your effort and time you spend on making this awesome project.  Looking forward to your reply.

[relay.zip](https://github.com/theyosh/TerrariumPI/files/8985469/relay.zip)

theyosh, 2022-06-26:
Hi,

I have added this relay to the code. But I am unable to test it. But you have changed the __init__.py file, which is not handy, and not needed. So please do the command: `git checkout hardware/relay/__init__.py` in the `/home/pi/TerrariumPI` folder.

Then you should be able to do the command: `git pull` and you should get the latest code, with support for this dimmer.

Restart TP4, and you should be able to add a new dimmer called `I2C 4Channel dimmer`. The address that you need to use is: `[relay_number],[i2c_address],[i2c_bus (optional)]`

Dragonfly-terra, 2022-06-26:
Thank you so much, unfortunately I am running it in a docker 😂
If I use the theyosh/terrariumpi:development image, should it have the changes?

theyosh, 2022-06-26:
Nope,

but I am building one now. But that takes about 2 hours :(. Also, using docker is not handy for debugging and testing. So you could try to pull the development version in about 2 hours. But I am not sure if it will work, and how easy you can see the errors.

Dragonfly-terra, 2022-06-27:
Oke I did the following steps.
First I deleted the docker and installed a fresh install (without the docker, a normal install because of the debugging)

After setting everything up, I tried adding the new dimmer I2C 4Channel LED AC dimmer.

I have tried the following address combinations:
1,0x27,22
0,0x27,22
0x80,0x27,22
0x27,22,0x80
0x27,22,0
0x27,22,1

The message I keep getting is:
```
Could not load hardware for relay I2C 4Channel LED AC dimmer dimmer named 'Dimmer 1-1' at address '0,0x27,22': can't set attribute, retrying in 0.5 seconds...
```
The dimmer is found in i2cdetect.
```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- 27 -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: UU -- -- -- -- -- 76 --
```

Is there anything I can provide or test for you? Thanks in advance.

Dragonfly-terra, 2022-06-27:
I have did a `git pull` and tested it.
The first channel is address `1,0x27,22` and then it was working.
Thank you so much. Still need to do some testing and configuring. I will keep you updated. Thanks again.

theyosh, 2022-06-27:
Ah yes. Humans start counting at '1'. Computers at '0' ;)

But it looks like it works? Very nice...

theyosh, 2022-06-28:
I think I found a small bug. So to make sure it stays working, please do a `git pull` once more. Than I think it should be ready.

Dragonfly-terra, 2022-06-30:
Hello,

I did the `git pull` and everything looks working fine. Still haven't had the change to fully test but everything looks great.
Thank you so much. I will keep you updated.

Dragonfly-terra, 2022-06-30:
I have done some testing with a incandescent lamp, 60Watt but unfortanatly the light is on and stays on, Perhaps because the start value is 0. There is no active dimming. Tried every channel but no result.

When I look at the logging I see the following:
`2022-06-30 22:42:28,703 - INFO    - terrariumEngine       - Updated relay i2c_4ch-dimmer dimmer named 'Dimmer 2-2' at address '4,0x27,22' with new value 50.00 in 0.01 seconds.'

I don't know if the value 50.00 is send or 50 (without the .00) but maybe this could be the issue?
Looking forward to your reply.

theyosh, 2022-07-02:
Hm,

with dimming I do not have all the answers. Not all lamps can be dimmed with all devices. So I test normally with a normal light bulb like: https://external-content.duckduckgo.com/iu/?u=https%3A%2F%2Fi5.walmartimages.com%2Fasr%2F73edaadb-b7e3-4ac6-b8be-62a0229b45a6_1.d648686056f101384f81e3449e1fd603.jpeg&f=1&nofb=1

But I do not have this kind of dimmer, so I cannot test it. So if you are able to hook up a 'normal' light bulb and see when you change the dimmer value if the light does dim. It could be that it works in reverse, then I have to change the code a bit.

But, first I need to know if the dimmer does dim. Just adding it, is not a valid test that it will work.

Dragonfly-terra, 2022-07-02:
I have tested it with a normal light bulb but the intensity of the light did not change with the dimmer settings.
Perhaps I can send you my dimmer (or bring it to you) I don't know where you are from.

theyosh, 2022-07-02:
So I ordered two to test with. I hope they will come soon so I can do some testing myself.

theyosh, 2022-07-13:
Hmm, I have updated my code based on your example. Do a `git pull` and see if the dimmer is working. Mine is still in transit somewhere. Track and trace is not telling me where it is right now ;(

Dragonfly-terra, 2022-07-13:
Did a ```git pull``` but it gets a error
This is the debug logging:
```
/home/terrariumpi/TerrariumPI/venv/lib/python3.7/site-packages/requests/__init__.py:104: RequestsDependencyW                                                                                                   arning: urllib3 (1.26.9) or chardet (5.0.0)/charset_normalizer (2.0.12) doesn't match a supported version!
  RequestsDependencyWarning)
2022-07-13 22:33:05,662 - INFO    - terrariumEngine       - Starting TerrariumPI 4.3.3 ...
2022-07-13 22:33:05,729 - INFO    - terrariumEngine       - Loaded 28 settings in 0.06 seconds.
2022-07-13 22:33:05,766 - INFO    - terrariumEngine       - Loading weather data from source https://api.ope                                                                                                   nweathermap.org/data/2.5/weather?q=Neerkant,NL&appid=d3bd66034ff8ca70ba8e301955ba9308
2022-07-13 22:33:06,416 - INFO    - weather.openweathermap_org_weather - Loaded new historical weather data                                                                                                    (48 measurements) from 2022-07-11 04:00:00 till 2022-07-13 03:00:00 in 0.35 seconds.
2022-07-13 22:33:06,417 - INFO    - weather               - Loaded new weather data in 0.647 seconds.
Traceback (most recent call last):
  File "terrariumPI.py", line 16, in <module>
    terrariumEngine = terrariumEngine(__version__)
  File "/home/terrariumpi/TerrariumPI/terrariumEngine.py", line 112, in __init__
    self.webserver = terrariumWebserver(self)
  File "/home/terrariumpi/TerrariumPI/terrariumWebserver.py", line 54, in __init__
    self.__routes()
  File "/home/terrariumpi/TerrariumPI/terrariumWebserver.py", line 237, in __routes
    self.api.routes(self.bottle)
  File "/home/terrariumpi/TerrariumPI/terrariumAPI.py", line 146, in routes
    all_sensor_types = '|'.join(terrariumSensor.sensor_types)
  File "/home/terrariumpi/TerrariumPI/terrariumUtils.py", line 38, in __get__
    return classmethod(self.fget).__get__(None, owner)()
  File "/home/terrariumpi/TerrariumPI/hardware/sensor/__init__.py", line 108, in sensor_types
    for sensor in __cls__.available_sensors:
  File "/home/terrariumpi/TerrariumPI/terrariumUtils.py", line 38, in __get__
    return classmethod(self.fget).__get__(None, owner)()
  File "/home/terrariumpi/TerrariumPI/hardware/sensor/__init__.py", line 91, in available_sensors
    for (hardware_type, sensor) in __cls__.available_hardware.items():
  File "/home/terrariumpi/TerrariumPI/terrariumUtils.py", line 38, in __get__
    return classmethod(self.fget).__get__(None, owner)()
  File "/home/terrariumpi/TerrariumPI/hardware/sensor/__init__.py", line 65, in available_hardware
    imported_module = import_module( '.' + file.stem, package='{}'.format(__name__))
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/terrariumpi/TerrariumPI/hardware/sensor/atlasscientific_sensor.py", line 10, in <module>
    from AtlasI2C import AtlasI2C
ModuleNotFoundError: No module named 'AtlasI2C'
```

theyosh, 2022-07-13:
crap.. rerun the installer. Should add the missing module

Dragonfly-terra, 2022-07-13:
I removed the atlasscientific_sensor.py for now and it booted up.
Tested the dimmer and it is working like a charm. Tomorrow I will do some additional testing. Thank you so much for all the great work.

theyosh, 2022-07-13:
Cool, Removing that sensor file is also a work around :).

So, lets see if it all works like expected. Then I can create a new release.

theyosh, 2022-07-21:
Ok then I close it now. Thanks for testing!

closed 2022-07-21 by theyosh