theyosh/TerrariumPI #715: Crashes following upgrade from 4.1 to 4.3

morrell7, 2022-06-30:
Hi Joshua,

I hope you are well.

When I first upgraded I had a crash about 20 hours after with similar messages to this:

2022-06-30 15:59:26,270 - ERROR   - hardware.sensor       - Sensor BME280 sensor temperature named 'Temperature' at address '77,1' has exception: [Errno 24] Too many open files: '/dev/i2c-1'

I believe this then causes a watchdog - but this morning it's allowed temps to reach 4 degrees higher than the threshold. 

Could you please have a look when you have a moment, I can forward ssh again for you.

Best regards,
Mark

morrell7, 2022-07-01:
Critical error while processing request: /
Error:
OSError(24, 'Too many open files')
Traceback:
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/bottle.py", line 962, in wsgi
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/bottle.py", line 915, in _cast
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 234, in handle_error
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/bottle.py", line 3655, in template
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/bottle.py", line 3264, in __init__
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/bottle.py", line 3354, in prepare
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 724, in exec_module
  File "<frozen importlib._bootstrap_external>", line 859, in get_code
  File "<frozen importlib._bootstrap_external>", line 916, in get_data
OSError: [Errno 24] Too many open files: '/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/jinja2/__init__.py'

The above showed as html on the last crash

theyosh, 2022-07-01:
Hi,

this looks like a broken or faulty I2C device. The first error: `2022-06-30 15:59:26,270 - ERROR - hardware.sensor - Sensor BME280 sensor temperature named 'Temperature' at address '77,1' has exception: [Errno 24] Too many open files: '/dev/i2c-1'`

is somewhat know to happen when there I2C issues. It does not have to be that sensor, but some device is not closing the I2C correctly.

You can check this by installing `lsof`: `sudo apt install lsof` and then run it: `lsof | grep i2c | wc -l` If the resulting number is higher then 100, then that is the case. If it is lower, then we have to do some more research.

If you run the command `lsof | grep i2c | wc -l` every minute, the number will probably increase, which should not be the case.

morrell7, 2022-07-01:
I've just done a new update a moment ago. 

after 8 minutes uptime this value is up to 146 but there are no errors yet.

What else could be driving this up? I do have things like Unix agent running.

morrell7, 2022-07-01:
302 now after 18 minutes still no errors in terrariumPi

theyosh, 2022-07-01:
Hmm the only thing I can think of is that you use a sensor that I do not have. And for that sensor the library maybe updated, which is somehow causing this. But this is hard to figure out.

There are no errors in the log files?

About the gui is strange, as it does/should work. So maybe a browser cache.

Also, maybe it could help to stop TP4, en rerun the installer.  And then do a reboot.

And sometimes it can also help to cut the power at all to reset all connected devices.

And you could try to ignore 1 sensor at a time and run for 15 min. And measure the lsof command. Hoping that at a certain point the i2c list is not growing. And then you know which sensor is causing this. In order to un-exclude a sensor, you have to change the setting and restart TP4. Else the ignored sensor will stay ignored even when the settings are saved.

theyosh, 2022-07-07:
Did you find any clues? A tip could be, if possible, to clean all the GPIO pins and other pins with a alcohol cleaning fluid. I use nail polish remover. It could be that there is some dust on it which makes a bad electrical connection. Cleaning them with alcohol could help sometimes.

morrell7, 2022-07-23:
Sorry for not responding sooner.

I've since done another update and the problem seems to have stopped..

Had a power trip and seems one relay on the board is failing and stuck in an on position.
So I need to replace that. But strange, I thought if it were broken the relay would be always off not on?

I'll close this for now thanks.

closed 2022-07-23 by morrell7