theyosh/TerrariumPI #716: [BUG]: Enclosure - Area settings button disappears of the screen and small typo

Dragonfly-terra, 2022-06-30:
**Setup:**
 - TerrariumPI version: 4.3.3 [main]
 - Python version: 3
 - Raspberry PI: 4
 - Webbrowser: Opera GX version LVL 4 (core: 88.0.4412.65)

**Describe the bug**
Display of the settings button disappears add the enclosures list.
Small typo in the Area settings Dimmer. Dimmer **of** duration should be Dimmer **off** duration

**To Reproduce / Installation steps**
Setting up a light with day and night timing. Long text resulting in disappearing of the settings button.
Add a Area with a Dimmer.

**Expected behavior**
Perhaps a Night settings could be placed below the Day settings. Or move the settings button before the area settings.

**Screenshots**
[image]
[image]

Thanks for this great software and all your support.

theyosh, 2022-07-02:
Hi,

if you do a `git pull` and restart TP4, both issue should be fixed.

Dragonfly-terra, 2022-07-09:
Yes this is fixed 👍

closed 2022-07-09 by theyosh