theyosh/TerrariumPI #718: [FEATURE]: Support for Atlas Scientific PT-1000 Submersible thermometer?

victorhooi, 2022-07-07:
What do you think about adding support for the Atlas Scientific PT-1000 subermsible thermometer?

https://atlas-scientific.com/kits/pt-1000-temperature-kit/

(I originally saw it referenced on the [Mycodo build log](https://hackaday.io/project/11997-mycodo-environmental-regulation-system/log/45473-software-updates-new-sensors-and-methods)).

It's definitely pricier than say, the DS18B20, but it's I2C, industrial-grade, and meant to be quite accurate/reliable.

What do you think?

victorhooi, 2022-07-07:
Also - you'll probably have less chance of fakes/counterfeits - which are rife with the DS18B20:

https://github.com/cpetrich/counterfeit_DS18B20

victorhooi, 2022-07-07:
Got it - that's awesome!

I didn't realise it was already supported =).

Does it make sense to add it to this list then?

https://theyosh.github.io/TerrariumPI/hardware/

Yeah, thinking about getting some - you're right they are pricey though, aha.

theyosh, 2022-07-08:
Euh, I just added the code. So it is new. The documentation will be updated when I release a new version.

So, it would be nice it you would be able to test it. Else, there is not much more I can do.

victorhooi, 2022-07-09:
Aha, got it - I didn't realised you'd just added - thanks!

I'll see if I can order a PT-1000 sensor and test then.

theyosh, 2022-07-09:
Just added the documentation also: https://theyosh.github.io/TerrariumPI/hardware/sensor/atlasscientific-i2c-sensor/

victorhooi, 2022-07-20:
Quick update - I've ordered one of these, will update when I get a chance to try it out! =).

theyosh, 2022-09-23:
Hi, are there any update on this? Did you get a hardware sensor to test with?

theyosh, 2022-12-11:
Closing this issue as the hardware is tested and working with release 4.6.0

closed 2022-12-11 by theyosh