theyosh/TerrariumPI #719: [BUG]:  Display stuttering in the dashboard overview.

Dragonfly-terra, 2022-07-09:
Setup:

TerrariumPI version: 4.3.3 [main]
Python version: 3
Raspberry PI: 4
Webbrowser: Opera GX version LVL 4 (core: 88.0.4412.65)

Describe the bug
Display stuttering in the dashboard overview.

To Reproduce / Installation steps
Display settings 1680x1050, full screen.
It looks like it has something to do with the scroll-bar

Expected behavior
No stuttering

Screenshots
[image]

Thanks for this great software and all your support.

theyosh, 2022-07-09:
Hi,

I am afraid this is a browser issue. I have tested this with Opera Version:89.0.4447.38 (Linux) and I can reproduce. As I saw that the GX version is a special version for Windows only. So I suggest to test this with the normal Opera browser and see if it is still the case.

I can't reproduce this stuttering, and you are the first in years that is mentioning this. So I get the feeling this is a browser issue which I cannot fix.

Dragonfly-terra, 2022-07-09:
Hello,

I have tested it on a different browser and you are right, looks like the issue is coming from the browser.
Thanks for you reply.

closed 2022-07-09 by Dragonfly-terra