theyosh/TerrariumPI #720: [FEATURE]: LED-Warrior18 Module support

Jexkin, 2022-07-15:
Hey there, I am back again with another module to hopefully allow for "better" LED dimming.

I'll link the store page and github.

https://shop.codemercs.com/en/led-warrior18-module.html
https://github.com/codemercs-com/lw18

I already ordered two of them and I hope that those will be "better" than the IRF520/IRF3708 module.
But they of course need to be supported by TerrariumPi first.

So if it's possible, I can do some testing again once the modules arrive (hopefully next tuesday or wednesday).

I assume the "0.0015% to 100%" PWM will not fully work, but only the usual 1-100% in1% steps right?

Kind regards,
Jexkin

theyosh, 2022-07-15:
Hi,

yeah I think I can add this one. The code is not that nice to implement, so I will write my own driver. But the example code is clear enough to support the dimmer.
So I will try to add it this weekend.

And you are correct that I will just support steps of 1% each from 0% till 100%

Jexkin, 2022-07-15:
Thank you!

I will keep you updated on when they arrive at my home. Then I can test if everything works, if you find some time to add the code this weekend :)

I'm still unsure what to expect from the module but who knows. Hopefully it'll be a worthy purchase.

JoshuaR1337, 2022-07-21:
I have added some code, but looking at their examples at https://github.com/codemercs-com/lw18 it looks like you cannot set the frequency of a single dimmer. Because when setting the frequency, you need to set both dimmers.

The same goes with the current state, but that can be read out from the hardware. Therefore the code to support this dimmer is a bit strange, and maybe buggy. But at the moment I cannot test it.

So then it will up to you to test the support of this dimmer.

Jexkin, 2022-07-21:
Hey there,

I think that'll be fine! Thank you!
Not sure why someone would like to have two different frequencies?
But maybe there'd be some use for that. For me it'll be fine if both dimmers have the same frequency.
Just asking, but I'd have to set the frequency to 730Hz anyway right? Since it says `PWM output at 730 Hz` on their GitHub?

I'm still waiting on some things to arrive but I hope I can test it this weekend or next monday.
Will keep everyone updated on this!

Jexkin, 2022-07-22:
Hi,

the rest of the parts arrived!
I already tried to get it running but I can't seem to add the device? Screenshot below.
[image]

And those are my settings right now, please let me know if I need to change something!
(I already tried `0,0x20` but that didn't work either)
[image]

Am I missing something? The module should be connected properly to my Pi as it shows up in the i2c detector
[image]

theyosh, 2022-07-22:
Ah, I think I see it. The entered address is not correct. As you have 2 dimmers on 1 device, the address should be: `1,0x20` and `2,0x20`. The first number is the relay number, second part is the I2C address.

Jexkin, 2022-07-22:
Just tried it with both but now I get this message:
[image]

Same settings as before, just changed the address to `1,0x20` and after that to `2,0x20`.

Jexkin, 2022-07-22:
Did a `git pull` and restarted it, I can add the module now properly!

Already tested if the dimming works but so far no luck.
As soon as I put it to any % it just goes on without the chance of any dimming.
So even if I set it to 10%, it's working at 100% brightness.
The only other thing that works is setting it to 0%, that turns it off as it should. But anything besides 0% makes it work at 100%.

theyosh, 2022-07-22:
Hmm, I think then the best way is maybe to test it with this script: https://github.com/codemercs-com/lw18/blob/main/src/rpi/python/lw18.py

Because 1 thing that is strange to me, is the frequency settings.
At line https://github.com/codemercs-com/lw18/blob/main/src/rpi/python/lw18.py#L26= we have the max possible frequency ?
Then the next line https://github.com/codemercs-com/lw18/blob/main/src/rpi/python/lw18.py#L27= we have the default frequency? But this variable is never used.... 
According to: https://github.com/codemercs-com/lw18/blob/main/src/rpi/python/lw18.py#L151= the max frequency value is 65535. Does that mean 65535 / 48000000 = 0,001365312 for every 1 hz. So for 730Hz I would expect: 0,996678125 which is lower then zero, which is not valid according to the code.

Then on line https://github.com/codemercs-com/lw18/blob/main/src/rpi/python/lw18.py#L238= it say that 730 hz is default which is the value of 65535. Which is the max. That means that the max at line 26 is bogus.. not valid.

So I get the feeling that the example code is not valid. But I cannot test it. So can you try to test the dimmer with https://github.com/codemercs-com/lw18/blob/main/src/rpi/python/lw18.py and see if that works?

Jexkin, 2022-07-23:
Hey there, sorry for the kinda late reply.

I did test the script right now and I am kinda confused on the output but regardless, here:
[image]

To explain what happened to the LEDs:
With every higher value for the `PWM value` the brighter the LEDs got.
The LEDs did not get brighter than value 730, seems like that was the max PWM value.

The `FREUQENCY` value confused me.
There it seems to be way different than the `PWM value`.
The lower the number, the higher the brightness. 
[image]
Even at a value of `1000000` the LEDs were still on but on a veeeery low brightness.

Hope this somewhat helps?

theyosh, 2022-07-26:
Sorry for the delay, needed some time for other things :)

All the numbers are a bit confusing, and I think that the code is not really correct. But it seems to work. So I think we need to do some tests. I will try to explain what I am looking for.

Looking at the script, the values for `pwm` and `frequency` can be 0 - 65535 https://github.com/codemercs-com/lw18/blob/main/src/rpi/python/lw18.py#L57= and https://github.com/codemercs-com/lw18/blob/main/src/rpi/python/lw18.py#L96=
So can you to start, set both pwm and frequency value of 1 dimmer to the value 32767 which should be 50% pwm dimming and half the frequency which is valid based on code.
Then you should lower the pwm value to zero in steps of 50 or so. That should either make the light go out or brighter. The goal here is to see if the value 65535 is indeed the max pwm value you can use. A higher number, should not make brighter light. As I read the code, a value higher then 65535 will be lowered to 65535. So value 1000000 is actually 65535 and should produce the same brightness or darkness.

So, things I need to know (this is all with the same frequency of 32767):
- using value 0 for pwm, does this dim the light and the light is off? Or is this full on
- is value 65535 the max for pwm. Higher values do not change the light. Not sure if you can see this. Could be to bright for your eyes.

Then the next test would be, what will happen when you put the pwm value to 32767 (if this is the middle of the pwm range), and play with the frequency value. Again the range to play with is 0 - 65535. Same tests again. Is 65535 the max? The important thing here, is that the frequency value is not a value in Hz. But it is just a number.

- using value 0 for frequency, does this dim the light and the light is off? Or is this full on
- is value 65535 the max for frequency

All these tests are done with 1 light/led connected.

Then the next test would be, what will happen when you connect the second led and start playing with that frequency values. As I can read the code, when you set the first dimmer to a frequency value of 32767 and the second one you change it, in what ever value, the first frequency will be reset to 65535 which would make the first light change, even when you do not want that.
So that means, when you change the frequency of a single dimmer, it will also reset the other dimmer frequency, which is not what I want.

So the frequency part is a bit confusing and make it difficult. The pwm part does not look that hard, I only need to know what value 0 and 65535 do. Which value if full on.

Jexkin, 2022-07-27:
I tried to do everything you asked for but I currently don't have a spare LED strip for the testings. But so far here's what I did:

setting both the PWM and Frequency to 32767:
[image]

There was no increase or decrease in brightness when I also set the frequency to 32767 or vice versa. Seems like it only responds to either the PWM or Frequency value. (This wasn't something you asked for but- maybe good to know)

Lowering the PWM value:
Lowering the PWM value decreased the brightness.
I really had to do very big steps because it was really not noticeable with small steps.
At a value of `0` the LEDs were off.
[image]

Is 65535 the max PWM value:
Yes. Though it seems like it just restarts at 0 if you put in a number higher than `65535`?
At a value of `70000` it wasn't off. At a value of `65536` it was off.
So I am thinking at a value of `65536` it just counts it as 0? And e.g. `70000` would be 4465 (if my math stands correct)?
At least from my testing `70000` and `4465` seem to be the same brightness.
[image]

Frequency testing:
Setting the frequency value to `0` while the PWM is set to `32767` makes it shine the brightest.
[image]

Interestingly, increasing the frequency even in `5000` steps, I did not see any change in brightness even at `30000`.
Once I set the value to `65535` the LED decreased in brightness visibly. 
I expected it to turn off but it did not. So 65535 seems not to be the max for the frequency?

There is no change in brightness regardless if I even set the PWM value to `32757` or `65535` when changing the frequency value. Really makes me think that one can control the LEDs with either a Frequency Value or a PWM Value.

I hope this helps! And sorry for not being able to test it with two LEDs.

theyosh, 2022-08-01:
Hi,
sorry for the slow response, but I do not understand why it is not working. I copied the original code in order to change the dimmer.

So, I added some extra debug. Could you run TP4 manual / in debug: https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/ and then try to change the dimmer in the gui. Every change should produce some debug output. I hope that the debug makes clear what I am doing wrong...
Also, can you post the debug outcome here?

Also, use for frequency value in the gui the value 32757 in stead of 730. I do not understand how the frequency calculation is done, and therefore I use the same values as in the lw18.py script.

Jexkin, 2022-08-01:
Trying to set the frequency value to `32757` in the GUI doesn't work because the value is above `10000`.

theyosh, 2022-08-01:
Ah crap, changed that. Do a git pull and try again. A restart is not needed but cleaning browser cache could be needed

Jexkin, 2022-08-01:
Done that, did some increments of 10%:
(0-50%:)
[image]

(50-100%:)
[image]

It seems like to work way better now!
I guess the key really was to increase the frequency in the GUI from 730 to 32757. Or maybe even the full 65535?
After 50% there was no increase in brightness anymore. As you can see in the output `50 -> 32767`, so at that point it was running at basically 100% power?
I'll try to set the frequency in the gui to 65535 right now. Will keep you updated!

Jexkin, 2022-08-01:
Set the value of the frequency in the GUI to 65535:
[image]

Now testing in 10% increments again
(0-50%:)
[image]

(50-100%:)
[image]

Definitely saw some increase in brightness for every 10% now.
So even after 50% it kept getting brighter! So I feel like it's important to set the frequency in the GUI to `65535` to get "true 100% dimming"?

In other "news": I really don't see any difference between the LED Warrior 18 and the IRF520/3708 module. Although, maybe if there's ever the possibilty to increase the dimming with decimals? That'd be cool to test but IIRC you said that's not possible without rewriting the code?

theyosh, 2022-08-05:
So it looks to work with a frequency value of `65535`. So that is what I put on the hardware information page https://theyosh.github.io/TerrariumPI/hardware/relay/led-warrior18/#setup

> In other "news": I really don't see any difference between the LED Warrior 18 and the IRF520/3708 module. Although, maybe if there's ever the possibilty to increase the dimming with decimals? That'd be cool to test but IIRC you said that's not possible without rewriting the code?

In my opinion is 100 steps for dimming enough. The problem is that dimming light does not seems to be a linear line from 0 till 100%. So for us it looks like the that at 75% it is already at max power. I can't explain it well, because it is a bit complicated. Look for dimming questions in the github issues, and you will find a lot of questions about it.

So, for now, I leave it as is. And that other dimmers looks like doing the same thing (good or wrong), is something I cannot explain or change. The only advantage of this LED18 warrior is that it not needs the Raspberry PI to make the PWM signal. This is a time critical action, and when the Pi is a bit busy, it could be a problem to keep the PWM stable. So with the LED18 warrior that is handled on the board, and a busy PI does not affect the PWM.

But I made a new release yesterday, and I think the support for this dimmer is now done, and I would like to close this issue.

Jexkin, 2022-08-05:
Thank you very much!

Feel free to close this issue :)

theyosh, 2022-08-05:
Thanks for testing!

closed 2022-08-05 by theyosh