theyosh/TerrariumPI #721: Multiple fake w1-sensors

kahuwi14, 2022-07-16:
Hey there
I've got a problem since the update to the recent version:

In my setup, I'm using one w1-DS18B20, but since the update, there appeare countless devices within terrariumpi.

[image]

It's not possible to delete them, I'm only able to put them to the "ignore"-list.
The second problem with this are the countless reading-failures: they lead to restarts because of the watchdog.

I already tried to find them in the database, but I don't know, how they are. The "sensor"-table only contains the actual available sensors.

I'd love to get some hints to fix this..

greeets

kahuwi14, 2022-07-16:
A snippet of my syslog:

```
Jul 16 14:44:02 terrariumpi python[17932]: 2022-07-16 14:44:02,177 - INFO    - weather.openweathermap_org_weather - Loaded new historical weather data (48 measurements) from 2022-07-14 04:00:00 till 2022-07-16 03:00:00 in 0.53 seconds.
Jul 16 14:44:02 terrariumpi python[17932]: 2022-07-16 14:44:02,180 - INFO    - weather               - Loaded new weather data in 1.057 seconds.
Jul 16 14:44:02 terrariumpi python[17932]: 2022-07-16 14:44:02,197 - INFO    - terrariumNotification - Connecting to MQTT Broker at address: iobroker.kahuwi:1885 ...
Jul 16 14:44:02 terrariumpi python[17932]: 2022-07-16 14:44:02,207 - INFO    - terrariumNotification - Logged in to MQTT Broker at address: iobroker.kahuwi:1885.
Jul 16 14:44:28 terrariumpi kernel: [ 4927.424242] w1_master_driver w1_bus_master1: Attaching one wire slave 04.620000000000 crc 2d
Jul 16 14:44:28 terrariumpi kernel: [ 4927.434901] w1_master_driver w1_bus_master1: Family 4 for 04.620000000000.2d is not registered.
Jul 16 14:45:17 terrariumpi kernel: [ 4976.905244] w1_master_driver w1_bus_master1: Attaching one wire slave 04.e20000000000 crc a1
Jul 16 14:45:17 terrariumpi kernel: [ 4976.914975] w1_master_driver w1_bus_master1: Family 4 for 04.e20000000000.a1 is not registered.
Jul 16 14:46:01 terrariumpi systemd[1]: terrariumpi.service: Watchdog timeout (limit 2min)!
Jul 16 14:46:01 terrariumpi systemd[1]: terrariumpi.service: Killing process 17932 (python) with signal SIGABRT.
Jul 16 14:46:01 terrariumpi systemd[1]: terrariumpi.service: Main process exited, code=killed, status=6/ABRT
Jul 16 14:46:01 terrariumpi systemd[1]: terrariumpi.service: Failed with result 'watchdog'.
Jul 16 14:46:06 terrariumpi systemd[1]: terrariumpi.service: Service RestartSec=5s expired, scheduling restart.
Jul 16 14:46:06 terrariumpi systemd[1]: terrariumpi.service: Scheduled restart job, restart counter is at 27.
Jul 16 14:46:06 terrariumpi systemd[1]: Stopped TerrariumPI.
Jul 16 14:46:06 terrariumpi systemd[1]: Starting TerrariumPI...
Jul 16 14:46:10 terrariumpi kernel: [ 5030.004617] w1_master_driver w1_bus_master1: Attaching one wire slave 04.120000000000 crc d5
Jul 16 14:46:10 terrariumpi kernel: [ 5030.012462] w1_master_driver w1_bus_master1: Family 4 for 04.120000000000.d5 is not registered.
Jul 16 14:46:13 terrariumpi systemd[1]: Started TerrariumPI.
Jul 16 14:46:13 terrariumpi python[18144]: 2022-07-16 14:46:13,598 - INFO    - terrariumEngine       - Starting TerrariumPI 4.3.3 ...
Jul 16 14:46:13 terrariumpi python[18144]: 2022-07-16 14:46:13,690 - INFO    - terrariumEngine       - Loaded 28 settings in 0.09 seconds.
Jul 16 14:46:13 terrariumpi python[18144]: 2022-07-16 14:46:13,730 - INFO    - terrariumEngine       - Loading weather data from source https://api.openweathermap.org/data/2.5/weather?q=kelberg,de&appid=a1cc120ef5971c303d2d31ef42e209cd
Jul 16 14:46:14 terrariumpi python[18144]: 2022-07-16 14:46:14,844 - INFO    - weather.openweathermap_org_weather - Loaded new historical weather data (48 measurements) from 2022-07-14 04:00:00 till 2022-07-16 03:00:00 in 0.57 seconds.
Jul 16 14:46:14 terrariumpi python[18144]: 2022-07-16 14:46:14,847 - INFO    - weather               - Loaded new weather data in 1.110 seconds.
Jul 16 14:46:14 terrariumpi python[18144]: 2022-07-16 14:46:14,869 - INFO    - terrariumNotification - Connecting to MQTT Broker at address: iobroker.kahuwi:1885 ...
Jul 16 14:46:14 terrariumpi python[18144]: 2022-07-16 14:46:14,884 - INFO    - terrariumNotification - Logged in to MQTT Broker at address: iobroker.kahuwi:1885.
Jul 16 14:47:01 terrariumpi kernel: [ 5081.125047] w1_master_driver w1_bus_master1: Attaching one wire slave 04.920000000000 crc 59
Jul 16 14:47:01 terrariumpi kernel: [ 5081.136053] w1_master_driver w1_bus_master1: Family 4 for 04.920000000000.59 is not registered.
Jul 16 14:47:27 terrariumpi dhcpcd[548]: enxb827eb11a09d: Router Advertisement from fe80::1eed:6fff:febe:af01
Jul 16 14:47:27 terrariumpi dhcpcd[548]: enxb827eb11a09d: adding default route via fe80::d472:cbff:fe2a:941b
Jul 16 14:48:05 terrariumpi kernel: [ 5144.635370] w1_master_driver w1_bus_master1: Attaching one wire slave 04.520000000000 crc 93
Jul 16 14:48:05 terrariumpi kernel: [ 5144.645867] w1_master_driver w1_bus_master1: Family 4 for 04.520000000000.93 is not registered.
Jul 16 14:48:13 terrariumpi systemd[1]: terrariumpi.service: Watchdog timeout (limit 2min)!
Jul 16 14:48:13 terrariumpi systemd[1]: terrariumpi.service: Killing process 18144 (python) with signal SIGABRT.
Jul 16 14:48:13 terrariumpi systemd[1]: terrariumpi.service: Main process exited, code=killed, status=6/ABRT
Jul 16 14:48:13 terrariumpi systemd[1]: terrariumpi.service: Failed with result 'watchdog'.
Jul 16 14:48:19 terrariumpi systemd[1]: terrariumpi.service: Service RestartSec=5s expired, scheduling restart.
Jul 16 14:48:19 terrariumpi systemd[1]: terrariumpi.service: Scheduled restart job, restart counter is at 28.
Jul 16 14:48:19 terrariumpi systemd[1]: Stopped TerrariumPI.
Jul 16 14:48:19 terrariumpi systemd[1]: Starting TerrariumPI...
Jul 16 14:48:26 terrariumpi systemd[1]: Started TerrariumPI.
Jul 16 14:48:26 terrariumpi python[18398]: 2022-07-16 14:48:26,223 - INFO    - terrariumEngine       - Starting TerrariumPI 4.3.3 ...
Jul 16 14:48:26 terrariumpi python[18398]: 2022-07-16 14:48:26,314 - INFO    - terrariumEngine       - Loaded 28 settings in 0.09 seconds.
[... cut ...]
Jul 16 14:49:34 terrariumpi kernel: [ 5233.555145] w1_master_driver w1_bus_master1: Family 4 for 04.320000000000.f6 is not registered.
Jul 16 14:49:34 terrariumpi python[18398]: 2022-07-16 14:49:34,460 - INFO    - terrariumEngine       - Loading existing sensors from database.
Jul 16 14:49:34 terrariumpi python[18398]: 2022-07-16 14:49:34,500 - WARNING - hardware.sensor       - Unable to load sensor 1-Wire sensor temperature named '1-Wire sensor measuring temperature' at address '10-0ae840040000': Unable to load sensor 1wire 1-Wire sensor measuring temperature at address 10-0
```

Python always adds new 1W-Sensors

theyosh, 2022-07-16:
Damm... this is strange. Not seen before. But there could be a quick and simple fix for this if you do NOT use 1-wire: https://pinout.xyz/pinout/1_wire#

So, if you do not need that, you can edit the file /boot/config ( `sudo nano /boot/config` ) and add a `#` or remove the line `    dtoverlay=w1-gpio` and reboot. This should disable the 1-wire in the OS, and should stop giving all kinds of fake 1-wire devices.

Normally, it should just see that there is nothing connected to pin 7. Could it be that you have another sensor/relay on that pin?

I can't explain why this is happening. Maybe a reboot could also solve it?

theyosh, 2022-07-16:
Oh crap, you are using it.... Hmmm then I can only think of a reboot.

theyosh, 2022-07-19:
Another brain wave....
Maybe the standard 1-wire GPIO pin is somehow, somewhat broken. You could consider to change the 1-wire GPIO pin https://pinout.xyz/pinout/1_wire# : `dtoverlay=w1-gpio,gpiopin=x` where x is the GPIO PIN number.

theyosh, 2022-07-22:
Any updates? Does the 1-wire work again? I am out of other options.

kahuwi14, 2022-07-29:
Sorry, I was a really busy..
I just reactivated the 1-wire bus and it seems to work...
I'm not sure, why...

I'll have to observe it I'll keep you in the informed..

theyosh, 2022-08-01:
Ok, luckily it is working again. But a bit confusing why. Lets hope it stays working

kahuwi14, 2022-08-07:
Sadly not...
Since my last message, I had the system running in "manual-mode":
I deactivated the misting-system and the notification via pushover. Everthing worked finde. No problems with restarts, n false redings from the dht22.
After the replacing of the defective waterpump and some testruns, I just disabled the manual mode of the waterpump-relais and reactiviated the notification and one moment later: restart notification in pushover and the gui is down....
 I'll do more research now...

kahuwi14, 2022-08-07:
One idea:
Some time ago, we had the problem, that the mqtt-service was connecting to the server, although it was deactivated.
I still have this problem.
I know deleted the mqtt-config and it seems to run stable now..
I'll have a look at this....

theyosh, 2022-10-08:
Updated the code to ignore disabled notification services during startup. So that should now be fixed.

Can we close this issue and when the mqtt is still giving issues, could you open a new issue?

kahuwi14, 2022-10-10:
Thanks!
I'm currently not at home, but I'll try it as soon as possible.

----------------------------------------

kahuwi14, 2022-10-23:
Last nights, we came back from our holidays, so I installed the update now.
I'll have a lok at the restarts, but I have several d1s that aren't not there....
I now ignored all d1-Sensors and have a look at the behaviour.

If you'd lie to close it anyway, don't mind, I'll keep you update within the issue.

theyosh, 2022-11-07:
Ok, for now I close this issue. It is a strange thing, but I think it is a hardware issue outside my software. I would suggest to use a clean install and start adding hardware with new connectors, or just a bit shorter cables to make sure you have clean connections with the cables.

closed 2022-11-07 by theyosh