theyosh/TerrariumPI #722: Problem with bme280 sensor in V. 4.3.3

vikrilow, 2022-07-16:
Hello
I've problem with bme280 sensor. I use these for humidity.
I can add sensors but they don't read nothing.
I can see i2c connections.
[image]

[image]

I can see that in log session:
INFO    - terrariumEngine       - Updated sensor bme280 humidity named 'HUMEDAD_COOL_ZONE' at address '77,4' with new value 0.00% in 0.05 seconds.
2022-07-16 17:44:59,881 - INFO    - terrariumEngine       - Updated sensor bme280 humidity named 'HUMEDAD_HOT_ZONE' at address '76,4' with new value 0.00% in 0.04 seconds.

I don't know what is the problem. Some help??
thanks

theyosh, 2022-07-16:
Hm, I think you could create a small python script and test the sensor manually. Copy the code below and put it in a file. Call it test-bme280.py: `nano test-bme280.py` and past the code below. Place the file in the TerrariumPI folder.

```python
import smbus2
import bme280

port = 1
address = 0x76
bus = smbus2.SMBus(port)

calibration_params = bme280.load_calibration_params(bus, address)

# the sample method will take a single reading and return a
# compensated_reading object
data = bme280.sample(bus, address, calibration_params)

# the compensated_reading class has the following attributes
print(data.id)
print(data.timestamp)
print(data.temperature)
print(data.pressure)
print(data.humidity)

# there is a handy string representation too
print(data)
```

Save the file with Ctrl+X. Then you need to activate the Python source environment: `source venv/bin/activate` . This should put the text '(venv)' in front of the command line.

Then run the test script: `python test-bme280.py` . This should produce some output. If not, it could be that the sensor died.

And what could also help is to shutdown the Pi, and take it of the power for a minute or so and then power it on again. Make sure all hardware is reset.

vikrilow, 2022-07-16:
Thanks!!!
I don't know if there is correct or not
pi@raspberrypi:~/TerrariumPI $ python test-bme280.py
Traceback (most recent call last):
  File "test-bme280.py", line 8, in <module>
    calibration_params = bme280.load_calibration_params(bus, address)
  File "/usr/local/lib/python2.7/dist-packages/bme280/__init__.py", line 153, in load_calibration_params
    compensation_params.dig_T1 = read.unsigned_short(0x88)
  File "/usr/local/lib/python2.7/dist-packages/bme280/reader.py", line 40, in unsigned_short
    return self._bus.read_word_data(self._address, register) & 0xffff
  File "/usr/local/lib/python2.7/dist-packages/smbus2/smbus2.py", line 474, in read_word_data
    ioctl(self.fd, I2C_SMBUS, msg)
IOError: [Errno 121] Remote I/O error

[image]

I think there is any problem with my rasp or wired, because I have problem with wired sensor too... It's very rare..

theyosh, 2022-07-16:
` IOError: [Errno 121] Remote I/O error` are mostly bad connections. I know it is strange after an update, but this is what is my experience.

Did you try to shutdown and removed the power from the pi. So that there no power at all at any devices. This could sometimes fix it. Needs a 'hard' reset.

It that does not help, I would suggest create a new SD card and install TP4 https://theyosh.github.io/TerrariumPI/install/ and copy the following files: https://theyosh.github.io/TerrariumPI/faq/which-files-to-backup/

Then you can just switch the SD card to see if that makes a difference.

If that also fails, then you have to look at your hardware and wiring. And make sure everything is still connected correctly, and that all connections are clean. Maybe there is some dust of grease on it from using a long time.
I clean the pin connections with nail polish remover. That removes a lot of dirt if there is some on it.

vikrilow, 2022-07-17:
I think physical cables are fine, that is to say, the cables are connected and none are loose. However, I think some cables are possible that they have rust. 
Really something I've done wrong, because wired sensors have problem too
2022-07-17 16:30:35,004 - WARNING - terrariumSensor      - Measured temperature value NoneC from w1 sensor 'Temp_hot' is outside valid range 0.00C - 100.00C in 0.00250 seconds.
FileNotFoundError: [Errno 2] No such file or directory: '/sys/bus/w1/devices/w1_slave'
with open(os.path.join(terrarium1WSensor.W1_BASE_PATH,self.get_address(),'w1_slave'), 'r') as w1data:
File "/home/pi/TerrariumPI/terrariumSensor.py", line 334, in load_data
Traceback (most recent call last):

I've changed SD with V3 and problem is same... 

I've always feared that this day would come, to check all the wiring... hehe

theyosh, 2022-07-17:
You had a backup of TP3 on a different SD card? And that is also causing sensor issues. Then I am afraid that some wiring or maybe the sensor it self is broken.

If you are going to re-wire, I think it is wise the clean all the GPIO pins with a nail polish remover fluid and swabs. That cleans is nicely and does not affect the electronics.

And if you need to buy new hardware, always buy extra spare parts at the same time. I have about 10+ BME280 and some other sensors at home. Not all sensors are of the same quality, and always wise to have some extra spares.

But this is sorry to hear. I hope you can fix it. For me, some sensors are not accessible for me if they brake. Keep my fingers crossed

vikrilow, 2022-07-17:
The original problem has been because I bougth some sensors BME280 and I wanted change old sensor bcs always showed 100% humidity...
But I've test with some sensors and problem continue...

theyosh, 2022-07-17:
And what I also noticed with another project, when it can get humid, and therefore a bit wet, you could use a plastic spray to cover the connection with a small plastic rubber coating. Make sure you protect the actual sensor when spraying with a peace of sticky paper. Then the hole sensor and its connector should stay a bit more protect against moist and water.

theyosh, 2022-07-17:
I also noticed you created new language Spanish in the TP3 version. I would suggest not to do, because TP3 is not supported anymore, and translating can take some time.

I am thank full for everybody that wants do to some translating, but then it would be better to do it only in the 4.x.y.z version, which is TP4.

vikrilow, 2022-07-17:
Yes I've created a new language to try to help the community.
But I only wanted to translate TP4, bcs I haven't V3 installed now.
I'm sorry for inconvenience, tell me how i can help for TP4 and delete TP3 option.
Yes I've created a new language to try to help the community.
But I only wanted to translate TP4, bcs I haven't V3 installed now.
I'm sorry for inconvenience, tell me how i can help for TP4 and delete TP3 option.

vikrilow, 2022-07-17:
Hello Again  @theyosh
In TP3 I had a fan programmed to work 5 minutes every 2 hours (to renew the air in the terrarium. Something essential for the species I keep) I really can't find this option, can you help me?

I show you a screenshot, to see better 
[image]

theyosh, 2022-07-17:
Use the On and Off duration fields and clear the Power on and Settle time.
On duration 5 and Off duration 115 . That should do the trick.
Maybe this could also help: https://theyosh.github.io/TerrariumPI/scenarios/ here you can play with the 6 timer fields to see how it works, and when the relays should be on.

vikrilow, 2022-07-20:
Hello friend!!
I've some stuff that comment!!
* About BME280 sensor, I think that the problem is that the new sensor were wrong (I remember that this sensor has been modified and maybe I didn't do correctly. 
* About 1wired sensors, the problem was in gpio. Moving the raspberry, I saw a loose GPIO wire and put it on the wrong gpio. Once placed well, the error was eliminated.
* In this moment, I have reinstalled the SD with TP3. I configured weather, but I put the wrong URL and I could never connect to the website anymore. I think I had same problem that #246 issue.... I tried some solutions, but I couldn't repair it. 
I think you can close this issue

Thanks

JoshuaR1337, 2022-07-21:
I would suggest to upgrade to TP4. As TP3 is not supported or updated anymore. The save way to do is to use a new SDcard, and start installing TP4 on it. So you can swap SD cards as a backup.

theyosh, 2022-07-22:
Ok, I close this issue as it is has been resolved. But I suggest to upgrade to TP4. As TP3 does not get any software updates.
https://theyosh.github.io/TerrariumPI/install/#migration-from-v3-to-v4

closed 2022-07-22 by theyosh