theyosh/TerrariumPI #723: [BUG]: BME 280 can not be deleted

sni89, 2022-07-17:
Hey! First thanks for the great project. 
since the last update one of my Bme 280 is making trouble. 

I figured out the sensor seems to be broken but I am unable to delete it to stop it from causing errors. I get the following notes 

[image]

[image]

I just ignored the sensor for now but would like to replace it but also 2 new sensors are showing the same errors and give me no data… 

`2022-07-17 11:06:23,733 - ERROR   - hardware.sensor       - Error updating sensor BME280 sensor pressure named 'Luftdruck 1' at address '0x76'. Check your hardware! Invalid reading from sensor BME280 sensor pressure named 'Luftdruck 1' at address '0x76'
2022-07-17 11:06:23,730 - ERROR   - hardware.sensor       - Sensor BME280 sensor pressure named 'Luftdruck 1' at address '0x76' has exception: [Errno 121] Remote I/O error`

Would like to hear some advice!

theyosh, 2022-07-19:
Hi,

the first error is something I can look into it. The second error is much more worse. If the database is malformed, which means broken, it could case issues the next time you restart TP4.

So, do you have backup files? https://theyosh.github.io/TerrariumPI/faq/which-files-to-backup/

It is fixable, but it is a bit of puzzle. So make sure you have a backup. And then you could do some tests on the backup sqlite file.

I copy the database while in use. And that should at least not hurt the running Pi version.

The I2C error `[Errno 121] Remote I/O error` is mostly hardware connection issue. What I do with new / changed sensors is running a small testing script to see if the sensor does work. For your BME: https://github.com/theyosh/TerrariumPI/issues/722#issuecomment-1186272815

sni89, 2022-07-19:
I figured out myself that somehow sensors that worked with 3,3v (2 x bme 280) now need 5v to work fine. I can not explain the issue but with all sensors plugged TP is trapped in a circle of trying to boot and ends up rebooting or starting with the errors shown. With the two sensors unplugged or plugged in on 5v everything is fine. I haven’t changed a thing so it’s a strange thing that came with the update. Guess coincidence…

I have very long cables so maybe the sensors worked on their limit so far.

theyosh, 2022-07-22:
Ok, I did some more testing, and found out that deleting sensor data does not work due to the long time it takes. And that will trigger a restart due to a watchdog check. If there is not updates for 2 minutes, TP will be restarted. The delete action for testing took about 20 minutes :(. That is about 1 year of sensor history data for 1 sensor.

So when you want to delete a sensor that is being used for more then 2 months, it is saver the run TP in manual mode: https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/ and then do the delete action. In manual mode it does not matter how long the action takes. It will just continue afterwards.

It is always handy to have the database backed-up before start to delete the sensor.

So, what is the current state? All sensors are working again? Do you still need to delete a sensor? Is the database still working because of the malformed error.

theyosh, 2022-08-01:
Hi,

any updates on this? Or at least is you TP still working? I am a bit afraid about the 'malformed database' message.

But I did some testing, and it turns out, that deleting old data is just taking to much time. So I added an extra FAQ page for it: https://theyosh.github.io/TerrariumPI/faq/how-to-delete-a-sensor-relay-button-terrariumpi/

Because I can't do much about it, and would like to close this issue.

sni89, 2022-08-01:
Sorry I am in Norway with poor connection. 
TP works fine. I am running it with 1 instead of two Bme right now. I will try to add a second one after my holidays.

closed 2022-08-24 by sni89