theyosh/TerrariumPI #724: question: Can I add security cam? Ezviz, Ring, Blink??

vikrilow, 2022-07-17:
Hello
I've some cams to review my installations, type Ezviz, Blink or other...
Thease cams are possible to add to Terrarium?
Thanks

theyosh, 2022-07-17:
Hi,

this depends on the way you can watch the streams. If it is app only, it is not possible.

For the Ezviz I found this: https://www.ispyconnect.com/camera/ezviz

Click on the camera you have, and you can construct a rtsp url. You can test this url with VLC on your desktop. If that works, you can add this as a RTSP webcam. It will only take snapshots every 30 seconds, and can be streamed live.

For Blink I could not find any info. But the mail rule is, can you get a video stream url that can be used by VLC. If that is not possible, the camera is not suitable.

theyosh, 2022-09-23:
Is this still an issue? Is there an update or can we close this issue?

theyosh, 2022-10-07:
Closing due to no response.

closed 2022-10-07 by theyosh