theyosh/TerrariumPI #725: [Question]: Submersible Wifi Temperature sensor?

vimes1984, 2022-07-19:
I'm running two terrariums: 

- Paludarium with tropical fish ( guppies). 
-  Veiled Chameleon 

I have the TerrariumPI  centralized and everything connected via wifi or bluetooth, and I need a submersible Wifi or Bluetooth thermostat to control the aquarium tank heater, but I can't seem to find one that confirms to be working with Terrarium Pi.
I've been looking at the [Inkbird](https://www.amazon.co.uk/Inkbird-Temperature-Controller-Thermostats-Tortoise%EF%BC%88No/dp/B07WV42NX2/), but really I want something like the [Merross MS100](https://www.meross.com/mTerminal/product/46/article/)  that I can just drop into the water. 
( Also not sure if this is the right place for this question to be asked please excuse me if it isn't)

JoshuaR1337, 2022-07-21:
Chris,

No sonoff is only supported when flashed with Tasmota software. Which can be a bit hard to do depending on your Sonoff device.

But I would like to inform that you need some technical background in order to use my software. I do not give support on hardware and how to flash of fix hardware issues. That is to complicated to support.

So all by suggesting above require some technical skills. For example flashing Sonoff: https://tasmota.github.io/docs/devices/Sonoff-Basic/#serial-flashing you need some soldering skills as well.

Using temperature sensors: https://tasmota.github.io/docs/DS18x20/#sensor-versions

As you can see, this is all technical. If this is to difficult for you, then it is not an option. Or you have to find a friend how is able to flash and wire the hardware for you. But that is outside my support scope.

vimes1984, 2022-07-21:
@JoshuaR1337 no worries on the technical aspect of it I know my way around a soldering iron and flashing a chip :)

theyosh, 2022-07-22:
So, I think then Sonoff with Tasmota and DS1820 sensors is the way to go. Also, that is easier for me, as that is already supported :).
And also, I do support any kind of remote sensor that can be readout through http(s) and json: https://theyosh.github.io/TerrariumPI/hardware/sensor/remote-sensor/

But I am not sure if you can get a hand on them. And what is available now a days.

vimes1984, 2022-07-29:
GOT IT WORKING!!!!!

theyosh, 2022-07-29:
With a Sonoff flashed device? Maybe post a picture? Could also help other people.

vimes1984, 2022-07-29:
@theyosh yup with a flashed Sonoff basic TL;DR to get it working is using a Sonoff :

- Aquire a Sonoff 4CHPROR3  ( carefull here the RX at the end denotes different revisions or versions of the hardware I did this with the R3 but you can do it with whichever) 
- Un screw the back, you'll see 5 little holes on the board, get a row of 5 female headers ( I [used](https://www.amazon.co.uk/Freenove-Electronics-Programming-Solderless-Breadboard/dp/B06W54L7B5/)  bottom row [thrid](https://m.media-amazon.com/images/I/81ESBA8qJIL._AC_SL1500_.jpg) item from the right) 
- You'll also need a "USB to TTL Serial Converter Adapter Module 3.3V" 
- Solder the header into the board and connect your serial adapter to it 
- Flash the sonoff ( I used [tasmotizer](https://github.com/tasmota/tasmotizer) ) 
- This video helped me through it: https://www.youtube.com/watch?v=1K1vTNzsTxI
- Once you have the Sonoff flashed grab your DS18 attach a set of dupont connectors to the end of them. 
- Plug the ds18 into the header: Red vcc/3.3v, black gnd, yellow signal (RX/TX) 
- Plug the Sonoff into mains (DC) 
- Open the Tasmota web UI in a browser and run the command  ( `SetOption74 1` ) this powers the tx connector up mitagating the need to use a resistor between the vcc and tx 
- Go to your Tasmota interface web UI and click `configure -> configure module` and set it to:
[image]
- Follow this: https://templates.blakadder.com/howto.html using this template: `{"NAME":"Sonoff 4CHPROR3","GPIO":[17,255,255,255,23,22,18,19,21,56,20,24,0],"FLAG":0,"BASE":23}`
- Reboot the Sonoff 
- Add your sensor into terrariumPI the hardware type I set to ` Remote sensor http/https ` the Address `http://192.168.X.X/cm?cmnd=sonoff/status%2010#StatusSNS/DS18B20/Temperature` 
 
and you are away to the races!!!!!

vimes1984, 2022-07-29:
@theyosh I hope that helps!!! you also have the relays in the tasmoto I have yet to add them to terrarium PI but I assume it's rather trival at this point :)

vimes1984, 2022-07-29:
@theyosh as a side note not relevant to this those Meross sensors are awful I swapped the battery in one of them and I can't get it to reconnect now.... very very very annoying...

vimes1984, 2022-07-29:
Also anyone is welcome to ping me on this thread and I'll gladly help :)