theyosh/TerrariumPI #728: [BUG]: Unable to load web interface suddenly

vimes1984, 2022-07-20:
**Setup:**
 - TerrariumPI version:4
 - Python version: 3.9
 - Raspberry PI: 4

**Describe the bug**
I am unable to view the web interface in the browser after a reboot 
A clear and concise description of what the bug is.

**Screenshots**
Please add screenshots of the [debug screens](https://github.com/theyosh/TerrariumPI/wiki/FAQ#how-to-debug-terrariumpi):

**Additional context**
I'm suffering from the same error as: #244 and #issuecomment-1190757148,  running `~/terrariumPI/terrariumPI.py` gives me: ```Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 4, in <module>
    from gevent import monkey
ModuleNotFoundError: No module named 'gevent'
```

the logs show: 

![Screenshot from 2022-07-21 00-00-25](https://user-images.githubusercontent.com/2351665/180097009-6112d24b-48ca-4f53-beaa-5e370f4e8062.png)

vimes1984, 2022-07-20:
also potentially this error: 
https://github.com/theyosh/TerrariumPI/issues/569

vimes1984, 2022-07-20:
Just found these FAQ's to debug TerrariumPi 4... such a moron I am... https://theyosh.github.io/TerrariumPI/faq/

theyosh, 2022-07-21:
Hi,
these errors about 'No module found' sounds like a wrong installation. I suggest to rerun the installer: `sudo ./install.sh` in the TerrariumPI folder.

Running it manual: https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/

But try a rerun of the installer.sh script first.

vimes1984, 2022-07-21:
@theyosh I've just re-run `installer.sh` and the web interface still fails to load. 

Running it manually did last night at lease eventually run the web interface though! 

I'm now running into a messos login issue `1301` https://github.com/albertogeniola/MerossIot/issues/96...

JoshuaR1337, 2022-07-21:
This all sounds broken installations. Meross software will do nothing when you did not configure. As you are saying the gui is not starting up, there is no Merross settings setup and therefore, it is not of influence.

As rerunning the installer did not fix this, I am afraid you need to reinstall the SD card itself. Make sure you have install Debian Buster (old legacy image). https://theyosh.github.io/TerrariumPI/install/#raspberry-pi

vimes1984, 2022-07-21:
hey there, 
this WAS working fine with Meross until last night after I set up the notifications, this now runs if i Manually run it eventually by following this: https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/. 
But it still won't run the web interface on boot...

theyosh, 2022-07-21:
Oh wait... I see the message `Temporary failure in name resolution` which means a DNS issue.

Here you can see the domains that are used to connect with Meross: https://github.com/albertogeniola/MerossIot/blob/0.4.X.X/meross_iot/model/constants.py

So somehow the domain iot.meross.com or mqtt.meross.com did not resolve, so the meross manager can't connect. That is unfortunately outside my reach to fix.

You can test this in a Pi running: `dig iot.meross.com`

```
dig iot.meross.com

; <<>> DiG 9.11.5-P4-5.1+deb10u7-Raspbian <<>> iot.meross.com
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 3383
;; flags: qr rd ra; QUERY: 1, ANSWER: 3, AUTHORITY: 4, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 4096
;; QUESTION SECTION:
;iot.meross.com.			IN	A

;; ANSWER SECTION:
iot.meross.com.		28	IN	A	54.247.85.191
iot.meross.com.		28	IN	A	34.241.14.20
iot.meross.com.		28	IN	A	34.253.63.97

;; AUTHORITY SECTION:
meross.com.		171798	IN	NS	ns-1092.awsdns-08.org.
meross.com.		171798	IN	NS	ns-1853.awsdns-39.co.uk.
meross.com.		171798	IN	NS	ns-755.awsdns-30.net.
meross.com.		171798	IN	NS	ns-83.awsdns-10.com.

;; Query time: 20 msec
;; SERVER: 192.168.5.1#53(192.168.5.1)
;; WHEN: Thu Jul 21 17:26:55 CEST 2022
;; MSG SIZE  rcvd: 227
```

```
dig mqtt.meross.com

; <<>> DiG 9.11.5-P4-5.1+deb10u7-Raspbian <<>> mqtt.meross.com
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 2005
;; flags: qr rd ra; QUERY: 1, ANSWER: 3, AUTHORITY: 4, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 4096
;; QUESTION SECTION:
;mqtt.meross.com.		IN	A

;; ANSWER SECTION:
mqtt.meross.com.	60	IN	A	54.77.214.248
mqtt.meross.com.	60	IN	A	54.75.5.168
mqtt.meross.com.	60	IN	A	52.49.129.225

;; AUTHORITY SECTION:
meross.com.		171764	IN	NS	ns-1092.awsdns-08.org.
meross.com.		171764	IN	NS	ns-1853.awsdns-39.co.uk.
meross.com.		171764	IN	NS	ns-755.awsdns-30.net.
meross.com.		171764	IN	NS	ns-83.awsdns-10.com.

;; Query time: 37 msec
;; SERVER: 192.168.5.1#53(192.168.5.1)
;; WHEN: Thu Jul 21 17:27:29 CEST 2022
;; MSG SIZE  rcvd: 228
```

It should produce lines with `IN A [IP_NUMBER]`. If you have the same outcome, it should work.
Dig can be installed by running: `sudo apt install dnsutils`

theyosh, 2022-07-21:
Ok, then it is still a broken installation issue I guess, because there is also an error `ModuleNotFoundError: No module named 'emails'` which is used to send email notifications. So if you created an email notification, it could happen that TP4 will not start due to a missing module. But that module should be installed by the installer.

Because when you created an email service, it will try to send emails that TP4 is starting up and when fully started. I thought that would be a nice extra service.

vimes1984, 2022-07-21:
Ok, what's the best way to  do a complete un install?

theyosh, 2022-07-21:
There is not a real uninstall ;) But in your case, you should delete the `venv` folder in the TerrariumPI folder. Then rerun the installer. This should install the python part freshly.

During the installation you see python modules installed. As long there is `attempt 1` showing it is going well. As soon attempt 2 or 3 are showing up, it could indicate installation problems. Then we need to do it by hand, but lets first try to remove the `venv` folder

vimes1984, 2022-07-21:
Just for future people reading this I'm about to do this: 
Stopping the terrariumPI program: `sudo service terrariumpi stop` 
Entering into the directory: `cd /home/pi/TerrariumPI`
Then removing the venv folder: `rm -rf venv`

vimes1984, 2022-07-21:
@theyosh right that is re-installed now all on attempt 1, ( as a side note I booted it manually deleted the Messos integration earlier and ALL the relays and that caused it to boot correctly, adding the KASA relays back in and it booted fine but adding in the MESSOS sensors it failed to boot again.)

vimes1984, 2022-07-21:
and that still now fails to boot after the re-install. ( thanks for this @theyosh ): 
`2022-07-21 23:20:26,333 - ERROR   - terrariumCloud        - Error in cloud run: 'TerrariumMerossCloud' object has no >
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/aiohttp/connector.py", line 1154, in _create_direct_con>
    hosts = await asyncio.shield(host_resolved)
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/aiohttp/connector.py", line 880, in _resolve_host
    addrs = await self._resolver.resolve(host, port, family=self._family)
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/aiohttp/resolver.py", line 33, in resolve
    infos = await self._loop.getaddrinfo(
  File "/usr/lib/python3.9/asyncio/base_events.py", line 856, in getaddrinfo
    return await self.run_in_executor(
  File "/usr/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/gevent/_socketcommon.py", line 247, in getaddrinfo
    addrlist = get_hub().resolver.getaddrinfo(host, port, family, type, proto, flags)
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/gevent/resolver/thread.py", line 63, in getaddrinfo
    return self.pool.apply(_socket.getaddrinfo, args, kwargs)
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/gevent/pool.py", line 161, in apply
    return self.spawn(func, *args, **kwds).get()
  File "src/gevent/event.py", line 329, in gevent._gevent_cevent.AsyncResult.get
  File "src/gevent/event.py", line 359, in gevent._gevent_cevent.AsyncResult.get
  File "src/gevent/event.py", line 347, in gevent._gevent_cevent.AsyncResult.get
  File "src/gevent/event.py", line 327, in gevent._gevent_cevent.AsyncResult._raise_exception
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/gevent/_compat.py", line 65, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/gevent/threadpool.py", line 167, in __run_task
    thread_result.set(func(*args, **kwargs))
socket.gaierror: [Errno -3] Temporary failure in name resolution

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumCloud.py", line 182, in _main_process
    http_api_client = await MerossHttpClient.async_from_user_password(email=self._username, password=self._password)
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/meross_iot/http_api.py", line 106, in async_from_user_p>
    creds = await cls.async_login(email=email,
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/meross_iot/http_api.py", line 203, in async_login
    response_data = await MerossHttpClient._async_authenticated_post(url=url,
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/meross_iot/http_api.py", line 277, in _async_authentica>
    async with session.post(url, json=payload, headers=headers, proxy=http_proxy) as response:
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/aiohttp/client.py", line 1138, in __aenter__
    self._resp = await self._coro
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/aiohttp/client.py", line 535, in _request
    conn = await self._connector.connect(
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/aiohttp/connector.py", line 542, in connect
    proto = await self._create_connection(req, traces, timeout)
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/aiohttp/connector.py", line 907, in _create_connection
    _, proto = await self._create_direct_connection(req, traces, timeout)
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/aiohttp/connector.py", line 1166, in _create_direct_con>
    raise ClientConnectorError(req.connection_key, exc) from exc
aiohttp.client_exceptions.ClientConnectorError: Cannot connect to host iot.meross.com:443 ssl:default [Temporary fail>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumCloud.py", line 49, in _run
    data.result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 440, in result
    return self.__get_result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 389, in __get_result
    raise self._exception
  File "/home/pi/TerrariumPI/terrariumCloud.py", line 236, in _main_process
    self.manager.close()
AttributeError: 'TerrariumMerossCloud' object has no attribute 'manager'
2022-07-21 23:20:26,363 - WARNING - terrariumCloud        - Reconnecting to Meross cloud. Somehow the connection was >
2022-07-21 23:20:26,364 - INFO    - terrariumCloud        - Stopping Meross cloud ...
`

from the log file this is what it appears to do. I'll wait a few mins then try a manual load again.

theyosh, 2022-07-21:
Hmm, this looks like a dns issue again. But I am not so sure anymore. You mention that you have both Kasa en Meross devices?

Then you are the first, and then you could have found a bug. Because both libraries for that hardware use asyncio. And I have not tested multiple asyncio devices together. And asyncio and gevent are both systems for concurrency, which could bite each other.
I do not have enough knowledge about that.

So that means, if you have only meross or only kasa and it is working, but both are giving issues, then I have to look into the code how the fix this. Unfortunately the current error is saying that it could not resolve dns. So it is not helping me much.

vimes1984, 2022-07-21:
Nmapping to the problematic url `iot.meross.com:403` 
returns this: 
```
Starting Nmap 7.80 ( https://nmap.org ) at 2022-07-21 23:44 BST
Nmap scan report for iot.meross.com (34.241.14.20)
Host is up (0.046s latency).
Other addresses for iot.meross.com (not scanned): 34.253.63.97 54.247.85.191
rDNS record for 34.241.14.20: ec2-34-241-14-20.eu-west-1.compute.amazonaws.com

PORT    STATE SERVICE
443/tcp open  https

Nmap done: 1 IP address (1 host up) scanned in 0.17 seconds
```

vimes1984, 2022-07-21:
Running it manually now errors with: 
```
(venv) pi@raspberrypi:~/TerrariumPI $ python terrariumPI.py
2022-07-21 23:39:10,017 - INFO    - terrariumEngine       - Starting TerrariumPI 4.3.3 ...
2022-07-21 23:39:10,083 - INFO    - terrariumEngine       - Loaded 28 settings in 0.07 seconds.
2022-07-21 23:39:10,115 - INFO    - terrariumEngine       - Loading weather data from source https://api.openweathermap.org/data/2.5/weather?q=Aden,YE&appid=f90159bfaaaa87d37096faefbb41e4df
2022-07-21 23:39:10,762 - ERROR   - weather               - Error loading online weather data! Please check your source address: https://api.openweathermap.org/data/2.5/weather?q=Aden,YE&appid=*********
2022-07-21 23:39:10,769 - INFO    - terrariumEngine       - Loading Meross cloud connection.
2022-07-21 23:39:10,774 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ... 
2022-07-21 23:39:11,775 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ... 
2022-07-21 23:39:12,776 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ... 
2022-07-21 23:39:13,778 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ... 
2022-07-21 23:39:14,780 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ... 
2022-07-21 23:39:15,782 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ... 
2022-07-21 23:39:16,784 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ... 
2022-07-21 23:39:17,785 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ... 
2022-07-21 23:39:18,786 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ... 
2022-07-21 23:39:19,787 - INFO    - terrariumCloud        - Waiting for Meross cloud connection ... 
2022-07-21 23:39:20,789 - INFO    - terrariumCloud        - Meross cloud is connected! Found 2 devices in 10.02 seconds.
2022-07-21 23:39:21,194 - INFO    - terrariumEngine       - Loading existing sensors from database.
2022-07-21 23:39:21,216 - INFO    - terrariumEngine       - Loaded sensor ms100 temperature named 'Meross MS100 paludarium 1 measuring temperature' at address '120027D1F3C3' with value 20.90C in 0.02 seconds.
2022-07-21 23:39:21,219 - INFO    - terrariumEngine       - Loaded sensor ms100 humidity named 'Meross MS100 paludarium 1 measuring humidity' at address '120027D1F3C3' with value 100.00% in 0.00 seconds.
2022-07-21 23:39:21,225 - INFO    - terrariumEngine       - Loaded sensor ms100 temperature named 'Meross MS100 homer mid enclosure measuring temperature' at address '120027D20830' with value 18.20C in 0.01 seconds.
2022-07-21 23:39:21,227 - INFO    - terrariumEngine       - Loaded sensor ms100 humidity named 'Meross MS100 homer mid enclosure measuring humidity' at address '120027D20830' with value 88.00% in 0.00 seconds.
2022-07-21 23:39:21,228 - INFO    - terrariumEngine       - Scanning for new sensors ...
2022-07-21 23:39:33,639 - INFO    - terrariumEngine       - Loaded 4 sensors in 12.44 seconds.
2022-07-21 23:39:33,640 - INFO    - terrariumEngine       - Loading existing relays from database.
2022-07-21 23:39:34,184 - INFO    - terrariumEngine       - Loaded relay tplinkkasa relay named 'K2 UVB' at address '192.168.0.145,1' value 0.00 in 0.53 seconds.
2022-07-21 23:39:34,190 - INFO    - terrariumEngine       - Loaded relay tplinkkasa relay named 'K2 Basking' at address '192.168.0.145,2' value 0.00 in 0.00 seconds.
2022-07-21 23:39:34,194 - INFO    - terrariumEngine       - Loaded relay tplinkkasa relay named 'K2 Mist' at address '192.168.0.145,3' value 0.00 in 0.00 seconds.
2022-07-21 23:39:34,349 - INFO    - terrariumEngine       - Loaded relay tplinkkasa relay named 'K3 Mist' at address '192.168.0.155,2' value 0.00 in 0.15 seconds.
2022-07-21 23:39:34,353 - INFO    - terrariumEngine       - Loaded relay tplinkkasa relay named 'K3 Canopy' at address '192.168.0.155,3' value 0.00 in 0.00 seconds.
2022-07-21 23:39:34,355 - INFO    - terrariumEngine       - Scanning for new relays ...
No GEMBIRD SiS-PM found. Check USB connections, please!
2022-07-21 23:39:39,684 - INFO    - terrariumCloud        - Got an update from the Meross Cloud. Setting temperature to 20.9 and humidity to 100.0
2022-07-21 23:39:39,685 - INFO    - terrariumCloud        - Got an update from the Meross Cloud. Setting temperature to 18.2 and humidity to 88.0
2022-07-21 23:39:49,015 - INFO    - terrariumEngine       - Loaded 5 relays in 15.38 seconds.
2022-07-21 23:39:49,015 - INFO    - terrariumEngine       - Loading existing buttons from database.
2022-07-21 23:39:49,023 - INFO    - terrariumEngine       - Loaded 0 buttons in 0.01 seconds.
2022-07-21 23:39:49,023 - INFO    - terrariumEngine       - Loading existing webcams from database.
2022-07-21 23:39:49,029 - INFO    - terrariumEngine       - Loaded 0 webcams in 0.01 seconds.
2022-07-21 23:39:49,029 - INFO    - terrariumEngine       - Loading existing enclosures from database.
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 16, in <module>
    terrariumEngine = terrariumEngine(__version__)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 150, in __init__
    self.__load_existing_enclosures()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 1031, in __load_existing_enclosures
    new_enclosure = self.add(terrariumEnclosure(
  File "/home/pi/TerrariumPI/terrariumEnclosure.py", line 22, in __init__
    self.load_areas(areas)
[... cut ...]
  File "src/gevent/_abstract_linkable.py", line 451, in gevent._gevent_c_abstract_linkable.AbstractLinkable._switch_to_hub
  File "src/gevent/_greenlet_primitives.py", line 61, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_greenlet_primitives.py", line 65, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_gevent_c_greenlet_primitives.pxd", line 35, in gevent._gevent_c_greenlet_primitives._greenlet_switch
KeyboardInterrupt: 
(venv) pi@raspberrypi:~/TerrariumPI $ 
```

theyosh, 2022-07-22:
Hmm, this time, it could not load the weather data from online. It is strange, but this again looks like a network related issue.

But both meross and kasa seems to work together... It is getting a bit strange. And I am almost sure that when the weather does not load, it is a network issue. Sorry to say, but all errors have something to do with the internet connection.
So, I can only say, try again a manual start, and see if it does get the weather data.

I tested it with your url and it works for me.

vimes1984, 2022-07-23:
Yeah, it's so weird it's working fine today with no changes.... My other
comps are all showing a good connection... But today it just works,I kind
of agree with you here that it's DNS on my side for some weird reason...
I'll dig in tomorrow to see :) thanks for ya help!

theyosh, 2022-08-01:
So it is a week later, still stable? So we can close this issue? I am afraid it was a some kind of network problem.

vimes1984, 2022-08-01:
Yeah it must have been a network issue on my end with metros please

vimes1984, 2022-08-01:
I might open a ticket and put in a pull request to build out of the software runs into a network issue like this it errors on screen in the browser instead of in the console..

closed 2022-08-01 by vimes1984