theyosh/TerrariumPI #729: [FEATURE]: Add support for Atlas Scientific EZO-HUM temperature/humidity sensor?

victorhooi, 2022-07-21:
Would it be possible to add support for the Atlas Scientific EZO-HUM probe?

https://atlas-scientific.com/probes/humidity-probe/

Technical data sheet is here:

https://files.atlas-scientific.com/EZO-HUM-Datasheet.pdf

It supports both UART and I2C.

It's pricier than some of the cheaper sensors, but it's industrial-grade, IP67 rated, and should do well in wet environments like you'd get in terrariums/vivarium/paludariums etc =).

(MyCodo has support for the same device - I believe the source is [here](https://github.com/kizniche/Mycodo/blob/master/mycodo/inputs/atlas_humidity.py), in case that helps at all).

victorhooi, 2022-07-21:
And yes, going to try to get my hands on one of these to verify.

I can't find any Australian distributors - seems to be easier to get in US/EU.

JoshuaR1337, 2022-07-21:
Hi Victor,

this is also supported. As the AtlasScientific I2C solution is generic, all devices should be supported. But as I cannot test it, I only supported  'ph','temperature','humidity','co2','conductivity','pressure' those are the most used types.

theyosh, 2022-09-23:
Hi, are there any update on this? Did you get a hardware sensor to test with?

Cinett87, 2022-10-13:
just seeing this  chain I asked the question under discussion just earlier, I will order all the parts  to test them but https://atlas-scientific.com/electrical-isolation/whitebox-t3/ I am intrigued especially by the whitebox and you can attach 5 sensors at once and whether that is compatible

theyosh, 2022-10-15:
@Cinett87 yes. As far I can see those boxes are also supported. But I think you cannot have multiple times the same sensor connected. So 2 temperature sensors could fail. As I expect that they use the same I2C address, which will collide with each other.

I do not see any options to change the sensor address, and there fore you cannot use 2 sensors or more of the same type.

closed 2022-12-11 by theyosh