theyosh/TerrariumPI #733: [FEATURE]: Single ESP32 board with all sensors included

vimes1984, 2022-08-18:
**Is your feature request related to a problem? Please describe.**
Currently to run multiple enclosures on terrarium PI it requires either individual off the shelf sensors of differing types or wiring directly into the Raspberry PI. 

**Describe the solution you'd like**
Building a esp32 breakout board that will except a collection of sensors and offer a json endpoint with tasmota so we can have one sensor array PER terrarium. 

**Additional context**
We've started the basic design [here](https://github.com/vimes1984/terrarium_sensor/), and added got a basic breadboarded prototype working with Tasmota... 
You can see the basic wiring diagram here: 
[image]

As we work through this from breadboard to breakout ESP32 board we'll keep that repo updated with more details.! 
Just thought we'd let you know @theyosh :)

theyosh, 2022-08-18:
Hi @vimes1984 

This is cool. I am currently playing with a ESP8266 ( https://components101.com/development-boards/nodemcu-esp8266-pinout-features-and-datasheet ) and Micropython to read out some senors and provide a JSON endpoint.

My experience with the ESP8266 is that there is not much memory to run everything. And that a ESP32 is a bit stronger.

But this is very cool to make. And makes is more easier to add more sensors on difficult to reach locations.

vimes1984, 2022-08-18:
Have you looked into tasmota yet? I'm pretty sure it'll work out the esp 8266's and it'll do the JSON output for you out of the box:) getting this esp32 wired up with UVB/temp/temp+humidity only took a day or two of futzing on the esp32's

theyosh, 2022-08-18:
He, I like to program myself :P

vimes1984, 2022-08-18:
Fair enough!!

theyosh, 2022-10-08:
But I can confirm using Tasmota works way better :P And it is that easy to do.

Based on Tasmota, adding a sensor would go through https://theyosh.github.io/TerrariumPI/hardware/sensor/remote-sensor/ and using the following url: http://[TasmotaIP]/cm?cmnd=Status%200#StatusSNS/DS18B20**-2**/Temperature

Where the part of **-2** is the second DS18B20 sensor.

STGMavrick, 2024-08-23:
Following! This seems great!