theyosh/TerrariumPI #734: Database Corruption [BUG]:

G4KCM, 2022-08-23:
On two occasions I have experienced database corruption as a result of power failures.  Whilst loss of historic measured data is of little concern what has worried me is that whilst the system restores with all the sensor configurations intact the alams/control setpoints have been corrupted.  This has resulted in setpoints being randomly adjusted outside of the 'comfort zone'.

This does not happen everytime there is a power failure so I cannot reproduce it upon demand.  I just wonder if there is a way to do some integrity check on the system control parameters i.e. a sort of checksum that could flag up incorrect system startup or perhaps embedding setpoint data in some more robust form within the database.  It appears that both historic and setting data maybe held within the same database (I think) maybe a seperate db file for setpoint data that is not so frequently accessed and hence less likly to be scribbled upon during an uncontrolled shutdown?

Best regards and as always appreciate the great work you are doing.

Clive

theyosh, 2022-08-27:
Hi, this is a bit complicated. In order to get the software fast, I used some Sqlite speed ups, which can harm the data when there are crashes.

And I did create some code for database recovery, but that does not play well with the systemd startup scripts and timeouts.

So I was thinking just to make a small script that can fix this. And that you have to run the script manual due to the size and how much time it can take.

In order to test this, do you have the broken DB file still? And could you upload it here? Any passwords in it should not be readable for me as it is encrypted with keys on your PI. Then I can create a work around for this.

And maybe I can make some code changes where the SQLite tuning is a bit more flexible.

G4KCM, 2022-08-27:
> Hi, this is a bit complicated. In order to get the software fast, I used some Sqlite speed ups, which can harm the data when there are crashes.
>
> And I did create some code for database recovery, but [...]

Understand the issues,  the only problem from my perspective is that the only parameters that seemed to be changed were the setpoints.  This could prove dangerous if someone had a terrarium which had enough heating/cooling/misting to turn the environment hostile to whatever lives in it.  Coming from a background of process control and the early days of computerised automation I have left the manual thermostat set to catch any failure and the consequences of having to explain cooked gecko to my wife :-) 
Have uploaded the corrupt db files in a zip here as I thought  they might be useful for you.  We live in a rural area and unfortunately suffer quite a lot of power failures.  Appreciate your help.

The file is too large for Github hopefully you can retrieve it here

https://drive.google.com/file/d/1_ONvS-o3JFZB7B00MkXQoYarldmyk2wU/view?usp=sharing

Best rgds - Clive

theyosh, 2022-08-29:
Hi,

I made a new script that should fix/restore the database. You can find it at `contrib/fix_db.py` after a `git pull`. Run the scripts with the following steps:

1. First enable python environment: source ../venv/bin/activate
2. Run this script with the -h for more information: ./fix_db.py -h

But I am not sure if I understand your issue. The remark `the only parameters that seemed to be changed were the setpoints`. What do you mean? When I looked in your db, it all looks correct. Could you explain a bit more.

Because when the database gets corrupted, I expect that some (last) history data is lost. But if you did not change the sensors/relays/areas, they should work normal after the db fix.

I will look into a way to make the database more resilience against power outages.

theyosh, 2022-10-08:
Do you have an update about `the only parameters that seemed to be changed were the setpoint` remark of you? I still do not understand what you mean with that.

I have updated some code in order to make the SQLITE 3 database better resistant to power outages. More info is at: https://theyosh.github.io/TerrariumPI/faq/#tune-database-settings . Also I updated the database recovery process: https://theyosh.github.io/TerrariumPI/faq/#sqlite3-databaseerror-database-disk-image-is-malformed

This is the most I can do.

G4KCM, 2022-10-11:
Dear Yosh, Thank you for this, much appreciated.   With regards to the
settings I can't explain it other than the fact that the alarm settings
were changed in system after it had restored.  Maybe if I had shut it down
and restarted it again it would have run with the correct values but I
immediately loaded a backup copy of the db files and restarted with those.

If it happens again I will try to adopt a more analytical approach but at
the time some of the terrarium parameters were quite outside of the norm
and I was keen to get it back to normal.

Best regards

Clive

theyosh, 2022-10-15:
Clive,

thanks for the extra info. I do understand that the animals comes first. :) Same here. But what you could do when it happens again, just try to make a copy of the broken db, and then restore it. So you will have a copy of the broken db. Then I could try to see what the state of the database is.

But, I think still that only historical data should be lost, as that is the last changed data. Unless you had changed the sensor or area settings within a few minutes before a crash. Then the .wal file could maybe be broken, and not yet synced back to the main database file.

But with the new database tuning settings, I hope that the setting synchronous=FULL will make the database a bit more crash resistant. But it will also make the system a bit slower, as there are more disk operations.

theyosh, 2022-10-23:
Can we close this issue? Because you try to test a power outage, to see if it works better. Else there is not much more I can do about this.

G4KCM, 2022-10-23:
> Can we close this issue? Because you try to test a power outage, to see if it works better. Else there is not much more I can do about this.

Yes of course I should have done it before. Thanks for the reminder

closed 2022-10-23 by G4KCM