theyosh/TerrariumPI #735: Web interface won't load anymore after power loss of the raspberry

Python823, 2022-08-27:
Hi,
I have a new raspberry Pi board running the application for the first time.
After a power loss on the raspberry pi, the web interface is not reachable any more.
I had a similar problem before today, so i formatted the SD, and started all over again.
The interface had been running for a couple of hours.

I only made some small configuration settings so far:
- Added a i2c multiplexer with 1 sensor to test (was working fine)
- Set a fixed IP address for the Wlan connection (did this before i installed the terrariumpi application)

Is there anything i can do to find out what the problem could be?
Thanks!

theyosh, 2022-08-27:
A quick fix could be deleting the databases in the `data` folder. But then you will not know what the error was.

So you could run it manually, and then it should produce some errors which you can post back here in this issue: https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/

Python823, 2022-08-27:
Hi Yosh,

Thanks for your very quick answer.
I'm new to raspberry pi and Python , so I don't understand everything yet, but i think this is what you need?
I do agree not to delete the database yet and rather find out what the problem is.

2022-08-27 20:29:05,066 - INFO    - terrariumEngine       - Starting TerrariumPI 4.4.0 ...
2022-08-27 20:29:05,135 - INFO    - terrariumEngine       - Loaded 28 settings in 0.07 seconds.
2022-08-27 20:29:05,673 - INFO    - terrariumEngine       - Loading existing sensors from database.
2022-08-27 20:29:05,703 - ERROR   - hardware.sensor       - Sensor Sensirion SHT3X sensor temperature named                                                               'Dembe_Basking_T' at address '44' has exception: I2C transceive failed: [Errno 121] Remote I/O error
2022-08-27 20:29:05,707 - WARNING - hardware.sensor       - Invalid reading from sensor Sensirion SHT3X sen                                                              sor temperature named 'Dembe_Basking_T' at address '44', retrying in 0.5 seconds...
2022-08-27 20:29:06,230 - ERROR   - hardware.sensor       - Sensor Sensirion SHT3X sensor temperature named                                                               'Dembe_Basking_T' at address '44' has exception: I2C transceive failed: [Errno 121] Remote I/O error
2022-08-27 20:29:06,231 - WARNING - hardware.sensor       - Invalid reading from sensor Sensirion SHT3X sen                                                              sor temperature named 'Dembe_Basking_T' at address '44', retrying in 0.5 seconds...
2022-08-27 20:29:06,755 - ERROR   - hardware.sensor       - Sensor Sensirion SHT3X sensor temperature named                                                               'Dembe_Basking_T' at address '44' has exception: I2C transceive failed: [Errno 121] Remote I/O error
2022-08-27 20:29:06,756 - ERROR   - hardware.sensor       - Error updating sensor Sensirion SHT3X sensor te                                                              mperature named 'Dembe_Basking_T' at address '44'. Check your hardware! Invalid reading from sensor Sensiri                                                              on SHT3X sensor temperature named 'Dembe_Basking_T' at address '44'
2022-08-27 20:29:06,758 - WARNING - terrariumEngine       - sht3x temperature named 'Dembe_Basking_T' at ad                                                              dress '44' had problems reading a new value during startup in 1.08 seconds. Will be updated in the next rou                                                              nd.
2022-08-27 20:29:06,760 - INFO    - terrariumEngine       - Scanning for new sensors ...
2022-08-27 20:29:15,971 - INFO    - terrariumEngine       - Loaded 1 sensors in 10.30 seconds.
2022-08-27 20:29:15,972 - INFO    - terrariumEngine       - Loading existing relays from database.
2022-08-27 20:29:15,987 - INFO    - terrariumEngine       - Scanning for new relays ...
No GEMBIRD SiS-PM found. Check USB connections, please!
2022-08-27 20:29:27,066 - INFO    - terrariumEngine       - Loaded 0 relays in 11.09 seconds.
2022-08-27 20:29:27,067 - INFO    - terrariumEngine       - Loading existing buttons from database.
2022-08-27 20:29:27,087 - INFO    - terrariumEngine       - Loaded 0 buttons in 0.02 seconds.
2022-08-27 20:29:27,088 - INFO    - terrariumEngine       - Loading existing webcams from database.
2022-08-27 20:29:27,098 - INFO    - terrariumEngine       - Loaded 0 webcams in 0.01 seconds.
2022-08-27 20:29:27,099 - INFO    - terrariumEngine       - Loading existing enclosures from database.
Traceback (most recent call last):
  File "terrariumPI.py", line 16, in <module>
    terrariumEngine = terrariumEngine(__version__)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 150, in __init__
    self.__load_existing_enclosures()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 1045, in __load_existing_enclosures
    new_enclosure.update()
  File "/home/pi/TerrariumPI/terrariumEnclosure.py", line 102, in update
    area_states[area_id] = self.areas[area_id].update(read_only)
  File "/home/pi/TerrariumPI/terrariumArea.py", line 570, in update
    light_state = ('on'     if self.enclosure.lights_on   else 'off')
  File "/home/pi/TerrariumPI/terrariumEnclosure.py", line 120, in lights_on
    return self.__light_status() == True
  File "/home/pi/TerrariumPI/terrariumEnclosure.py", line 48, in __light_status
    return main_lights.state['day']['powered']
KeyError: 'day'

Thanks.

theyosh, 2022-08-29:
Ok, I added an update which would hopefully fix this. So do a `git pull` and try again with a manual start.

I get the feeling that you have configured lights area that does not have relays on the day tab.

So when this is working again, please post a screenshot of your main lights area setup.

Python823, 2022-08-29:
Hi Yosh,

I stopped the service, git pull, started it again, but no webinterface so far.
I also did a reboot, but no change.

This is the log i get right now

2022-08-29 14:01:37,900 - INFO    - terrariumEngine       - Starting TerrariumPI 4.4.0 ...
2022-08-29 14:01:37,970 - INFO    - terrariumEngine       - Loaded 28 settings in 0.07 seconds.
2022-08-29 14:01:38,511 - INFO    - terrariumEngine       - Loading existing sensors from database.
2022-08-29 14:01:38,542 - ERROR   - hardware.sensor       - Sensor Sensirion SHT3X sensor temperature named 'Dembe_Basking_T                                                                          ' at address '44' has exception: I2C transceive failed: [Errno 121] Remote I/O error
2022-08-29 14:01:38,546 - WARNING - hardware.sensor       - Invalid reading from sensor Sensirion SHT3X sensor temperature n                                                                          amed 'Dembe_Basking_T' at address '44', retrying in 0.5 seconds...
2022-08-29 14:01:39,069 - ERROR   - hardware.sensor       - Sensor Sensirion SHT3X sensor temperature named 'Dembe_Basking_T                                                                          ' at address '44' has exception: I2C transceive failed: [Errno 121] Remote I/O error
2022-08-29 14:01:39,071 - WARNING - hardware.sensor       - Invalid reading from sensor Sensirion SHT3X sensor temperature n                                                                          amed 'Dembe_Basking_T' at address '44', retrying in 0.5 seconds...
2022-08-29 14:01:39,595 - ERROR   - hardware.sensor       - Sensor Sensirion SHT3X sensor temperature named 'Dembe_Basking_T                                                                          ' at address '44' has exception: I2C transceive failed: [Errno 121] Remote I/O error
2022-08-29 14:01:39,597 - ERROR   - hardware.sensor       - Error updating sensor Sensirion SHT3X sensor temperature named '                                                                          Dembe_Basking_T' at address '44'. Check your hardware! Invalid reading from sensor Sensirion SHT3X sensor temperature named                                                                           'Dembe_Basking_T' at address '44'
2022-08-29 14:01:39,600 - WARNING - terrariumEngine       - sht3x temperature named 'Dembe_Basking_T' at address '44' had pr                                                                          oblems reading a new value during startup in 1.08 seconds. Will be updated in the next round.
2022-08-29 14:01:39,602 - INFO    - terrariumEngine       - Scanning for new sensors ...
2022-08-29 14:01:48,810 - INFO    - terrariumEngine       - Loaded 1 sensors in 10.30 seconds.
2022-08-29 14:01:48,811 - INFO    - terrariumEngine       - Loading existing relays from database.
2022-08-29 14:01:48,827 - INFO    - terrariumEngine       - Scanning for new relays ...
No GEMBIRD SiS-PM found. Check USB connections, please!
2022-08-29 14:01:59,904 - INFO    - terrariumEngine       - Loaded 0 relays in 11.09 seconds.
2022-08-29 14:01:59,906 - INFO    - terrariumEngine       - Loading existing buttons from database.
2022-08-29 14:01:59,925 - INFO    - terrariumEngine       - Loaded 0 buttons in 0.02 seconds.
2022-08-29 14:01:59,926 - INFO    - terrariumEngine       - Loading existing webcams from database.
2022-08-29 14:01:59,936 - INFO    - terrariumEngine       - Loaded 0 webcams in 0.01 seconds.
2022-08-29 14:01:59,937 - INFO    - terrariumEngine       - Loading existing enclosures from database.
Traceback (most recent call last):
  File "terrariumPI.py", line 16, in <module>
    terrariumEngine = terrariumEngine(__version__)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 150, in __init__
    self.__load_existing_enclosures()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 1045, in __load_existing_enclosures
    new_enclosure.update()
  File "/home/pi/TerrariumPI/terrariumEnclosure.py", line 101, in update
    area_states[area_id] = self.areas[area_id].update(read_only)
  File "/home/pi/TerrariumPI/terrariumArea.py", line 574, in update
    self.state['is_day'] = self.is_day
  File "/home/pi/TerrariumPI/terrariumArea.py", line 553, in is_day
    is_day_time = main_lights_area.state['day']['begin'] < int(datetime.datetime.now().timestamp()) < main_lights_area.state['day']['end']
KeyError: 'day'

So your hunch is that i didn't configure my relays, area's and enclosures in the right order?
Thanks

theyosh, 2022-08-29:
Hmm, I think that you have created a setup which should not be possible. So for now, the easiest way is to remove the existing database and start over.

Delete all `terrariumpi.db*` in the folder data and restart TP4.

Python823, 2022-08-30:
Thanks for your help. This did indeed solve the problem. This ticket can be closed.

commit 6836f598 (2022-08-29): Fix #735
closed 2022-08-30 by Python823