theyosh/TerrariumPI #736: [BUG]:

Konrad84, 2022-09-22:
- Raspberry PI: 3B

can´t get weather information over openweather 

Create Account and API 
Tried following Calls:

https://api.openweathermap.org/data/2.5/weather?q=Erding,DE&appid=937cf8c4fcd394e0475ea09ef8ecafa7
https://api.openweathermap.org/data/2.5/weather?lat=44.34&lon=10.99&appid=52f3f67c1918cdde836256dd17753170
[image]

theyosh, 2022-09-22:
Hmm, I just tested with your weather url, but that does give an error. So I have to look into it.

Your first url is the correct one. That should work. I removed your appid value, as you should that keep secret ;)

theyosh, 2022-09-22:
Hmm, this is strange. I get a 401 error with your API key on a second call. That is something I cannot fix. What you can do is the following. You could try to generate a new API key in your account at Openweather. And then test the following two urls:

https://api.openweathermap.org/data/2.5/weather?q=Erding,DE&appid=[YOUR_NEW_API_KEY]
https://api.openweathermap.org/data/2.5/onecall?lat=48.306&lon=11.9069&units=metric&exclude=minutely&appid=[YOUR_NEW_API_KEY]&lang=nl

The first url is the url you enter in TP4. The second url is the url I construct to get more data. And the second url does give a 401 error with your old key. More info: https://openweathermap.org/faq#error401

But this is strange, as the first call is ok with your key, but the second is not. This is an issue at Openweather. So try to make a new key and see if that solves it. If not, you should contact Openweather why the first url is working, and the second is not.

Your free subscription plan is sufficient for all the online calls I do with TP4. So you should not have to have a paid plan.

I hope with a new API key it will work...

Konrad84, 2022-10-02:
Hi there,

 

thank you for your message! 

 

Not he new Key doesen´t work ether.

 

I get a error .. i wrote to openweather may they can give a hint.

 

https://api.openweathermap.org/data/2.5/weather?q=Erding,DE <https://api.openweathermap.org/data/2.5/weather?q=Erding,DE&appid=%5BYOUR_NEW_API_KEY%5D> &appid=[YOUR_NEW_API_KEY]
https://api.openweathermap.org/data/2.5/onecall?lat=48.306 <https://api.openweathermap.org/data/2.5/onecall?lat=48.306&lon=11.9069&units=metric&exclude=minutely&appid=%5BYOUR_NEW_API_KEY%5D&lang=nl> &lon=11.9069&units=metric&exclude=minutely&appid=[YOUR_NEW_API_KEY]&lang=nl

STGMavrick, 2022-10-03:
Hi  theyosh - I have the same issue as this guy. I sent you an email on 9/24 about it to the Gmail address listed.

Konrad84, 2022-10-05:
Hi there i sent a message to openweathermap and they answered that my api ist working fine they send me this:

 

Hello, thank you for contacting us.

Your API call works fine with no issues: https://api.openweathermap.org/data/2.5/weather?q=Erding,de <https://api.openweathermap.org/data/2.5/weather?q=Erding,de&APPID=de7a1fbd5f74afc83284ce53b8c9b504> &APPID=de7a1fbd5f74afc83284ce53b8c9b504

If you have any other questions, please don’t hesitate and contact us.

Kind regards,
Artūras Rosliakas
Technical Support Specialist

 

If i use the api URL in terrarium i get this Error

theyosh, 2022-10-05:
Hmm, it looks like they have changed there API pricing plans and therefore some urls. I need to investigate with them... :(

theyosh, 2022-10-05:
> Hi theyosh - I have the same issue as this guy. I sent you an email on 9/24 about it to the Gmail address listed.

Not sure where you mailed to, but I do not have a gmail address...

STGMavrick, 2022-10-05:
Sorry - not Gmail. It was to terrariumpi@theyosh.nl on 9/24.

theyosh, 2022-10-07:
Do a git pull and restart. It should work now for @Konrad84

For @STGMavrick you should change the url to: https://api.openweathermap.org/data/2.5/weather?q=Cincinnati,US&APPID=[YOUR_KEY]
Change is Cincinnati,OH,USA to Cincinnati,US

STGMavrick, 2022-10-07:
@theyosh Following your instructions I get the following. Just in case it was an appid issue I created another appid and I get the same result.

- 2022-10-07 18:38:32,725 - ERROR   - terrariumEngine       - Loading weather exception: 'appid'
2022-10-07 18:38:30,642 - INFO    - terrariumEngine       - Loading weather data from source https://api.openweathermap.org/data/2.5/weather?q=Cincinnati,US&APPID=34fe561b71f03364089c504440ce43fb

theyosh, 2022-10-08:
Totally my fault. Do another `git pull` and should be fixed. Sorry for the typo in code.

STGMavrick, 2022-10-08:
@theyosh That fixed it! Thank you!

theyosh, 2022-10-08:
There will be another update about getting more weather info. I think you should also subscribe for the One call api 3.0 in your account. But that needs entering a credit card :( I asked them about this, in order to see how I can still use their API.

Now you miss some forecast and the history data. But the basic data is there.

Konrad84, 2022-10-08:
Wow thank you a lot now it works !!!!

 

Hmm ok we´ll how long 😊

 

Nice !

theyosh, 2022-10-08:
Also updated information at the online help: https://theyosh.github.io/TerrariumPI/setup/#weather

Konrad84, 2022-10-14:
Hi There,

 

thanks for your help and sorry form y nooby questions.. i try to get camera started. 

 

I haver one IR Camera and dont get it working (Black Screen) and now i tried another one without IR and i get only Black screen. No Original Raspberry Pi cams .. is this a problem ? Should i chang somthing with the configuration?

theyosh, 2022-10-15:
Change the white balance to auto? The white balance setting is mostly the cause of a black screen

Konrad84, 2022-10-15:
Thats it Genius ! 😊 Thank YOu

Konrad84, 2022-10-16:
Hopefully Last Question for now. What do you use foor connecting to the Turtle Pin threw the Internet? I got a Fritzbox. So you Connect via vpn? You got an advice Here?

theyosh, 2022-10-23:
That is a bit outside the scope of TP4. But this should do it: https://www.edpnet.be/en/support/installation-and-usage/internet/manage-fritz!box/how-to-set-up-a-port-forwarding-on-a-fritzbox.html

It is called port forwarding. That means that everybody can find your TP4 setup online. It can take some time. but eventually it will be found. That is not a big problem, but you should know that.

Using VPN is a bit harder, and I cannot help you with that.

theyosh, 2022-10-30:
Closing as initial issue is fixed

closed 2022-10-30 by theyosh