theyosh/TerrariumPI #737: [FEATURE]: Sort Sensors, group average for different enclosures

sni89, 2022-09-27:
Hey, 

I have the following problem: 

I am running 4 enclosures with one TP System. If I want to have a quick view on parameters of a specific one it’s possible by manually checking sensor data. So here I am struggling with a possibility to regroup sensors to, for example, “enclosure 1, enclosure 2, …” these could hook up to a average for temperature, humidity and whatever. So my dashboard would give me 4 sections with average sensor data for individual enclosures. 

I hope my problem is understandable. And maybe there already is a way I don’t see :) 

I could also set up one TP for each enclosure. But that would be a little overdose.

theyosh, 2022-10-07:
Hi,

sorry for the delay. I think I see you wish. Let me try to rephrase.

Show the average sensor values per shown enclosure on the right side in stead of overall average.

I was thinking about this, but it is a bit hard to make in the current version. If I add more graphs on the dashboard, it get to much. So I think a nice thing could be that you can just toggle at the enclosure part (right side) to show the average sensors graphs for that enclosure only. And when you open another enclosure, the average sensors graphs changes to that enclosure.

But this will introduce a lot of sensor graphs loading. So for the current version I do not see this happening. But there is good news, as I am working on a new GUI. Which is reactive, and then it should be doable. But I am not sure when that will be ready.

And a question back, if you have averages per enclosure on the dashboard, do you need still the overall average graphs? Because I can also make a general setting, which shows only the averages of the active enclosure. But then you will miss the overall average sensors graphs.

sni89, 2022-10-07:
Hey, 
for me (keeping Uroplatus species) it’s important to see how the enclosure conditions are at a specific moment. For example pairs mating better than others, now the enclosure data would give a good base to work things out. 

For my use graphs are nice to have but just as a gimmick. No need to be visualized. It would be clearer if there’s only values.  

a bonus would be if you could add sensors as “Surrounding parameter” like room temperature. 
But this could be easy done if put all relevant sensors in a separate Enclosure. 

Example : 

Room
21,00C
45% humidity 

1.1 U. Phantasticus (enclosure name)
23,00C
78% humidity 
UVA 
UVB 

1.1 U. Fimbriatus (enclosure name)
28,00C
68% humidity 
UVA 
UVB 

…

thanks for your engagement and time!!! 
Can’t wait for that GUI you mentioned 🙂

theyosh, 2022-10-23:
Sorry for the late reply, but this sounds a bit more like an analytical thing, which is a bit outside the scope of TP4. I think that you could be better of getting the sensor database to a CSV file and use Excel or SPSS6 for it. Then you can combine all kind of sensors and compare to other sensor values. Creating a CSV file from the database is not that difficult to make.

Or use Graphana or other data graphing tool.

Else I still do not grasp your wish.

theyosh, 2022-11-02:
> So my dashboard would give me 4 sections with average sensor data for individual enclosures.

What about just showing the four enclosures next to each other, and no graphs on the dashboard. As the enclosure overview does show the average of the used sensors. But you will then only have the actual values. Going back in time is hard, and if you tried to get a month of data in the graphs, it starts to struggle. That is why I am a bit reluctant in combining all kind of sensor data and do calculation on it.

sni89, 2022-11-03:
The enclosures with actual sensor values would be great. it does not need to archive sensor data. i could do that by hand. 
I have no clue of programming but would it be possible to add a section while sensor installation where you can assign the sensor to a specific enclosure and then add it to the dashboard?  I made a short sketch to make it a little easier to understand what i want. =) The blue boxes would be my enclosures. 
[image]

At the moment is just set up scenarios with no effect where i use the sensors. So in my dashboard it shows the enclosures with all its scenarios and the actual value.  Like in the red sqare in the sketch

theyosh, 2022-11-07:
Hmm, that means you show all enclosures next to each other. That can be done. And do you want the average graphs still below it?

But you select the sensors in an area. That is right way. So per enclosure you create the areas. And than per area you can select the sensors you want.

sni89, 2022-11-22:
Hello, 

(for me!) it would be the best if it show all enclosures next to each other with actual values. I don’t need the average graphs because they include values of different enclosures. This makes them useless to me. Like setting up a display with actual sensor data just inside TP. 

I am a coding noob so sorry for being annoying. 
I tried to find a way to grab the values from database to show it in a separate gui but failed 😂 
If there would be a way to sort sensors or put them in folders that would be a good solution, too. If I get it right TP sorts them by their name. So for now my noobsolution is to give them numbers in their sensor names.

theyosh, 2022-11-28:
The only way I see is that you create the amount of enclosures you need to make all the areas you want. And you can reuse sensors in multiple areas. And you do not need to fill in the low or high alarm values. This will turn the areas in more like informative. And I thinks suitable for your wish.

theyosh, 2022-12-04:
If you are able to checkout the 4.x.y.z branch, rerun the installer and restart TP4, you should now be able to setup your dashboard that you can only show enclosures. This is the best I can do. You should then create the amount of enclosures and areas you need.

sni89, 2022-12-05:
Ty! Again I am a noob in coding. so I see the 4.xyz branch but how to use it? I did an update but no changes 🙈 I am sorry for wasting your time.

theyosh, 2022-12-05:
No problem. I think the following commands will do it. Make sure you are in the TerrariumPI folder and that TerrariumPI is stopped. https://theyosh.github.io/TerrariumPI/faq/how-to-start-stop-restart-and-disable-enable-terrariumpi/

Than run the following

1. `git fetch`
2. `git checkout 4.x.y.z`
3. `sudo ./install.sh`

And start TP4 again. A reboot is not needed and you should have the new gui. Looks a lot the same, but you will notice that a login is not reloading the page anymore ;). And than at the settings page, you can change the dashboard look.

sni89, 2022-12-05:
One last problem 
When I do 
`git checkout 4.x.y.z`
I get 
`error: pathspec '4.x.y.z' did not match any file(s) known to git `

😢

theyosh, 2022-12-05:
Does `git fetch origin` and then `git checkout 4.x.y.z` work?

sni89, 2022-12-05:
`pi@raspberrypi:~/TerrariumPI $ git fetch origin
pi@raspberrypi:~/TerrariumPI $ git checkout 4.x.y.z
error: pathspec '4.x.y.z' did not match any file(s) known to git `
Sadly it won’t

theyosh, 2022-12-05:
Ok, after some searching this should do the trick:

1. `git remote set-branches origin '*'`
2. `git fetch`
3. `git checkout 4.x.y.z`

And then rerun the installer: `sudo ./install.sh`

sni89, 2022-12-05:
Works! Thank you so much 😊

theyosh, 2022-12-08:
Do you have multiple enclosures on the dashboard? Could you post a screenshot? I only have 2 at max..

sni89, 2022-12-09:
Here are some from my phone. Will add better ones if needed. 

Btw there is a funny translation mistake in the Uptime section „zweites“ should be „Sekunden“ 😝 

[image]
[image]

theyosh, 2022-12-09:
Wrong translations.... https://weblate.theyosh.nl/projects/terrariumpi/gui/ ;)

You can just login to it with your existing Github account. And change the translation ... I did a lot with Google translate automatically, so any translation was valid for me :)

sni89, 2022-12-10:
Ty. Went through this several times before to help translate 😊 I will have a look into it again.

theyosh, 2022-12-10:
I also made a small update for the dashboard. On mobile, they should now be above each other. And when the screen gets bigger, more and more next to each other.

sni89, 2022-12-10:
Nice. I do the normal update to update this „special“ version?!

theyosh, 2022-12-10:
Just do a `git pull` and refresh in the browser. That should do it.

theyosh, 2022-12-11:
I just released 4.6.0. So that means for you that you can switch back to the main branch and have the same functionality. This is also better for the future as new released will be done in the main branch.

Thanks for testing.

sni89, 2022-12-12:
Great just got up to date and had a look at the translation! Went trough all and changed a lot. I tried `./update_translation.sh` but have no effect. I guess my input has to be confirmed or something to be integrated. 

here’s a quick view on my dashboard. 
[E09540F7-19B6-4B31-BD87-FC5338B8B36F](https://user-images.githubusercontent.com/56272047/207055695-d25d919a-fdf7-4ac4-875b-76aa6e80b134.jpeg)`

theyosh, 2022-12-12:
If you did the translations with weblate.theyosh.nl then they are now merged to main branch. The translations did go to the 4.x.y.z branch, but I have changed that.

The ./update_translations.sh is for developers finding new to translate strings in Python.

So, for you, all the translations actions are done at https://weblate.theyosh.nl

theyosh, 2022-12-12:
The screen looks better now, at least I think. You can now read all the data at once.

theyosh, 2022-12-12:
Ok, the German translation is done, which I am very thank full for. And I think that the enclosures on the dashboard is as good as I can make it.

So can we close this issue?

theyosh, 2022-12-12:
Oh, I think I just lost some of your translations on the gui. That was my fault. I guess it should not be much, but if it is the case, I am sorry.

sni89, 2022-12-13:
Went trouh the translation again. should be fine now. 

Thanks and closed :)

closed 2022-12-13 by sni89