theyosh/TerrariumPI #738: [BUG]: docker crash

cmonicob, 2022-10-07:
**Setup:**
 - TerrariumPI version: 4
 - Python version:  3
 - Raspberry PI: 4 8G RAM

**Describe the bug**
the latest version of terrarium installed from scratch, when booting the system on doker it restarts the raspberry

theyosh, 2022-10-07:
Hmm, that is everything? You can't see an error? And you run it in docker?  Because you say that you have installed it from scratch. But I guess you mean a clean Raspberry PI os, and then just docker and install the image.

What Debian version did you use? I have only tested on the legacy OS.

cmonicob, 2022-10-12:
the operating systems being tested are the following: Raspberry Pi OS Lite (Bullseye) 32-bit and 64-bit

theyosh, 2022-10-15:
Hmm, not sure of Bullseye is the issue. I tested on Buster. That is the legacy OS. Also I think you should stick to the 32bit. Else I am not sure if the camera will work.

You could also try the development version which is more recently build. https://hub.docker.com/r/theyosh/terrariumpi/tags But that is just a guess.

theyosh, 2022-10-18:
Damm, that is a shame... But, could you also try with the old legacy Buster OS?

cmonicob, 2022-10-18:
I was thinking if the solution would be to create the docker container directly in the Bullseye system since the volumes are highly connected with the system configuration
      - /boot/config.txt:/boot/config.txt
      - /boot/cmdline.txt:/boot/cmdline.txt
      - /etc/modules:/etc/modules
      - /opt/vc:/opt/vc
      - /dev:/dev
I don't know if it would be feasible to relocate some of these volumes, for example:
      - /opt/terrariumpi/boot/config.txt:/boot/config.txt
      - /boot/cmdline.txt:/boot/cmdline.txt
      - /etc/modules:/etc/modules [delete]
      - /opt/vc:/opt/vc [delete]
      - /dev:/dev [delete]

theyosh, 2022-10-18:
Hmm, that is not possible.

/dev is needed for I2C and USB hardware. 
/opt/vc is needed for the Pi Camera

For modules I am not sure, but I did not make the docker image. Somebody else made it I think about 2 years ago. And I do not think that Docker is much easier than just running the installer. As I need support for all kind of hardware and their way of connecting, that I did not make a Docker image initially.

Also, I do know that Bullseye does not work with the PiCamera. So that is why I do not fully support Bullseye OS.

So I am afraid that I cannot change much to the Docker image.

theyosh, 2022-10-30:
I made a totally different image called `svelte`. This will be the new version in a couple of month. The only big difference is a rewritten gui.

You could see if this image works better. I only saw that the docker image will restart the Pi the first time as some boot settings have to be set. But that comes with a warning and a timeout of 60 seconds before it reboots. If that is not the case... it will be hard to find out what is causing the restart.

theyosh, 2022-10-31:
Just made a clean install with the **Raspberry Pi OS Lite (legacy)** and use the svelte image and that just works.

```
docker compose logs -f
tp4-terrariumpi-1  | Enabled I2C in boot config
tp4-terrariumpi-1  | Enable I2C modules
tp4-terrariumpi-1  | Enabled 1 Wire in boot config
tp4-terrariumpi-1  | Enabled Camera in boot config (1)
tp4-terrariumpi-1  | Enabled Camera in boot config (2)
tp4-terrariumpi-1  | Enabled Serial in boot config
tp4-terrariumpi-1  | Enabled CO2 sensors in boot config (2)
tp4-terrariumpi-1  | Some settings have been updated that require reboot, your Raspberry Pi will reboot in 60 seconds...
tp4-terrariumpi-1  | 2022-10-31 21:31:38,823 - INFO    - terrariumEngine       - Starting TerrariumPI 4.5.0 ...
tp4-terrariumpi-1  | 2022-10-31 21:31:38,981 - INFO    - terrariumEngine       - Loaded 29 settings in 0.16 seconds.
tp4-terrariumpi-1  | 2022-10-31 21:43:34,679 - INFO    - terrariumEngine       - Loading existing sensors from database.
tp4-terrariumpi-1  | 2022-10-31 21:43:34,686 - INFO    - terrariumEngine       - Scanning for new sensors ...
tp4-terrariumpi-1  | 2022-10-31 21:43:42,483 - INFO    - terrariumEngine       - Added new new sensor miflora temperature named 'MiFlora bluetooth sensor measuring temperature' at address '80:ea:ca:88:d4:d4' to database with value 22.90C in 0.31 seconds.
tp4-terrariumpi-1  | 2022-10-31 21:43:42,491 - INFO    - terrariumEngine       - Added new new sensor miflora light named 'MiFlora bluetooth sensor measuring light' at address '80:ea:ca:88:d4:d4' to database with value 3368.00lux in 0.00 seconds.
tp4-terrariumpi-1  | 2022-10-31 21:43:42,496 - INFO    - terrariumEngine       - Added new new sensor miflora moisture named 'MiFlora bluetooth sensor measuring moisture' at address '80:ea:ca:88:d4:d4' to database with value 45.00% in 0.00 seconds.
tp4-terrariumpi-1  | 2022-10-31 21:43:42,500 - INFO    - terrariumEngine       - Added new new sensor miflora fertility named 'MiFlora bluetooth sensor measuring fertility' at address '80:ea:ca:88:d4:d4' to database with value 223.00µS/cm in 0.00 seconds.
tp4-terrariumpi-1  | 2022-10-31 21:43:44,818 - INFO    - terrariumEngine       - Added new new sensor miflora temperature named 'MiFlora bluetooth sensor measuring temperature' at address 'c4:7c:8d:67:11:40' to database with value 20.70C in 0.40 seconds.
tp4-terrariumpi-1  | 2022-10-31 21:43:44,830 - INFO    - terrariumEngine       - Added new new sensor miflora light named 'MiFlora bluetooth sensor measuring light' at address 'c4:7c:8d:67:11:40' to database with value 90.00lux in 0.01 seconds.
tp4-terrariumpi-1  | 2022-10-31 21:43:44,840 - INFO    - terrariumEngine       - Added new new sensor miflora moisture named 'MiFlora bluetooth sensor measuring moisture' at address 'c4:7c:8d:67:11:40' to database with value 59.00% in 0.01 seconds.
tp4-terrariumpi-1  | 2022-10-31 21:43:44,848 - INFO    - terrariumEngine       - Added new new sensor miflora fertility named 'MiFlora bluetooth sensor measuring fertility' at address 'c4:7c:8d:67:11:40' to database with value 1380.00µS/cm in 0.01 seconds.
tp4-terrariumpi-1  | 2022-10-31 21:43:46,951 - INFO    - terrariumEngine       - Added new new sensor miflora temperature named 'MiFlora bluetooth sensor measuring temperature' at address 'c4:7c:8d:66:e3:52' to database with value 22.20C in 0.34 seconds.
tp4-terrariumpi-1  | 2022-10-31 21:43:46,959 - INFO    - terrariumEngine       - Added new new sensor miflora light named 'MiFlora bluetooth sensor measuring light' at address 'c4:7c:8d:66:e3:52' to database with value 1714.00lux in 0.00 seconds.
tp4-terrariumpi-1  | 2022-10-31 21:43:46,964 - INFO    - terrariumEngine       - Added new new sensor miflora moisture named 'MiFlora bluetooth sensor measuring moisture' at address 'c4:7c:8d:66:e3:52' to database with value 36.00% in 0.00 seconds.
tp4-terrariumpi-1  | 2022-10-31 21:43:46,969 - INFO    - terrariumEngine       - Added new new sensor miflora fertility named 'MiFlora bluetooth sensor measuring fertility' at address 'c4:7c:8d:66:e3:52' to database with value 287.00µS/cm in 0.00 seconds.
tp4-terrariumpi-1  | 2022-10-31 21:43:50,018 - INFO    - terrariumEngine       - Loaded 12 sensors in 15.34 seconds.
tp4-terrariumpi-1  | 2022-10-31 21:43:50,019 - INFO    - terrariumEngine       - Loading existing relays from database.
tp4-terrariumpi-1  | 2022-10-31 21:43:50,040 - INFO    - terrariumEngine       - Scanning for new relays ...
tp4-terrariumpi-1  | 2022-10-31 21:43:52,045 - INFO    - terrariumEngine       - Added new relay eg-pm-usb relay named 'Energenie USB device nr: 1(01:ff:ff:ff:ff), Socket: 1' at address '1,01:ff:ff:ff:ff' to database with current value 0.00.
tp4-terrariumpi-1  | 2022-10-31 21:43:52,167 - INFO    - terrariumEngine       - Added new relay eg-pm-usb relay named 'Energenie USB device nr: 1(01:ff:ff:ff:ff), Socket: 2' at address '2,01:ff:ff:ff:ff' to database with current value 0.00.
tp4-terrariumpi-1  | 2022-10-31 21:43:52,280 - INFO    - terrariumEngine       - Added new relay eg-pm-usb relay named 'Energenie USB device nr: 1(01:ff:ff:ff:ff), Socket: 3' at address '3,01:ff:ff:ff:ff' to database with current value 100.00.
tp4-terrariumpi-1  | 2022-10-31 21:43:52,388 - INFO    - terrariumEngine       - Added new relay eg-pm-usb relay named 'Energenie USB device nr: 1(01:ff:ff:ff:ff), Socket: 4' at address '4,01:ff:ff:ff:ff' to database with current value 0.00.
[... cut ...]
tp4-terrariumpi-1  | 2022-10-31 21:44:04,654 - INFO    - terrariumEngine       - Loaded 0 webcams in 0.01 seconds.
tp4-terrariumpi-1  | 2022-10-31 21:44:04,655 - INFO    - terrariumEngine       - Loading existing enclosures from database.
tp4-terrariumpi-1  | 2022-10-31 21:44:04,660 - INFO    - terrariumEngine       - Loaded 0 enclosures in 0.00 seconds.
tp4-terrariumpi-1  | 2022-10-31 21:44:04,853 - INFO    - terrariumEngine       - TerrariumPI is up and running at address: http://0.0.0.0:8090 in 747.26 seconds.

```

theyosh, 2022-11-02:
Ok, after some more testing, I think the docker image with the tag `svelte` should work. So, you could try to try that.

Also, I found out that previous versions could break your /boot/config file. If it is all on a single line, you have to restore it. But this is only on a Raspberry Pi 4 and Bullseye.

cmonicob, 2022-11-05:
after doing several tests to get the portainer logs, I start directly from docker compose.

```
docker compose logs -f
TerrariumPi  | 2022-11-05 19:30:22 initInitialise: bind to port 8888 failed (Cannot assign requested address)
TerrariumPi  | Can't initialise pigpio library
TerrariumPi  | Traceback (most recent call last):
TerrariumPi  |   File "/TerrariumPI/terrariumPI.py", line 13, in <module>
TerrariumPi  |     from terrariumEngine import terrariumEngine
TerrariumPi  |   File "/TerrariumPI/terrariumEngine.py", line 2, in <module>
TerrariumPi  |     from hardware.webcam.rpilive_webcam import terrariumRPILiveWebcam
TerrariumPi  |   File "/TerrariumPI/hardware/webcam/__init__.py", line 20, in <module>
TerrariumPi  |     import cv2
TerrariumPi  |   File "/opt/venv/lib/python3.7/site-packages/cv2/__init__.py", line 181, in <module>
TerrariumPi  |     bootstrap()
TerrariumPi  |   File "/opt/venv/lib/python3.7/site-packages/cv2/__init__.py", line 153, in bootstrap
TerrariumPi  |     native_module = importlib.import_module("cv2")
TerrariumPi  |   File "/usr/local/lib/python3.7/importlib/__init__.py", line 127, in import_module
TerrariumPi  |     return _bootstrap._gcd_import(name[level:], package, level)
TerrariumPi  | ImportError: libmmal_core.so: cannot open shared object file: No such file or directory
TerrariumPi  | 2022-11-05 19:30:41 initInitialise: bind to port 8888 failed (Cannot assign requested address)
TerrariumPi  | Can't initialise pigpio library
TerrariumPi  | Traceback (most recent call last):
TerrariumPi  |   File "/TerrariumPI/terrariumPI.py", line 13, in <module>
TerrariumPi  |     from terrariumEngine import terrariumEngine
TerrariumPi  |   File "/TerrariumPI/terrariumEngine.py", line 2, in <module>
TerrariumPi  |     from hardware.webcam.rpilive_webcam import terrariumRPILiveWebcam
TerrariumPi  |   File "/TerrariumPI/hardware/webcam/__init__.py", line 20, in <module>
TerrariumPi  |     import cv2
TerrariumPi  |   File "/opt/venv/lib/python3.7/site-packages/cv2/__init__.py", line 181, in <module>
TerrariumPi  |     bootstrap()
TerrariumPi  |   File "/opt/venv/lib/python3.7/site-packages/cv2/__init__.py", line 153, in bootstrap
TerrariumPi  |     native_module = importlib.import_module("cv2")
TerrariumPi  |   File "/usr/local/lib/python3.7/importlib/__init__.py", line 127, in import_module
TerrariumPi  |     return _bootstrap._gcd_import(name[level:], package, level)
TerrariumPi  | ImportError: libmmal_core.so: cannot open shared object file: No such file or directory
TerrariumPi  | 2022-11-05 19:30:56 initInitialise: bind to port 8888 failed (Cannot assign requested address)
TerrariumPi  | Can't initialise pigpio library
TerrariumPi  | Traceback (most recent call last):
TerrariumPi  |   File "/TerrariumPI/terrariumPI.py", line 13, in <module>
TerrariumPi  |     from terrariumEngine import terrariumEngine
TerrariumPi  |   File "/TerrariumPI/terrariumEngine.py", line 2, in <module>
TerrariumPi  |     from hardware.webcam.rpilive_webcam import terrariumRPILiveWebcam
TerrariumPi  |   File "/TerrariumPI/hardware/webcam/__init__.py", line 20, in <module>
TerrariumPi  |     import cv2
TerrariumPi  |   File "/opt/venv/lib/python3.7/site-packages/cv2/__init__.py", line 181, in <module>
TerrariumPi  |     bootstrap()
TerrariumPi  |   File "/opt/venv/lib/python3.7/site-packages/cv2/__init__.py", line 153, in bootstrap
TerrariumPi  |     native_module = importlib.import_module("cv2")
TerrariumPi  |   File "/usr/local/lib/python3.7/importlib/__init__.py", line 127, in import_module
TerrariumPi  |     return _bootstrap._gcd_import(name[level:], package, level)
TerrariumPi  | ImportError: libmmal_core.so: cannot open shared object file: No such file or directory
TerrariumPi  | 2022-11-05 19:31:10 initInitialise: bind to port 8888 failed (Cannot assign requested address)
TerrariumPi  | Can't initialise pigpio library
TerrariumPi  | Traceback (most recent call last):
TerrariumPi  |   File "/TerrariumPI/terrariumPI.py", line 13, in <module>
TerrariumPi  |     from terrariumEngine import terrariumEngine
TerrariumPi  |   File "/TerrariumPI/terrariumEngine.py", line 2, in <module>
TerrariumPi  |     from hardware.webcam.rpilive_webcam import terrariumRPILiveWebcam
TerrariumPi  |   File "/TerrariumPI/hardware/webcam/__init__.py", line 20, in <module>
TerrariumPi  |     import cv2
TerrariumPi  |   File "/opt/venv/lib/python3.7/site-packages/cv2/__init__.py", line 181, in <module>
TerrariumPi  |     bootstrap()
TerrariumPi  |   File "/opt/venv/lib/python3.7/site-packages/cv2/__init__.py", line 153, in bootstrap
TerrariumPi  |     native_module = importlib.import_module("cv2")
TerrariumPi  |   File "/usr/local/lib/python3.7/importlib/__init__.py", line 127, in import_module
TerrariumPi  |     return _bootstrap._gcd_import(name[level:], package, level)
TerrariumPi  | ImportError: libmmal_core.so: cannot open shared object file: No such file or directory
TerrariumPi  | 2022-11-05 19:31:25 initInitialise: bind to port 8888 failed (Cannot assign requested address)
TerrariumPi  | Can't initialise pigpio library
[... 167 lines cut ...]
TerrariumPi  |   File "/opt/venv/lib/python3.7/site-packages/cv2/__init__.py", line 181, in <module>
TerrariumPi  |     bootstrap()
TerrariumPi  |   File "/opt/venv/lib/python3.7/site-packages/cv2/__init__.py", line 153, in bootstrap
TerrariumPi  |     native_module = importlib.import_module("cv2")
[... cut ...]
TerrariumPi  |     bootstrap()
TerrariumPi  |   File "/opt/venv/lib/python3.7/site-packages/cv2/__init__.py", line 153, in bootstrap
TerrariumPi  |     native_module = importlib.import_module("cv2")
TerrariumPi  |   File "/usr/local/lib/python3.7/importlib/__init__.py", line 127, in import_module
TerrariumPi  |     return _bootstrap._gcd_import(name[level:], package, level)
TerrariumPi  | ImportError: libmmal_core.so: cannot open shared object file: No such file or directory
TerrariumPi exited with code 1
```

theyosh, 2022-11-05:
Remove the volume line `- /opt/vc:/opt/vc` from your docker-compose.yml and try again. That would at least start it.

cmonicob, 2022-11-05:
With this modification it no longer restarts the raspberry and starts the TerrariumPi, which causes an error

```
2022-11-05 22:31:24 initInitialise: bind to port 8888 failed (Cannot assign requested address)
Can't initialise pigpio library
2022-11-05 22:31:42,795 - INFO    - terrariumEngine       - Starting TerrariumPI 4.5.0 ...
2022-11-05 22:31:42,911 - INFO    - terrariumEngine       - Loaded 29 settings in 0.11 seconds.
2022-11-05 22:31:44,578 - INFO    - terrariumEngine       - Loading existing sensors from database.
2022-11-05 22:31:44,592 - INFO    - terrariumEngine       - Scanning for new sensors ...
2022-11-05 22:31:45,142 - WARNING - hardware.sensor       - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!
2022-11-05 22:31:45,543 - WARNING - hardware.sensor       - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!
2022-11-05 22:31:45,950 - WARNING - hardware.sensor       - Bluetooth scanning is not enabled for normal users or there are zero Bluetooth LE devices available.... bluetooth is disabled!
2022-11-05 22:31:46,062 - INFO    - terrariumEngine       - Loaded 0 sensors in 1.48 seconds.
2022-11-05 22:31:46,063 - INFO    - terrariumEngine       - Loading existing relays from database.
2022-11-05 22:31:46,079 - INFO    - terrariumEngine       - Scanning for new relays ...
No GEMBIRD SiS-PM found. Check USB connections, please!
2022-11-05 22:31:57,667 - INFO    - terrariumEngine       - Loaded 0 relays in 11.60 seconds.
2022-11-05 22:31:57,670 - INFO    - terrariumEngine       - Loading existing buttons from database.
2022-11-05 22:31:57,703 - INFO    - terrariumEngine       - Loaded 0 buttons in 0.03 seconds.
2022-11-05 22:31:57,705 - INFO    - terrariumEngine       - Loading existing webcams from database.
2022-11-05 22:31:57,735 - INFO    - terrariumEngine       - Loaded 0 webcams in 0.03 seconds.
2022-11-05 22:31:57,737 - INFO    - terrariumEngine       - Loading existing enclosures from database.
2022-11-05 22:31:57,749 - INFO    - terrariumEngine       - Loaded 0 enclosures in 0.01 seconds.
2022-11-05 22:31:57,800 - INFO    - terrariumEngine       - TerrariumPI is up and running at address: http://0.0.0.0:8090 in 16.35 seconds.
2022-11-05 22:31:57,000 - INFO    - terrariumWebserver    - Running webserver at 0.0.0.0:8090
2022-11-05 22:31:58,600 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,604 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,608 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,612 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,616 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,620 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,624 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,628 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,632 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,636 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,640 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,644 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,648 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,652 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,656 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,660 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,664 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,668 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,672 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,676 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,680 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,685 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,689 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,693 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,697 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,701 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,705 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,709 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,713 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
2022-11-05 22:31:58,717 - WARNING - terrariumEngine       - Killing hanging bluetooth helper process
/usr/local/lib/python3.7/threading.py:874: ResourceWarning: unclosed file <_io.TextIOWrapper name='/dev/null' mode='w' encoding='UTF-8'>
  del self._target, self._args, self._kwargs
ResourceWarning: Enable tracemalloc to get the object allocation traceback
/opt/venv/lib/python3.7/site-packages/bottle.py:3380: DeprecationWarning: Absolute template path names are deprecated.
  fname = self.search(name, self.lookup)
```

theyosh, 2022-11-07:
The webinterface should work. The only strange thing is the bluetooth killers. What kind of Pi are you using? Get the feeling it is a Raspberry PI zero.

cmonicob, 2022-11-09:
the web works correctly, the tests are done with my old raspberry pi 3 b+, since I am doing several tests to link with other machines within docker

theyosh, 2022-11-10:
Ok, I think I found the issue, and I have already created new images for the latest and 4.5.0 tag. If there are people with issues about rebooting Pis and docker images, I am now sure the latest should work.

theyosh, 2022-11-10:
And can we close this issue?

cmonicob, 2022-11-12:
The problem with bluetooth is because I don't use the (network_mode: host), the environment that I am preparing to be able to secure thanks to docker is that the accesses to the TerrariumPi are through nginx and to have a private network for the exchange of data between the machines of docker.
Is there a way for bluetooth to work without host mode?

theyosh, 2022-11-28:
I think running bluetooth without host mode will be hard. This will need installing and running a bluetooth daemon in the image, and not on the Pi it self. That is what I can find here: https://stackoverflow.com/questions/28868393/accessing-bluetooth-dongle-from-inside-docker

Or maybe this could be the solution: https://stackoverflow.com/a/71541458 This needs an extra mount point, and no host network mode anymore.

Extra info: https://stackoverflow.com/questions/24225647/docker-a-way-to-give-access-to-a-host-usb-or-serial-device/24231872#24231872

But I am unable to test it myself. I do not have that much time. Also not sure what other issues could arise by changing the network mode...

theyosh, 2022-12-08:
Closing this issue as the main cause is found and fixed. The bluetooth issues with Docker setup are outside my support scope

closed 2022-12-08 by theyosh