theyosh/TerrariumPI #744: [FEATURE]:  add tolerance/desviation in area actions

cmonicob, 2022-10-18:
In order to adjust the actions of the areas more precisely, it would be good to have a field that modifies the state of the relay/actuator.
leaving it at 0 would be normal operation but setting it to positive or negative values ​​modifies the actuation point on the minimum/maximum alarm

pongo aqui una posible solucion del terrariumArea.py para agregar esta caracteristica:

      if ignore_low_alarm:
        # Use the max alarm value to changing the relays
        self.state['sensors']['alarm_low']  = self.state['sensors']['current'] + self.state['sensors']['desviation'] < self.state['sensors']['alarm_max']
      else:
        # Normal state
        self.state['sensors']['alarm_low']  = self.state['sensors']['current'] + self.state['sensors']['desviation'] < self.state['sensors']['alarm_min']

      if ignore_high_alarm:
        # Use the min alarm value to changing the relays
        self.state['sensors']['alarm_high'] = self.state['sensors']['current'] + self.state['sensors']['desviation'] > self.state['sensors']['alarm_min']
      else:
        self.state['sensors']['alarm_high'] = self.state['sensors']['current'] + self.state['sensors']['desviation'] > self.state['sensors']['alarm_max']

theyosh, 2022-10-23:
I think I use the term offset for this. So in order to see if I understand it well.

You want to change the alarm values of the sensors in the area by an X amount. Is this just 1 value? This is then like the day/night difference field. When I drop or rise the alarm values based on day or night period.
Or could it be better to have this value per alarm type (low/high). So the low alarm shifts 3 and high alarm shifts 6 for example.

Because as you describe it now, the alarm values will just be raised or lowered by X amount.

Could you clarify a bit more?

G4KCM, 2022-10-23:
Just a comment here… in conventional on/off control it is normal to have a set point value i.e. the desired value and  a switching hysteresis value. Alarm values should be just alarms and have no impact upon the control.  Using this strategy would also allow further refinement by introducing time proportioned integral control which works well with heating control especially if solid state relays are used, it does though require a bit of effort tuning.  I used this previously in a terrarium control developed in node-red and it worked very well.

cmonicob, 2022-10-23:
The idea is to have these values ​​both in high and low alarm, even if it is possible in each relay, the idea is to modify the action parameters of the relays to achieve actions before reaching the alarm value.
For example:
I want to turn on a heater before reaching the minimum alarm "we could call a preventive action" in case of reaching the minimum, turn on other more powerful heaters.

The idea is that our reference points [maximum alarm] 30ºC and [minimum alarm] 20ºC, with these two modifiers in each relay we could have configurations that would allow us to turn on a heater [maximum alarm deviation] -2 [minimum alarm deviation] 5 in The heater would be working between 25ºC and 28ºC, or even turn on backup heaters with a [minimum alarm deviation] -5. This would mean that if it reaches 15 degrees, take more severe action to compensate for the loss of temperature.

What @G4KCM comments would be perfect to have a tab to have a target temperature with PID hysteresis settings for the heaters and all the common functions of a current resistance heating system, but I think with this modification we could have a pretty good approximation and could be used in all actions involving sensors

G4KCM, 2022-10-23:
Yes @cmonicob 🤙, what you suggest does offer a good option without going as far as PID.  What you have suggested is effectively on/off control with adjustable positive/negative hysteresis values so would be a good move forward and allow the alarm set point to be used as alarms or for remedial control actions.

cmonicob, 2022-11-08:
I understand the logic that you mention about alarms as actions, but for those of us who come from industrial control logic, we always work in four fields:
- low setpoint: actions we perform to achieve the desired value from below
- low alarm: support actions we take when we can't get the setpoint when this low alarm goes off (could just sound an alarm)
- high setpoint: actions we perform to achieve the desired value from above
- high alarm: support actions we take when we can't get the setpoint when it exceeds this high alarm (could just sound an alarm)

I don't know if it would be possible and/or logical to add these four tabs and the deviation only in the high setpoint and low setpoint tabs.
Thus, in the same area we could include all the control related to a probe.

anyway this is just an idea that I raise from my point of view

theyosh, 2022-11-10:
But, then with the two extra fields, you could do this I think. The alarm values are still there, but with the two deviation fields, you can lower or rise the alarm values based on the selected sensors.

Also, my idea is that the low alarm tab is used for controlling a minimum (temperature) value. So when it is getting to hot, the low alarm relays toggles off. But if you want active cooling, you can either use the high alarm tab, and activate there some cooling stuff. And with the two extra fields for each alarm tab, you can change the behavior independently of each other, where now, they share the same alarm and therefore toggle actions.
And in the code there is hard coded logic that either low alarm or high alarm is active. So it is not possible to have the relays of both alarm tabs active at the same time. So if you change the alarm values, and get some strange overlapping in values, I am not sure what will happen.

But my hole system is based on the the sensor logic, which a lot of people are using now, so it will be hard to change that logic or make it variable logic so you have even more options and choices.

And how do you display this in the gui. I think just showing the toggle values so you know the operating range. But the alarm icon will only show up when the alarm value is passed. That means I have to split this up, as now, the alarm values are also the operating range.

Long story, but I hope that with adding the two deviation fields you suggested on both alarm tabs, it can work for you. And that is for now what I can do.

cmonicob, 2022-11-10:
perfect, thank you very much.
I understand that as soon as the new gui is functional we will be able to work on these new functionalities and restructuring of the logic, I hope that soon we will be able to open a branch for this topic.
for anything that can help in the development I am here

theyosh, 2022-11-28:
Sorry for the late reply. I am trying to add the deviation support, but I guess my initial setup is not allowing me to have different deviations for the high and low alarm tabs. As I the alarm values on different places, which are only based on the sensor values, it is hard to have deviation values per low or high alarm tab. Than I need to have some logic on which alarm you want. So that is not working.

I can only support deviation as general value for the hole area. So, now I am thinking, when you select 1 or more sensors you get the option to have 2 deviation fields where you can change the low and high alarm values. But this will than be used for both low and high alarm tabs.

So, I hope, and think, you should still be able to make your setup, but a less flexible as I wanted...

theyosh, 2022-12-04:
I made a test version on the 4.x.y.z branch. This has now also the new GUI, so the area form could look a bit different.

But you have not to enter positive and negative values to change the low and high alarm value, and therefore all the relay triggering. This is the best I can do.

theyosh, 2022-12-12:
Hmm, this looks like a javascript error. Could you open your browser console, and see if you can find a javascript error. Please post it here.

Also, it could help to reload the page with holding the shift so you have the latest version. Maybe it is a caching issue.

Cinett87, 2022-12-12:
tested this in chrome and edge producing the same Java errors in both browsers

[image]

theyosh, 2022-12-12:
Ok, I found the issue. Do a `git pull` and restart TP4. This should fix it.

Cinett87, 2022-12-12:
fixed! sorry for hijacking this thread it was just an area thread

theyosh, 2022-12-12:
No worries... it is somewhat related ;)

theyosh, 2022-12-19:
@cmonicob Any change of checking this out? I ended up using this more like an offset, because it will also control the alarm trigger.. Else you would not see that the area is toggled on.

If you think is not correct, I can change it back. But this feeds more natural to me,

cmonicob, 2022-12-20:
I've been away and I haven't been able to deeply review the operation of the integration.
What would be good instead of marking as an alarm, I don't know if to monitor having a new status.
I really look at the Enclosures state, which I think the alarm state could be removed, which would indicate that it is working in normal mode.
I still think that being able to add four tabs (Low alarm, Low deviation, High deviation, High alarm) instead of two (Low alarm, High alarm) would be ideal so that all the Areas would contemplate the four possible states and the type of programming would be more visual applied

theyosh, 2023-03-10:
Closing due to no activity and code is implemented as best as possible.

closed 2023-03-10 by theyosh