theyosh/TerrariumPI #745: [BUG]: Daylight savings?

G4KCM, 2022-10-30:
Hi,
U.K. daylight savings ended last night…
Sunday, 30 October 2022, 02:00:00 clocks were turned backward 1 hour to Sunday, 30 October 2022, 01:00:00 to local standard time instead.  Around that time the heaters have caused quite a large temperature excursion.  I can’t see anything else happening around that time so thought I would mention it.[
[image]
[image]
](url)

theyosh, 2022-10-30:
I guess it is. I was a bit less effected as for me the relays did not toggle on. And we shift at 3 o'clock to 2 o'clock back. 

[image]

But the strange thing with your issue, is that it did not toggled down the relay based on temperature. The only thing I can think of is that there was a negative difference value, as you go back 1 hour. And that would case some triggers not to fire for 1 hour...

I can have a look at that difference calculation. But it is hard to test, and it only happens once 1 year... So not sure if I can fix this.

G4KCM, 2022-10-30:
Yes, what you say appears to be what happened.  Just thought I would mention it as it looks like the control loop cycle is linked to real time which might explain it.  Not a major issue for me either but if someone was pumping water it might be a bit more of an issue!

theyosh, 2022-11-07:
Made small adjustment in de settle time calculation. That is the only thing I can think off. Thanks for reporting.

closed 2022-11-07 by theyosh